esphome/issues #1276: Upgrade to 1.14.4 cause ESP32 bootloop

definitio, 2020-06-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Arch Linux, Python 3.8.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

https://esphome.io/components/wifi.html ?

**Description of problem:**
ESP32 doesn't start after upgrade.

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  ssid: !secret ssid
  password: !secret wifi_password
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[19:45:10]Rebooting...
[19:45:10]ets Jun  8 2016 00:22:57
[19:45:10]
[19:45:10]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:45:10]configsip: 0, SPIWP:0xee
[19:45:10]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:45:10]mode:DIO, clock div:2
[19:45:10]load:0x3fff0018,len:4
[19:45:10]load:0x3fff001c,len:1044
[19:45:10]load:0x40078000,len:8896
[19:45:10]load:0x40080400,len:5828
[19:45:10]entry 0x400806ac
[19:45:10][E][pcf8574:012]: PCF8574 not available under 0x27
[19:45:10][E][component:092]: Component was marked as failed.
[19:45:10][E][component:092]: Component was marked as failed.
[19:45:10][E][component:092]: Component was marked as failed.
[19:45:10][E][component:092]: Component was marked as failed.
[19:45:10][E][component:092]: Component was marked as failed.
[19:45:10]Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
[19:45:10]Core 1 register dump:
[19:45:10]PC      : 0x4008c366  PS      : 0x00060134  A0      : 0x8008b523  A1      : 0x3ffb0970  
WARNING Decoded 0x4008c366: vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:188
[19:45:10]A2      : 0x3ffb7ddc  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001  
[19:45:10]A6      : 0x00060123  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074  
[19:45:10]A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00000020  A13     : 0x00000000  
[19:45:10]A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x00000006  
[19:45:10]EXCVADDR: 0x00000000  LBEG    : 0x40001609  LEND    : 0x4000160d  LCOUNT  : 0x00000000  
[19:45:10]
[19:45:10]Backtrace: 0x4008c366:0x3ffb0970 0x4008b520:0x3ffb0990 0x400897ab:0x3ffb09b0 0x400fde97:0x3ffb09f0 0x400fdf62:0x3ffb0a10 0x40087966:0x3ffb0a30 0x40087e9a:0x3ffb0a50 0x40110565:0x3ffb0ac0 0x4010ee2d:0x3ffb0ae0 0x4010f27b:0x3ffb0b00 0x4010dc05:0x3ffb0b70 0x4010e001:0x3ffb0bc0 0x4010e138:0x3ffb0c40 0x4010d0ce:0x3ffb0ca0 0x4010d735:0x3ffb0ce0 0x4015c9e6:0x3ffb0d00 0x4015cbda:0x3ffb0d50 0x4015cc2d:0x3ffb0d80 0x40110a23:0x3ffb0da0 0x40110b76:0x3ffb0dc0 0x400fc289:0x3ffb0de0 0x400eca58:0x3ffb0e00 0x400df7b5:0x3ffb0f00 0x400df814:0x3ffb0f30 0x400debe8:0x3ffb0f60 0x40173051:0x3ffb0f80 0x40173139:0x3ffb0fa0 0x400e10cd:0x3ffb0fc0 0x400ea9aa:0x3ffb1010 0x400f9333:0x3ffb1fb0 0x400899a5:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008c366: vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:188
WARNING Decoded 0x4008b520: vTaskPlaceOnEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
WARNING Decoded 0x400897ab: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1996
WARNING Decoded 0x400fde97: esp_ipc_call_and_wait at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:116
WARNING Decoded 0x400fdf62: esp_ipc_call at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:123
WARNING Decoded 0x40087966: spi_flash_disable_interrupts_caches_and_other_cpu at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:120
WARNING Decoded 0x40087e9a: spi_flash_guard_start at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:162
 (inlined by) spi_flash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:508
WARNING Decoded 0x40110565: nvs::nvs_flash_read(unsigned int, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_ops.cpp:70
WARNING Decoded 0x4010ee2d: nvs::Page::readEntry(unsigned int, nvs::Item&) const at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:849
WARNING Decoded 0x4010f27b: nvs::Page::findItem(unsigned char, nvs::ItemType, char const*, unsigned int&, nvs::Item&, unsigned char, nvs::VerOffset) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:849
WARNING Decoded 0x4010dc05: nvs::Storage::findItem(unsigned char, nvs::ItemType, char const*, nvs::Page*&, nvs::Item&, unsigned char, nvs::VerOffset) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp:557
WARNING Decoded 0x4010e001: nvs::Storage::readMultiPageBlob(unsigned char, char const*, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp:557
WARNING Decoded 0x4010e138: nvs::Storage::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp:557
[... 36 lines cut ...]
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vPortCPUAcquireMutex at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:394
WARNING Decoded 0x4008a5ca: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
WARNING Decoded 0x4008c208: _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:406
WARNING Decoded 0x4008c1be: _frxt_int_exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:206
[19:45:11]
[19:45:11]Rebooting...
[19:45:11]ets Jun  8 2016 00:22:57
[19:45:11]
[19:45:11]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[19:45:11]configsip: 0, SPIWP:0xee
[19:45:11]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[19:45:11]mode:DIO, clock div:2
[19:45:11]load:0x3fff0018,len:4
[19:45:11]load:0x3fff001c,len:1044
[19:45:11]load:0x40078000,len:8896
[19:45:11]load:0x40080400,len:5828
[19:45:11]entry 0x400806ac
[19:45:11][E][ota:369]: Boot loop detected. Proceeding to safe mode.
[19:45:12][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
```

**Additional information and things you've tried:**
Flash erasing doesn't help. There are PCF8574 and other errors because nothing is connected to ESP32. I don't have DHT sensors, so it's not related to https://github.com/esphome/issues/issues/1269.

DoctorOctagonapus, 2020-06-06:
I have the same issue on mine, again no DHT sensor. Mine eventually boots into I'm guessing safe mode and connects to wifi, but will not talk to ESPHome over OTA. Can't upload a new flash and can't read the logs. (I'm getting the logs using serial.) I'll attach a dump of my logs.

```
INFO Reading configuration /config/esphome/fishtank.yaml...
INFO Detected timezone 'GMT' with UTC offset 0 and daylight savings time from 28 March 01:00:00 to 31 October 02:00:00
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[17:27:37]ets Jun  8 2016 00:22:57
[17:27:37]
[17:27:37]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:27:37]configsip: 0, SPIWP:0xee
[17:27:37]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:27:37]mode:DIO, clock div:2
[17:27:37]load:0x3fff0018,len:4
[17:27:37]load:0x3fff001c,len:1044
[17:27:37]load:0x40078000,len:8896
[17:27:37]load:0x40080400,len:5828
[17:27:37]entry 0x400806ac
[17:27:37][I][logger:156]: Log initialized
[17:27:37][C][status_led:014]: Setting up Status LED...
[17:27:37][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[17:27:37][I][app:028]: Running through setup()...
[17:27:37][D][binary_sensor:034]: 'Fishtank Water Level': Sending initial state OFF
[17:27:37][D][binary_sensor:034]: 'Light Switch': Sending initial state ON
[17:27:37][D][binary_sensor:034]: 'Pump Switch': Sending initial state ON
[17:27:37][C][switch.gpio:011]: Setting up GPIO Switch 'Fishtank Light'...
[17:27:37][D][switch:021]: 'Fishtank Light' Turning ON.
[17:27:37][D][switch:045]: 'Fishtank Light': Sending state ON
[17:27:37][D][switch:021]: 'Fishtank Light' Turning ON.
[17:27:37][C][switch.gpio:011]: Setting up GPIO Switch 'Fishtank Pump'...
[17:27:37][D][switch:021]: 'Fishtank Pump' Turning ON.
[17:27:37][D][switch:045]: 'Fishtank Pump': Sending state ON
[17:27:37][D][switch:021]: 'Fishtank Pump' Turning ON.
[17:27:38][C][ads1115:014]: Setting up ADS1115...
[17:27:38][W][i2c:070]: Received NACK on transmit of address 0x48
[17:27:38][E][component:092]: Component was marked as failed.
[17:27:38][C][bh1750.sensor:012]: Setting up BH1750 'Fishtank Light Level'...
[17:27:38][W][i2c:070]: Received NACK on transmit of address 0x23
[17:27:38][E][component:092]: Component was marked as failed.
[17:27:38][C][dallas.sensor:032]: Setting up DallasComponent...
[17:27:38][E][dallas.sensor:200]: Reading scratchpad failed: reset
[17:27:38][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[17:27:38][W][i2c:076]: Unknown transmit error 5 for address 0x3C
[17:27:38][E][component:092]: Component was marked as failed.
[17:27:38][C][wifi:033]: Setting up WiFi...
[17:27:38]Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
[17:27:38]Core 1 register dump:
[17:27:38]PC      : 0x4008c32a  PS      : 0x00060f34  A0      : 0x8008b4e7  A1      : 0x3ffb0df0  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /config/esphome/fishtank -t idedata
WARNING Decoded 0x4008c32a: vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:188
[17:27:42]A2      : 0x3ffb7d7c  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001  
[17:27:42]A6      : 0x00060f23  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074  
[17:27:42]A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00000020  A13     : 0x3ffbf938  
[17:27:42]A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x0000001a  EXCCAUSE: 0x00000006  
[17:27:42]EXCVADDR: 0x00000000  LBEG    : 0x40001609  LEND    : 0x4000160d  LCOUNT  : 0x00000000  
[17:27:42]
[17:27:42]Backtrace: 0x4008c32a:0x3ffb0df0 0x4008b4e4:0x3ffb0e10 0x4008976f:0x3ffb0e30 0x400ec6f2:0x3ffb0e70 0x400ec7b2:0x3ffb0e90 0x4008792a:0x3ffb0eb0 0x40087e5e:0x3ffb0ed0 0x400fdfed:0x3ffb0f40 0x400fc8b5:0x3ffb0f60 0x400fcd03:0x3ffb0f80 0x400fb68d:0x3ffb0ff0 0x400fc197:0x3ffb1040 0x400fab11:0x3ffb10a0 0x400fb1bd:0x3ffb10e0 0x40148092:0x3ffb1100 0x40148286:0x3ffb1150 0x401482d9:0x3ffb1180 0x400fe49b:0x3ffb11a0 0x400fe5ee:0x3ffb11c0 0x400eac65:0x3ffb11e0 0x400e49b8:0x3ffb1200 0x400dd125:0x3ffb1300 0x400dd184:0x3ffb1330 0x400dc4b0:0x3ffb1360 0x4015d805:0x3ffb1380 0x4015d8c1:0x3ffb13a0 0x400dea95:0x3ffb13c0 0x400e259e:0x3ffb1410 0x400e8067:0x3ffb1fb0 0x40089969:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008c32a: vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:188
WARNING Decoded 0x4008b4e4: vTaskPlaceOnEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
WARNING Decoded 0x4008976f: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1996
WARNING Decoded 0x400ec6f2: esp_ipc_call_and_wait at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:116
WARNING Decoded 0x400ec7b2: esp_ipc_call at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:123
WARNING Decoded 0x4008792a: spi_flash_disable_interrupts_caches_and_other_cpu at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c:120
[... cut ...]
From here it will boot multiple times, sometimes connecting to wifi and waiting for an OTA update that it can't receive (because ESPHome won't talk to it). Eventually it boots into safe mode but ultimately boot loops again.

I have a number of components connected to this; a sensor via an ADC, an OLED display and a light sensor all over i2c, a onewire temperature sensor, plus various binary sensors, couple of switches, couple of lights, and two GPIO controlled relays. 1.14.3 was able to handle it all without issue but I flashed it with 1.14.4 via OTA and it never worked again.

MarcoV-git, 2020-06-06:
Same problem here!
It does connect to Wifi, because ping is working. It seems that the internal webserver is not starting, therefor you cannot access API- nor HTTP-endpoint.
OTA flashing is still working, as long as you get the firmware uploaded before the ESP reboots... ;-) Currently no FTDI at hand, so no serial logs from me... v1.14.3 is still working fine.

DoctorOctagonapus, 2020-06-06:
Anyone know if it's possible to roll the firmware back to 1.14.3? I'm running mine on HassOS.

HugoVonHory, 2020-06-06:
I have the same issue caused by Dallas component, after upgrade it reboots several times with "Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)" and switches to safe mode. It looks it isn't wifi issue, when Dallas removed from configuration ESP32 works again (binary sensors and switch in config).

DoctorOctagonapus, 2020-06-06:
Can confirm. I have a DS18b20 sensor, mine booted fine once I commented out all the code for that and reflashed.

Can't find anything in the changelogs relating to the Dallas component, so no idea what caused it to break.

I don't have any other DS18b20s on any other devices so no way of checking if this problem also affects ESP8266s.

definitio, 2020-06-06:
I have one esp8266 with ds18b20 sensor. It works fine after update. The
esp32 doesn't have 1-wire sensors.

matthewcooper, 2020-06-06:
This problem also affects the ethernet connection I have POE ESP32 from Olimex and updating cause them not to connect.
[loggs.txt](https://github.com/esphome/issues/files/4740986/loggs.txt)

Pulli0N, 2020-06-06:
I have 2 divice
First:  
esphome:
  name: meteostation
  platform: ESP8266
  board: nodemcuv2
dallas:
  - pin: D6
sensor:
  - platform: dallas
    address: 0xE80307979432F628
    name: "RoofTempOut"

After update on 1.14.4 work fine

But second:
esphome:
  name: aqua
  platform: ESP32
  board: esp32doit-devkit-v1
dallas:
 - pin: GPIO25
sensor:
 - platform: dallas
   address: 0xAC03099794586128
   name: "Aqua Temp"

After update send error in log
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 10.0.0.62: [Errno 111] Connection refused). Re-Trying in 1 seconds

But it exists on the network and OTA works
If I comment the dallas section everything else works fine

asadenton, 2020-06-07:
I am seeing this too. 1.14.4 works fine on most of my devices but the one that has a DHT and Dallas is failing if I include either of those in the config. I am getting the same error message as above :"Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)". The first entry says it is not related to the issue with DHTs but for me either of these components being present is causing the issue.

csmart, 2020-06-07:
I have the same problem, when dallas component is enabled version 1.14.3 works but version 1.14.4 is broken.

Git bisect seems to point to commit https://github.com/esphome/esphome/commit/072b2c445cd33a13d7029e9ce36dad9122fce162

```
[12:12 csmart ~/code/esphome ((072b2c445cd33a13...)|BISECTING)]$ git bisect bad
072b2c445cd33a13d7029e9ce36dad9122fce162 is the first bad commit
commit 072b2c445cd33a13d7029e9ce36dad9122fce162
Author: Otto Winter <otto@otto-winter.com>
Date:   Wed Dec 4 19:30:10 2019 +0100

    Add ESP8266 core v2.6.2 (#905)
    
    * Add ESP8266 core v2.6.2
    
    * Upstream ESP8266 Wifi fixes
    
    * Replace disable_interrupt with InterruptLock C++ class
    
    * Update code to use InterruptLock
    
    * Lint
    
    * Update dht.cpp
    
    * Improve InterruptLock docs, mark as ICACHE_RAM_ATTR
    
    * Fixes

 esphome/components/dallas/dallas_component.cpp     |  68 +++++-----
 esphome/components/dallas/esp_one_wire.cpp         |  26 ++--
 esphome/components/dht/dht.cpp                     | 141 ++++++++++++---------
 esphome/components/hx711/hx711.cpp                 |  31 ++---
 .../remote_transmitter_esp8266.cpp                 |  24 ++--
 esphome/components/uart/uart.cpp                   |  39 +++---
 esphome/components/wifi/wifi_component_esp8266.cpp |  59 ++++++++-
 esphome/core/helpers.cpp                           |  28 ++--
 esphome/core/helpers.h                             |  34 ++++-
 esphome/core/preferences.cpp                       |  22 ++--
 esphome/platformio_api.py                          |   1 +
 11 files changed, 291 insertions(+), 182 deletions(-)
```

csmart, 2020-06-07:
@OttoWinter indeed, reverting this commit solves the problem for me.

SebastianBar, 2020-06-07:
The same happened to me.

I'm running ESPHome on Docker. The `esphome/esphome:1.14.3` tag is broken, not being able to download resources from the internet.

I "solved" it by switching to the `esphome/esphome:dev` tag.

DoctorOctagonapus, 2020-06-07:
@SebastianBar Yeah I just tried to roll back as well. Says I'm not connected to the internet and won't compile. Starting to wish I'd saved a .bin file...

rostikk80, 2020-06-07:
same here - adding DHT platform creates a boot loop.

DaveTiff, 2020-06-07:
Same here wont run unless I rem out DHT sensor.

rasweh, 2020-06-07:
Same here. If I comment out the DHT sensor in the yaml configuration file then the ESP32 device boots fine - as soon as I try to enable the DHT sensor in the configuration the device goes into a boot loop with a guru meditation error.

PMaxx-Gaming, 2020-06-07:
Having the same issue here; hopefully its figured out soon, as my dallas sensor is integral to my device and I can't seem to find a way to rollback the update.

asadenton, 2020-06-08:
For those unable to roll back. I ran
pip install esphome==1.14.3
pip uninstall platformio
pip install platformio
rm -rf .esphome builddirformyyaml
Then it works fine.

DoctorOctagonapus, 2020-06-08:
@asadenton is there an equivalent set of steps for those of us running on HassOS? I tried restoring 1.14.3 from a snapshot but it failed with the "You are not connected to the internet" issue when I tried compiling.

mbo18, 2020-06-08:
Is there a workaround for HassOS users? 
On 1.14.4, can't use DHT and with 1.14.3 I can't compile. @glmnet or @OttoWinter please help

mario-tux, 2020-06-08:
Remove the DHT references in the yaml of the project; rebuild it and upload using USB (also using another computer with esphome-flasher). In this ways you can use your ESP even with esphome 1.14.4 but without H&T from DHT.

mbo18, 2020-06-08:
That's what I did, but I need my DHT :)

SebastianBar, 2020-06-08:
@mario-tux I wouldn't say that as a proper fix for the issue.

@DoctorOctagonapus I've never used HassOS, but I'm certain that it is a Docker supervisor. Can you set Docker configurations on it? If so, my previous message should help you.

mario-tux, 2020-06-08:
@SebastianBar Yes, it is just a workaround waiting for the official fix. :-)

Using HassOS and the ESPHome addons you can revert to a previous version of ESPHome. Uninstall the stable addon, install the `dev` one and use in the addon configuration the field `esphome_version`. It's a powerful tool and you can even use a build-on-the-fly branch from a generic fork on GitHub. In this case you should use something like:
```
{
  "esphome_version": "v1.14.3"
}
```
I'm not using it as I can survive without DHT sensors but I used it in the past.

mbo18, 2020-06-08:
Just tried that but ESPHome 1.14.3 contains Platform.io 4.0.3 which is obsolete
and does not compile. So ESPHome 1.14.3 is dead. I will wait for a fix

mario-tux, 2020-06-08:
@mbo18 this surprise me as I was thinking the all was self-contained in the addons model. I used this system to use a personal fork (with PR or fix) waiting for the official adoption. One could fork and revert the bad commitment to get a personal working version of ESPHome 1.14.4.

HugoVonHory, 2020-06-08:
I have switched to dev (1.15.0) as workaround and it works fo all my configs. Waiting for official fixed version.

larrelandin, 2020-06-09:
> I have switched to dev (1.15.0) as workaround and it works fo all my configs. Waiting for official fixed version.

Can confirm that this worked for me as well. I first tried reverting back to 1.14.3 but got errors when uploading. Added "esphome_version: dev" to my configuration and then it worked fine.

DoctorOctagonapus, 2020-06-09:
I'll have to have another go at switching to dev. I tried adding the esphome/esphome:dev tag to my instance within Portainer but it just timed out.

fhassine, 2020-06-12:
> pip install esphome==1.14.3

> For those unable to roll back. I ran
> pip install esphome==1.14.3
> pip uninstall platformio
> pip install platformio
> rm -rf .esphome builddirformyyaml
> Then it works fine.
Thanks for the hint... my pio got corrupt as well.. Do you run this in a container, or online? (i am presuming your installation is docker'ed?)

DaveTiff, 2020-06-12:
How can roll back to 1.14.3 using HASSIO on a PI

Many thanks

DoctorOctagonapus, 2020-06-12:
@DaveTiff If you have a snapshot from before you updated you can restore from that, however none of your configs will compile due to a broken Platform.IO dependency, which is fixed in 1.14.4.

If you install the ESPHome (dev) addon, you can start that up (obviously stop the main one first) and flash your borked device with the dev firmware. That's what I ended up doing to get my bad ESP32 back on.

DaveTiff, 2020-06-13:
> @DaveTiff If you have a snapshot from before you updated you can restore from that, however none of your configs will compile due to the broken Platform.IO dependency.
> 
> If you install the ESPHome (dev) addon, you can start that up (obviously stop the main one first) and flash your borked device with the dev firmware. That's what I ended up doing to get my bad ESP32 back on.

Hi, Thanks for the reply, I may have a snapshot taken 4 weeks ago, I will check and look at the DEV version.

Regards, Dave

rostikk80, 2020-06-13:
Is there any visibility as to when this might be fixed?

Haringstad, 2020-06-13:
Tried the dev version, and uncommented the DHT section. But it's still russian roulette..... Bootloop, freeze, same type of errors as already mentioned before in this tread. I truly hope this gets solved! I love my ESPHome enabled devices!

enriquefernandez, 2020-06-14:
I'm having the same issue with just the Dallas sensor (no DHT).
Removing the Dallas platform and sensor from the configuration completely solves the issue.

stigvi, 2020-06-14:
> Is there any visibility as to when this might be fixed? My DHT is supposed to look after the wine cellar...

I am also wondering about this. If the dev version solves this, why isn't a fix pushed out immediately?

TessyPowder, 2020-06-14:
My Node MCU ESP32 also gets stuck in a bootloop, if i add a dht11 sensor to the yaml configuration. After 10 failed boots it switches to safe mode.

230delphi, 2020-06-14:
I can confirm same behaviour with Dallas components. switched to Dev branch and it is working perfectly on 1.15

glmnet, 2020-06-14:
Ok, https://github.com/esphome/esphome/commit/ea652e35879e554a478422eb0832d737ca6cc26c is the commit in dev which fixes this issue introduced in 1.14.4

As you know it only affects ESP32 the components
 - DHT
 - Dallas 
 - hx711

thanks @csmart for pointing the issue.

As workaround I could only suggest to use dev version

mbo18, 2020-06-14:
\o/ This is it! I can confirm that my DHT is back online after changing those 2 lines! And I'm still on 1.14.4

mario-tux, 2020-06-14:
> Ok, [esphome/esphome@ea652e3](https://github.com/esphome/esphome/commit/ea652e35879e554a478422eb0832d737ca6cc26c) is the commit in dev which fixes this issue introduced in 1.14.4

I don't understand: this looks an old commit (already in 1.14.4) dated December 2019. Do you mean that this is the bad commitment and that the fix is revert it?

glmnet, 2020-06-14:
> I don't understand: this looks an old commit (already in 1.14.4) dated December 2019. Do you mean that this is the bad commitment and that the fix is revert it?

no, it's not in 1.14.4

mario-tux, 2020-06-14:
> no, it's not in 1.14.4

Ok, you are right: it's an old commit in a not merged branch.  But why people says that the current dev fix the problem: does the master include this commit?

glmnet, 2020-06-14:
branch master is 1.14.4 
branch dev (default) is dev

jaysalazar, 2020-06-18:
Wish I would have seen this sooner, updated and using Dallas sensor. Any updates on official fix?

jazlichy, 2020-06-20:
I had a Dallas one wire sensor working fine under 1.14.3. Upgraded to 1.14.4 & the ESP32 would no longer connect to to wifi. I manually wiped & reflashed over USB and still the same issue under 1.14.4. I assumed the ESP32 had bricked so flashed a new ESP32 and had the same error. I then commented all Dallas entries and the ESP32 returned to life. So now it works again but I have no Dallas temp sensors. 

Hope this is helpful diagnosing the issue. 

Still love ESPHome, bugs and all, amazing product.

Config with dallas entries commented out:

# Enable 1-wire DS18b20 sensors
#dallas:
#  - pin: GPIO15

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret ota_api_password

ota:
  password: !secret ota_api_password
  
# Enable BlueTooth scanning for the ESP32
esp32_ble_tracker:

sensor:
  # kitchen
  - platform: xiaomi_lywsdcgq
    mac_address: 4C:65:A8:DA:85:D3
    temperature:
      name: "XM85_D3 Temperature"
    humidity:
      name: "XM85_D3 Humidity"
    battery_level:
      name: "XM85_D3 Battery"
  
  # living room
  - platform: xiaomi_lywsdcgq
    mac_address: 58:2D:34:34:71:78
    temperature:
      name: "XM71_78 Temperature"
    humidity:
      name: "XM71_78 Humidity"
    battery_level:
      name: "XM71_78 Battery"

#  - platform: dallas
#    address: 0xCF02131DB904AA28
#    name: "Hot Water"
#  - platform: dallas
#    address: 0x6802131DB096AA28
#    name: "Lower Tank"

WoLpH, 2020-06-23:
I do hope a fix for this gets released soon as this can easily brick devices that are hard (or impossible) to reflash without OTA

tathamoddie, 2020-06-23:
> as this can easily brick devices that are hard (or impossible) to reflash without OTA

In my experience with this bug, that's not the case. The firmware crashes on boot up, causes a boot loop, and then after 10x rounds automatically enters safe mode. In safe mode, OTA updates are still functional. Logging is not.

WoLpH, 2020-06-23:
> In my experience with this bug, that's not the case. The firmware crashes on boot up, causes a boot loop, and then after 10x rounds automatically enters safe mode. In safe mode, OTA updates are [...]

Maybe you're lucky or I was unlucky because I had to flash over USB to get my ESP32 working again.

DoctorOctagonapus, 2020-06-23:
> Maybe you're lucky or I was unlucky because I had to flash over USB to get my ESP32 working again.

Same here. OTA was completely broken, wouldn't even attempt to flash.

glmnet, 2020-06-23:
Yes maybe this issue breaks OTA too. It’s quite a silly mistake actually handling concurrency in the code. The bug was spotted right away when introduced but Otto misspicked the fix when released 1.14.4 in a rush.

glmnet, 2020-06-24:
I just partially released v1.14.5 with just this fix, you need to go to the configuration tab in the ESPHome add on and add

````yaml
"esphome_version": "v1.14.5"
````

DaveTiff, 2020-06-24:
Thank you, now back working again.

Re Dave

Haringstad, 2020-06-24:
Using v1.14.5 makes it work again! Thank you!

Regards,

Jacco van Koll

jamestutton, 2020-06-24:
Any idea when 1.14.5 docker image will be released?

fribse, 2020-06-24:
I just tried it here with the 1.14.5 setting in the config, works like a charm!

nicop4, 2020-06-24:
I started using a DTH11 with esp32 a few days ago and the issue. Thanks to all of you for fixing it so quicly :)
However I cannot get the new version via pip (so not in Hassio, just a standard install on macos), is there a `pip` thing to do to have it ?

```
pip install --upgrade esphome
Requirement already up-to-date: esphome in /usr/local/lib/python3.7/site-packages (1.14.4)
```

If I ask for the specific new version, it seems not declared:
```
pip install --upgrade 'esphome==1.14.5'
ERROR: Could not find a version that satisfies the requirement esphome==1.14.5 (from versions: 1.10.1, 1.11.0b1, 1.11.0b2, 1.11.0b3, 1.11.0, 1.11.1, 1.11.2, 1.12.0b1, 1.12.0b2, 1.12.0b3, 1.12.0b4, 1.12.0, 1.12.1, 1.12.2, 1.13.0b1, 1.13.0b2, 1.13.0b3, 1.13.0b4, 1.13.0b5, 1.13.0b6, 1.13.0b7, 1.13.0, 1.13.1, 1.13.2, 1.13.3, 1.13.4, 1.13.5, 1.13.6, 1.14.0b1, 1.14.0b2, 1.14.0b3, 1.14.0b4, 1.14.0b5, 1.14.0, 1.14.1, 1.14.2, 1.14.3, 1.14.4)
ERROR: No matching distribution found for esphome==1.14.5
```

napieraj, 2020-06-24:
@nicop4 `pip3 install https://github.com/esphome/esphome/archive/v1.14.5.zip`

@jamestutton unfortunately, we need Otto to update the Docker repositories, as he is the only one with access to the account

MimbaMonkeyHouse, 2020-06-29:
> ```yaml
> "esphome_version": "v1.14.5"
> ```

Thanks for this. Put an end to a day of troubleshooting. Even went through a full rewiring/soldering as i thought that may be the issue.

[1 more reply]

closed 2020-07-01 by glmnet