esphome/issues #1278: fan and tuya_fan components lack support for modules having 4 speed states

m1k3n34l, 2020-06-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Windows 10 VirtualBox host
**ESP (ESP32/ESP8266, Board/Sonoff):**

Treatlife model DS03 Ceiling Fan Dimmer Switch (Tuya MCU)
**Affected component:**

**Description of problem:**
Hardware supports 4 speed states 0x0,1,2,3
tuya_fan and fan components support only 3 speed settings 0x0,1,2

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
# -----------------------------------------------------------------------------
# TUYA SETUP FOR TREATLIFE DS03 CEILING FAN DIMMERS
# -----------------------------------------------------------------------------

substitutions:
  devicename: test_fan_treatlife
  lightname: Test Controller
  ssid: !secret wifi-ssid
  ssidpass: !secret wifi-password

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: $ssid
  password: $ssidpass

logger:
  baud_rate: 0
api:

ota:

web_server:
  port: 80
  
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 115200

tuya:

sensor:
  - platform: wifi_signal
    name: $lightname WiFi Signal
    update_interval: 60s

  - platform: uptime
    name: $lightname Uptime

light:
  - platform: "tuya"
    name: $lightname Light
    dimmer_datapoint: 10
    switch_datapoint: 9
    min_value: 100
    max_value: 1000

fan:
  - platform: "tuya"
    name: $lightname Fan
    switch_datapoint: 1
    speed_datapoint: 3

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[18:08:42][C][logger:175]: Logger:
[18:08:42][C][logger:176]:   Level: DEBUG
[18:08:42][C][logger:177]:   Log Baud Rate: 0
[18:08:42][C][logger:178]:   Hardware UART: UART0
[18:08:42][C][light:177]: Light 'Test Controller Light'
[18:08:42][C][light:179]:   Default Transition Length: 0.0s
[18:08:42][C][light:180]:   Gamma Correct: 1.00
[18:08:42][C][tuya.light:027]: Tuya Dimmer:
[18:08:42][C][tuya.light:029]:    Dimmer has datapoint ID 10
[18:08:42][C][tuya.light:031]:    Switch has datapoint ID 9
[18:08:42][C][tuya.fan:048]: Tuya Fan:
[18:08:42][C][tuya.fan:050]:   Speed has datapoint ID 3
[18:08:42][C][tuya.fan:052]:   Switch has datapoint ID 1
[18:08:42][C][web_server:131]: Web Server:
[18:08:42][C][web_server:132]:   Address: test_fan_treatlife.local:80
[18:08:42][C][ota:029]: Over-The-Air Updates:
[18:08:42][C][ota:030]:   Address: test_fan_treatlife.local:8266
[18:08:42][C][api:095]: API Server:
[18:08:42][C][api:096]:   Address: test_fan_treatlife.local:6053
[18:08:42][C][wifi_signal.sensor:009]: WiFi Signal 'Test Controller WiFi Signal'
[18:08:42][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[18:08:42][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[18:08:42][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[18:08:42][C][tuya:023]: Tuya:
[18:08:42][C][tuya:032]:   Datapoint 9: switch (value: OFF)
[18:08:42][C][tuya:034]:   Datapoint 105: int value (value: 100)
[18:08:42][C][tuya:034]:   Datapoint 10: int value (value: 99)
[18:08:42][C][tuya:032]:   Datapoint 1: switch (value: ON)
[18:08:42][C][tuya:036]:   Datapoint 3: enum (value: 2)
[18:08:42][C][tuya:036]:   Datapoint 106: enum (value: 0)
[18:08:42][C][tuya:034]:   Datapoint 101: int value (value: 0)
[18:08:42][C][tuya:034]:   Datapoint 103: int value (value: 0)
[18:08:42][C][tuya:046]:   Product: '{"p":"5oja7jtexzcvv9d1","v":"3.3.10","m":0}'
[18:08:42][E][uart:049]:   Invalid baud_rate: Integration requested baud_rate 9600 but you have 115200!
[18:08:42][C][tuya.fan:023]: Speed has invalid value 3
[18:08:42][D][tuya.fan:024]: MCU reported speed of: 3
[18:08:42][D][tuya.fan:064]: Setting switch: ON
[18:08:42][D][tuya.fan:084]: Setting speed: 2
[18:08:42][D][tuya.fan:024]: MCU reported speed of: 2
[18:08:42][D][tuya.fan:064]: Setting switch: ON
[18:08:42][D][tuya.fan:084]: Setting speed: 2

```

**Additional information and things you've tried:**

Interesting baud rate warning as well, though the Tuya MCU will not respond when UART is set to 9600 baud.  Unrelated to this issue; kept in log output for completeness.

brandond, 2020-06-06:
Home Assistant only has 3 speed states (hi/med/low, off) so there's no point having more in esphome.

Sanfam, 2020-08-25:
Some food for thought: what about enabling ESPHome to custom map each of the three available speeds to one of the switch's speed data values.  This would provide an adequate stopgap for switches with `N` speeds while the home assistant community hashes out a long term direction. 

For example, a fan I have attached to this is so slow at the    ###DS0 switch speed 1 that it barely turns, rendering 2's leisurely spin as my effective "low", 3's relaxed pace as my effective medium, and 4's full-bore gust as high.  Were I to flash it with ESPHome in its current state, I'd only have two usable speeds, neither of them terribly good at moving air.  

In the case of the Tuya fan controllers supporting 5-7 speeds, this would enable users to define a comfortable and usable range from low to high.

cprussin, 2020-10-26:
I am using the TreatLife DS03 wall fan controller & dimmer.  I learned today that esphome actually prevents accessing the top fan speed, even when manually using the wall switch--if I increase to the top speed on the wall switch with esphome installed, the device immediately reverts back to the third speed. That's pretty unfortunate--while I'd love to be able to access all speeds from HA I think at least being able to somehow access them manually would be a huge improvement.

cprussin, 2020-10-26:
Alternatively as @Sanfam mentioned being able to configure what the 3 speeds map to would be great

skitzo2000, 2020-10-29:
I'm with @cprussin on the request for an option to assign, or even just an example/hack to use to start on speed 2-4 rather than 1-3.  On my fan level 1 is pretty much useless.

Sanfam, 2020-10-30:
It looks like my take on this might be on pause while I replace the switch.  I was starting to map that out in code, but it turns out this dimmer doesn't react well to powering a vacuum (surprise wiring discovery!)

Based on my weeks of use, it really seems like allowing a map of speed values using explicit data value assignments is the safe decision that permits current and future models to work.   More Tuya switches appear to be on the horizon, at least one using potentially 7 states and one with as few as 2 (off/low/high).  Even with a 2-speed switch you can redundantly `0x0` or `0x1` to medium and retain `0x0` for low and `0x1` for high.

Changes would need to be in the `tuya_fan.h` init file (add speed variable), `tuya_fan.cpp` to refactor maps to variable sources, and `tuya/fan/__init__.py` to handle detection of yaml arguments

pcartwright81, 2020-11-23:
I was able to write the code out, but there is still work to do to the docs to open the pr.  I took it from a simple change of the variables to what was described above.

https://github.com/pcartwright81/esphome/commit/b002897abe0df49a7c960cec3935425075246c70

ssieb, 2021-01-28:
HA now has support for more speeds, so support can be added in esphome.

mjoshd, 2021-05-05:
I just got one of these TreatLife DS03 devices and noticed that ESPHome's [Speed Fan](https://esphome.io/components/fan/speed.html#configuration-variables) now (as of 1.17.0) has the ability to define the number of speeds however it seems that it is still not compatible with the `Tuya Fan` component.

```
 [speed_count] is an invalid option for [fan.tuya]. Did you mean [speed_datapoint], [speed_command_topic]?
  speed_count: 4 [source /config/esphome/treatlife_ds03_fan_light.yaml:194]
```

Does anyone know if it's actually possible to do now, and I'm just doing it wrong, or if there's still something that needs to be added in order to make it work?

0x0a11c0de, 2021-05-05:
Support for more speeds was just [merged](https://github.com/esphome/esphome/pull/1654). I don't know if it will make it in 1.18.0, though.

mjoshd, 2021-05-05:
@0x0a11c0de  Thanks for the heads up!

I installed the 1.18.0 dev version to test and it compiles, installs, and works perfectly on this device 🎉

0x0a11c0de, 2021-05-05:
@mjoshd Thanks for confirming. It has been working great for me as well.

Sounds like this issue can be closed.

closed 2021-05-05 by m1k3n34l