esphome/issues #1281: use GPIO2 for ACD on esp32 wroom32

jasonblewis, 2020-06-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Debian, building from command line:

`esphome  iotuz.yaml run  --upload-port=iotuz.local`

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32 wroom 32 on an [IoTuz](https://github.com/CCHS-Melbourne/iotuz-esp32-hardware/wiki) board
**Affected component:**

[adc](https://esphome.io/components/sensor/adc.html)

**Description of problem:**
if you set it to use pin 2, esphome produces the following error:

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: iotuz
  platform: ESP32
  board: esp32dev

sensor:
  - platform: adc
    pin: 2
    name: "Battery Level"
    update_interval: 60s
    attenuation: 11db
```

**Logs (if applicable):**

```
jason@sage:~/ownCloud/projects/esphome/config$ esphome  iotuz.yaml run  --upload-port=iotuz.local
INFO Reading configuration iotuz.yaml...
Failed config

sensor.adc: [source iotuz.yaml:70]
  platform: adc
  
  ESP32: Only pins 32 though 39 support ADC.
  pin: 2 [source iotuz.yaml:71]
  name: Battery Level
  update_interval: 60s
  attenuation: 11db

```

according to this page, gpio2 can be used for ADC if I understand correctly:

https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

on adc_2

JSchuetzke, 2020-06-11:
I just managed to get my temperature sensor to work so I'm not an expert but for me I had to specify the pin with "GPIOx"

So in your case the solution could be:
```yaml
sensor:
  - platform: adc
    pin: GPIO2
```

hydrocontica, 2020-07-05:
I'm having a similar issue - when I try to use the ADC component with GPIO2 on my ESP32, ESPHome is insisting that "Only pins 32 though 39 support ADC". Yet that is only true for ADC1. ADC2 is also documented as being available on GPIO pins 0, 2, 4, 12-15, and 25-27.

One issue is that the ESP32 can't handle wifi and ADC2 simultaneously, so using any of those pins could prove problematic in many use cases - however, I'm trying to set up a ph sensor on an AI Thinker ESP32-Cam board with wifi and the sensor only being used at specified intervals that do not overlap, and pins 32-39 on such a board are all used internally for the camera module. For ADC, my only option is to use one of the ADC1 pins, yet ESPHome won't let me because it seems to only recognise ADC2 as existing on the ESP32.

closed 2020-11-17 by stale[bot]

gogenzuller, 2021-08-28:
1.5 years have passed, and the problem has not been resolved. Very sad.

CostaRich, 2021-10-09:
Can confirm this is still a problem.  There's no workaround?