esphome/issues #1283: Regression on 1.14.4 with a specific version of Xiaomi CGG1 sensor

mario-tux, 2020-06-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hass.io 110.5 / Esphome 1.14.4

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32

**Affected component:** xiaomi_cgg1

**Description of problem:**
There is a regression in esphome 1.14.4 about the support of one-of-many variants of the CGG1 H&T sensor. I introduced it in https://github.com/esphome/esphome/pull/930 but the last rewrite by @ahpohl is missing it. I got other 5 devices (I guess different variants) that worked and work with current version.

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: xiaomi_cgg1
    mac_address: 58:2D:34:10:3A:67
    temperature:
      name: "Bathroom Temperature"
      filters:
        - lambda: |-
            if (x >= -5 && x <= 60 ) return x;
            else return {};
        - filter_out: nan
    humidity:
      name: "Bathroom Humidity"
      filters:
        - lambda: |-
            if (x >= 10 && x <= 100 ) return x;
            else return {};
        - filter_out: nan
    battery_level:
      name: "Bathroom Battery"
```

**Logs (if applicable):**

This is the unsupported variant:
```
[08:45:34][VV][esp32_ble_tracker:282]: Parse Result:
[08:45:34][VV][esp32_ble_tracker:299]:   Address: 58:2D:34:10:3A:67 (PUBLIC)
[08:45:34][VV][esp32_ble_tracker:301]:   RSSI: -81
[08:45:34][VV][esp32_ble_tracker:302]:   Name: 'ClearGrass Temp & RH'
[08:45:34][VV][esp32_ble_tracker:310]:   Ad Flag: 6
[08:45:34][VV][esp32_ble_tracker:319]:   Service data:
[08:45:34][VV][esp32_ble_tracker:320]:     UUID: FE:FE95
[08:45:34][VV][esp32_ble_tracker:321]:     Data: 50.20.47.03.16.67.3A.10.34.2D.58.0D.10.04.E1.00.87.02.0A.10.01.15 (22)
[08:45:34][VV][esp32_ble_tracker:324]: Adv data: 02.01.06.19.16.95.FE.50.20.47.03.16.67.3A.10.34.2D.58.0D.10.04.E1.00.87.02.0A.10.01.15.15.09.43.6C.65
.61.72.47.72.61.73.73.20.54.65.6D.70.20.26.20.52.48 (51)
```

This is from one of the supported one:
```
```

**Additional information and things you've tried:**

The main difference on the frame I raised in https://github.com/esphome/esphome/pull/930 was the difference on `raw[1]` (0x30 vs. 0x20) but now, in the new code, I can't spot the way such byte is considered. May be the frame is discharged for other checks. @ahpohl  could you help?

mario-tux, 2020-06-10:
@ahpohl, we should continue the discussion here as this is the only open issue report. Previous messages are present at https://github.com/esphome/issues/issues/1282#issuecomment-641789284

> The relevant code in `xiaomi_ble.cpp` is this:
> 
> ```
>   } else if ((raw[2] == 0x47) && (raw[3] == 0x03)) {  // round body, e-ink display
>     result.type = XiaomiParseResult::TYPE_CGG1;
>     result.name = "CGG1";
> ```
> 
> Previously raw bytes 1-3 were checked for the type, but now only bytes raw[2] and raw[3] are checked. As `raw[2] == 0x47` for both CGG1 variants, it should work regardless.
> 
> Could you please confirm that your CGG1 variant is not working with the current dev branch. If this is really the case, I'll have to investigate further.

I tested the stable 1.14.4 and it doesn't work (as before my old PR). I can't spot in the dev branch any related new commit. Are they?

Could the issue be present in the structure of the following bytes (at the moment I don't understand the underlying logic).

ahpohl, 2020-06-10:
The packet is rejected here:
```
result.has_data = (raw[0] & 0x40) ? true : false;
```
As `raw[0] == 0x20` for this variant, the check evaluates to `false`. And later
```
  if (!result.has_data) {
    ESP_LOGVV(TAG, "parse_xiaomi_header(): service data has no DATA flag.");
    return {};
  }
```
One solution would be to remove the check for the data flag from the code.
However, this issue raises the question whether
a) the check for the data flag (& 0x40) is wrong for all types
b) the CGG1 variant behaves odd.
How does the packet data frame for the other CGG1 variant look like in the verbose log?

mario-tux, 2020-06-10:
> The packet is rejected here:
> 
> ```
> result.has_data = (raw[0] & 0x40) ? true : false;
> ```
> 
> As `raw[0] == 0x20` for this variant, the check evaluates to `false`. And later

I'm a bit lost (maybe I'm looking at the wrong reference) but in both my data packets the `raw[0]` is `0x50`.

ahpohl, 2020-06-10:
You are right, I just dug out my notes and raw[0] is indeed 0x50 here.

mario-tux, 2020-06-10:
I just installed the esphome-dev addons on Hassio and, I don't understand how, but it started to work again. A mess in my addons structure or something wrong with the official stable package of 1.14.4. 

Sorry for the noise. I'm going to investigate trying to install again the stable addon.

mario-tux, 2020-06-10:
There is something wrong in the last 1.14.4 addon package for HassOS. It reports 1.14.4 in the web interface but it presents the old web dashboard and my CGG1 variant doesn't work...

@OttoWinter we miss you!

closed 2020-06-10 by mario-tux