esphome/issues #1291: Substitutions not working with ints and floats

brahmafear, 2020-06-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esphome running via HassIO addon on HomeAssistant 0.111.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8266EX

**Affected component:**

Substituition
Don't see any docs for this.

**Description of problem:**
Cannot be used with integer or float values.  

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
substitutions:
  transition_default: '2000ms'
  min_pwm: '0.0'
  max_pwm: '0.99'

output:
  - platform: esp8266_pwm
    id: o_0
    # inverted: True
    min_power: $min_pwm
    max_power: $max_pwm
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Reading configuration /config/esphome/fairyjar.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 566, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 544, in run_esphome
    config = read_config()
  File "/opt/esphome/esphome/config.py", line 808, in read_config
    res = load_config()
  File "/opt/esphome/esphome/config.py", line 668, in load_config
    return _load_config()
  File "/opt/esphome/esphome/config.py", line 656, in _load_config
    result = validate_config(config)
  File "/opt/esphome/esphome/config.py", line 561, in validate_config
    validated = comp.config_schema(input_conf)
  File "/opt/esphome/esphome/voluptuous_schema.py", line 32, in __call__
    res = super(_Schema, self).__call__(data)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/opt/esphome/esphome/voluptuous_schema.py", line 131, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/opt/esphome/esphome/config_validation.py", line 837, in percentage
    value = possibly_negative_percentage(value)
  File "/opt/esphome/esphome/config_validation.py", line 845, in possibly_negative_percentage
    if value > 1:
TypeError: '>' not supported between instances of 'ESPStr' and 'int'
```

**Additional information and things you've tried:**
Tried removing quotes from sub definition.
Tried to convert to float or int at usage location.

brahmafear, 2020-06-13:
Further testing -- this might be a specific problem only with the min_power and max_power statements of esp8266_pwm.

brahmafear, 2020-06-15:
Workaround found by setting substitution to "99%" instead of "0.99" which works for the esp8266pwm component even though not documented as such.

closed 2020-06-19 by glmnet

bcwhite0, 2020-07-25:
It seems that substitutions are only allowed to be strings but if I want to use one in a place where only a float is allowed...

Ex:
sensor > filters > calibrate_linear
   - 0.0 -> 0.0
   - 1.0 -> ${sensor_scale}

... then the substitution will fail because it's not the expected type.  Is there any way to make a "float" substitution or convert it at the point where it is used?

ssieb, 2020-07-25:
It does work, so you're doing something else wrong.  I'll contact you on Discord.

bcwhite0, 2020-07-28:
In trying to narrow down the problem, I found a solution that works.  I can't actually say what I changed because it looks exactly the same as my memory of what didn't work and, unfortunately, I didn't keep the non-working files for comparison.  It even failed to work many times while trying to create that minimal test-case but I still can't reproduce it now.  So, uhhh...  never mind?

brahmafear, 2020-07-28:
> In trying to narrow down the problem, I found a solution that works. I can't actually say what I changed because it looks exactly the same as my memory of what didn't work and, unfortunately, I [...]

Can you post your working config For future reference?

ssieb, 2020-07-28:
There's not really anything to show.  The substitutions just work as you would expect.
```yaml
substitutions:
  scale_max: "5"

sensor:
[...]
    filters:
      - calibrate_linear:
          - 0.0 -> 0.0
          - 40.0 -> ${scale_max}
```

bcwhite0, 2020-07-28:
Yup.  That's pretty much it.  I must have had a bad indent or character somewhere.