esphome/issues #1294: SH1106 128x64 I2C Display - garbage at 2 first columns

u17194907425, 2020-06-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 DevKit v4

**Affected component:**

https://esphome.io/components/display/ssd1306.html

**Description of problem:**
I observe strange artifacts on the first two rows of the display. And it seems that the column 0 and the column 1 are not accessible for writing. If I draw a line from coordinates 0;0 it is started with the 3rd pixel from the left. I can't seem to draw anything beyond the column 125 even though I am expecting 126 and 127 to be available on my 128x64 screen.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  location: esp32_2_4
  upper_location: Esp32_2_4

esphome:
  name: esp32_2_4
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "*******"
  password: "*********"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32 ${upper_location} Fallback Hotspot"
    password: "*********"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: GPIO21
  scl: GPIO22
  
font:
  - file: "fonts/Consolas.ttf"
    id: my_font
    size: 16

################################################################################

sensor:
  #-----------------------------------------------------------------------------
  # Wi-Fi Signal Sensor
  - platform: wifi_signal
    id: wifi_sgnl
    name: "${upper_location} WiFi Signal Sensor"
    update_interval: 5s
    accuracy_decimals: 0
    filters:
      - sliding_window_moving_average:
          window_size: 60
          send_every: 60
          send_first_at: 2

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    #reset_pin: D0
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(my_font), "Hello World!");
      it.printf(0, 26, id(my_font), "Wi-Fi: %.1f", id(wifi_sgnl).state);
      

      it.line(1, 0, 124, 0);
      it.line(1, 63, 124, 63);
      it.line(0, 1, 0, 62);
      it.line(125, 1, 125, 62);
      
      it.line(126, 1, 126, 62); // I DO NOT SEE THIS LINE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I was trying to draw lines at the edges of the screen and diagonally. It seems that the available drawing area is 0;0 to 125;63

[image]

Troon, 2020-06-29:
I have exactly the same issue with what looks to be the same screen, set up thus:

`
esphome:
  platform: ESP8266
  board: d1_mini

i2c:
  sda: D2
  scl: D1

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    address: 0x3C
`

Screen works just fine if I don't try to write beyond 125px across: if I do, it puts junk into the LH columns.

kbx81, 2020-07-02:
@u17194907425 @Troon I do not have one of these specific displays, but I've been working on a fixes to this component. I would encourage you to try it and see if the behavior is resolved or has otherwise changed. Please reply and let me know -- if it's not fixed, I can try tweaking some other things during initialization and/or display writes to see if it helps.

Troon, 2020-07-04:
I'm happy to test but don't have much Github experience. I've installed
esphome into a python virtualenv, currently on 1.14.5 — how do I best get
your changes into my installation?

glmnet, 2020-07-05:
I can reproduce something similar here, but I fix it by changing to ``SSD1306 128x64``
can you post a picture how it looks if you use 

```
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
```

u17194907425, 2020-07-06:
> I can reproduce something similar here, but I fix it by changing to `SSD1306 128x64`
> can you post a picture how it looks if you use
> 
> ```
> display:
>   - platform: ssd1306_i2c
>     model: "SSD1306 128x64"
> ```

Wow... that did the trick to me. My mistake was that I blindly trusted to the data provided by the module seller. It was stated that the module is based on SH1106. So I haven't even thought of trying `SSD1306`.

Now It works as expected. No junk pixels. and the entire screen area is available.

Thanks again @glmnet !

Troon, 2020-07-06:
Opposite for me: I bought my screen not even aware of the two versions, but
got no output with the SSD1306 setting. Only when I changed to SH1106 could
I get any output at all.

glmnet, 2020-07-06:
Can you upload a picture?

closed 2020-12-05 by stale[bot]