esphome/issues #1295: Constant device reboots

picto3000, 2020-06-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome Docker - DEV
Home Assistant Core Docker - 0.111.2
Lubuntu - 20.04

**ESP (ESP32/ESP8266, Board/Sonoff):**

4x Sonoff S26 - ESP8266
1x Sonoff Mini - ESP8266
4x Kogan KASPEMHUSBA - ESP8266

**Affected component:**

wifi - https://esphome.io/components/wifi.html
api - https://esphome.io/components/api.html

**Description of problem:**
All my devices are consistently disconnecting from Home Assistant API, and as a result randomly reboot every 5 to 40 minutes.
- HA availability to flick between available/unavailable
- Uptime counters to reset
- Connected devices to switch off/on. e.g. lights flickering, appliances momentarily loosing power

I feel I'm lost in the Bermuda Triangle... I don't know if this issue is with 1. ESPHome, 2. Home Assistant or 3. my home wifi. I've systematically modified all options across all three systems with no resolution. 

1. First thought: It's an ESPHome issue related to case #877 - https://github.com/esphome/issues/issues/877#issuecomment-631834726
2. Then, I moved onto Home Assistant with case #36011 - https://github.com/home-assistant/core/issues/36011
3. At this point I discovered the devices weren't just disconnecting from the API but were also rebooting. The disconnects usually didn't mean all that much, but the reboots where causing flickering lights and reset up time timers etc.
4. I moved onto house wifi, checking for neighbour wifi conflicts, channel changes, 20Mhz only etc. all with no change. Other devices (e.g. Sonos) connected to my 2.4Ghz wifi have no issues at all. 

Now I've moved back to thinking it's an ESPHome issue, somewhere between device wifi and Home Assistant. I'd love to get more wifi logging info from ESPHome as to why there's a disconnect. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: $hostname
  comment: $comment
  platform: ESP8266
  board: esp01_1m
  # on_loop:
  #   then:
  #     - lambda: delay(5);

#CUSTOMISATIONS
substitutions:
  hostname: "sonoff_s26_01" #Internal designated name
  comment: "Bathroom LED" #Usually what it's currently used for
  device: "sonoff_s26_01" #What device is connected/controlled?
  icon: "mdi:power-socket-au" #MDI device icon e.g. mdi:power-socket-au via https://materialdesignicons.com
  restore_mode: "restore_default_off" #Power auto on with reboot? e.g. restore_default_on or ALWAYS_OFF
  ip: "192.168.1.80" #Static IP address
  old_ip: "192.168.1.80" #used when swaping static IPs  
 
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pwd
  power_save_mode: none
#  fast_connect: on
#  use_address: "$old_ip" #sonoff_04.local
  reboot_timeout: 30min
  manual_ip:
    static_ip: "$ip"
    gateway: !secret gateway
    subnet: !secret subnet

api:
ota:
logger:
  level: VERBOSE

[... 174 lines cut ...]
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
INFO Successfully connected to 192.168.1.80
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
INFO Successfully connected to 192.168.1.80
[04:11:20][V][sensor:013]: 'Uptime sonoff_s26_01': Received new state 512.284973
[04:11:25][V][api.connection:688]: Error: Disconnecting Home Assistant 0.111.2 (192.168.1.100)
[04:11:25][D][api:067]: Disconnecting Home Assistant 0.111.2 (192.168.1.100)
[04:11:25][V][api.connection:583]: Hello from client: 'Home Assistant 0.111.2 (192.168.1.100)'
[04:11:25][D][api.connection:599]: Client 'Home Assistant 0.111.2 (192.168.1.100)' connected successfully!
[04:11:32][D][text_sensor:015]: 'sonoff_s26_01 Uptime (formatted)': Sending state '0d 00h 08m 32s'
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.80:6053 (192.168.1.80)
INFO Successfully connected to 192.168.1.80
[04:11:48][V][api.connection:583]: Hello from client: 'Home Assistant 0.111.2 (192.168.1.100)'
[04:11:48][D][api.connection:599]: Client 'Home Assistant 0.111.2 (192.168.1.100)' connected successfully!
[04:12:09][V][sensor:013]: 'Uptime sonoff_s26_01': Received new state 30.309999
[04:12:09][D][sensor:092]: 'Uptime sonoff_s26_01': Sending state 30.31000 s with 0 decimals of accuracy
[04:12:10][D][text_sensor:015]: 'sonoff_s26_01 Uptime (formatted)': Sending state '0d 00h 00m 30s'
[04:13:09][V][sensor:013]: 'Uptime sonoff_s26_01': Received new state 90.309998
[04:13:09][D][sensor:092]: 'Uptime sonoff_s26_01': Sending state 90.31000 s with 0 decimals of accuracy
[04:13:10][D][text_sensor:015]: 'sonoff_s26_01 Uptime (formatted)': Sending state '0d 00h 01m 30s'
```

**Additional information and things you've tried:**

KYDronePilot, 2020-07-04:
I had pretty much the exact same problem, also running HA from a Docker container. It seemed to coincide with a HA upgrade, but downgrading didn't fix it. Also tried switching from the HA API to MQTT, but that didn't work either.

Rebooting the physical HA Docker host machine and affected ESP devices ended up fixing the issue. I'm guessing the network on the host had somehow gotten messed up during the upgrade, because other integrations also started having issues.

patmann03, 2020-07-07:
I've had similar issues.  I recently updated my firmware to 1.14.5 and reflashed the esp8266's.  I still have the occasional disconnect where the esp shows as unavailable for 1 second in the HA history.  I did move all the esp's to a different vlan / wireless SSID that is only broadcast on 2.4ghz thinking that may help, but it did not.  Prior to updating sometimes the esp would go unavailable for 10 minutes before reconnecting and sometimes I'd need to pull the plug to reboot it to get it to reconnect.  I haven't had this since updating.

I'll enable logging and see if I'm seeing the same issues you are, but I definitely still am having issues staying connected via the API.

picto3000, 2020-07-07:
I swapped out my Asus router with a full Ubuiti setup, and at the same time discovered a duplicate device in Home Assistant. I've been stable for 2-3 weeks and can't confirm which one solved my issue.

I'll close this off. In general WiFi just seems super flakey on these chips and I know there isn't much ESPhome can do about it.

closed 2020-07-07 by picto3000

mikalauskas, 2020-07-07:
I have no problem with tasmota firmware on these chips, while esphome hangs.

messier433, 2020-07-14:
Could we please reopen this topic? It was closed a bit too soon imo. I recently upgraded a couple of my ESP8266 nodes with the latest version of ESPHome and ~50% now show regular reboots (~2x per hour). All these nodes were working fine with a ~1 year old ESPHome version. For now I will simply go back to that old version, but its clear there are now new bugs in the ESPHome firmware and something could be done.

glmnet, 2020-07-15:
@messier433 no. But feel free to open a new one and provide all required information. Also review current issues labeled with communication issue. There are many. 
I have more than 20 at home and they all run quite well. All version v1.14 or newer