esphome/issues #1296: 7.5Inch E-Paper V2

Willi24c, 2020-06-14:
I bought Waveshare 7.5inch e-paper display. Attached to Lolin32 (or Waveshare ESP32 Driver Board) i can not get any content to the display. My investigation shows, that my display is Version2 which has different resolution. In your waveshare_epaper display platform the 7.5inch display is listed, but most likely it refers to the "old" 640x384 display. The V2 has 800x480px and does not work with your  platform. I was trying to buy old 640x384px display but it looks like, that it is not available any more.

My request: 
1.) Please indicate on"https://esphome.io/components/display/waveshare_epaper.html" that  your platform supports the 7.5in 640x384px version only.
2.) Please try to support the 7.5inch V2 version (800x480px) - Waveshare 13504 7.5inch e-Paper HAT.

For your reference:
1.) Before i was successfully using the 4.2inch Display with your platform (using ESP32).
2.) It looks like that the original Maker of the Waveshare display is Dalian and the 7.5inch 640x384px display (GDEW075T8) is also not available any more.

Thank you very much in advance for any help with this issue.
Regards
 

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hassio on Raspberry 3b+

**ESP (ESP32/ESP8266, Board/Sonoff):**
Lolin32  or e-Paper ESP32 Driver Board

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

sredfern, 2020-06-23:
Hey @Willi24c 

It's not very difficult to add support for new Waveshare modules. I added support for the new 5.83 model without much code change.
You need to figure out what controller version is in the glass and then copy the most similar initialize commands. You'll also need to grab the [data sheet](https://www.waveshare.com/w/upload/6/60/7.5inch_e-Paper_V2_Specification.pdf) and find the width and height hex codes and put them into the initialize commands. 

Here is my PR:
https://github.com/esphome/esphome/commit/7fa5cab8e368bdfacb07b40bb826bf19c8f8cf46

Willi24c, 2020-06-23:
Hi sredfern
first of all, thank you for your comment / informations!

I was trying to follow your link(s) but I can not find:
esphome/components/waveshare_epaper/waveshare_epaper.cpp 
esphome/components/waveshare_epaper/display.py

Using Home Assistant File Editor I can not see this files.
Actually I am using Raspberry Pi 3B+ with Hass OS 4.10

Please let me know, where to find this files. Modifying the
parameters should not be big issue.

Thank you in advance for helping the hardware guy
Regards

sredfern, 2020-06-24:
Hey @Willi24c,

I'd follow this guide here: https://esphome.io/guides/contributing.html 

I normally develop on a second linux box in my house. You could use an old raspberry pi to make the changes and then deploy them to the esp32 using the command line tools of esphome. Once you've got the local version you just need to cd into the display folder and follow the links from there. You can see the files from my PR. 

Once you publish your PR and the documentation it normally takes 1-2 months for the PR to be accepted into the dev branch. You can use this on your HA instance by choosing the dev branch of esphome in the supervisor.

Willi24c, 2020-06-24:
Hi sredfern,
I think that your methode is beyond my capabilities (or knowledge) at this time. I am interested user and not developer or software guru. What I understand is, that adopting your methode, I need to use second computer (Raspi or NUC or..). Install Linux version of Home Assistant and install ESPHome on that. Than I can modify the before mentioned files (waveshare_epaper.cpp and waveshare_epaper/display.py) to my needs and finally create with the installed ESPHome the necessary binary file for the ESP32 with7.5inch V2 epaper. Is that correct understanding? 

Or is it necessary to publish something and wait for acceptance and then I can use "normal methode with my existing Raspi"? Is there any guidance or manual for "normal user" or is the whole system only for nerds or gurus?

Sorry for eventually stupid questions or statements, but it seems to be that I need some kind of guidance how to proceed in case of non-standard (not listed in ESPHome) devices.
Regards

sredfern, 2020-06-24:
Hey @Willi24c 

Esphome is an open source project that thrives on beginners learning how to contribute to a real open source project! There is no requirement of being a guru or expert. Sadly, there is no paid team who develops modules support for interested parties. I would say, unless someone has the hardware there is very little interest in expanding the module scope. 

Yes, you would need a second computer to develop the changes (and test them - even more important). You would want just normal raspbian linux not hassio. Hassio is more of a consumption tool over a creation tool in this specific case. 

However, I understand that not everyone is keen to start developing. My suggestion would be to either find one of the older 7.5in displays or my commit in the dev branch supports the 5.83in units.

Willi24c, 2020-06-24:
Hi sredfern,
thank you so much, spending time for beginners / users. 
As mentioned im my first post, I could not find 7.5inch V1 black/white display in my country. Also the chinese suppliers offer V2 version only. 

Taking your example (5.83inch e-paper) what will happen next? When can I expect related entry at: https://esphome.io/components/display/waveshare_epaper.html? 

I mentioned earlier that there should be a remark in the list that the 7.5inch support is in this moment only for the 7.5inch V1(640x384px) Display. Who is taking care and how to trigger such an entry? 

Fortunately I could find a 7.5inch two color display (640x384px) which is "officially" not supported but works fine (slightly reduced sharpness) for my project. So I can wait for future development. I learned in the past that it is better to finalize first issue first and then take care next issue. 

Anyway, thank you very much for your help and understanding.

PaulAntonDeen, 2020-07-16:
https://github.com/esphome/feature-requests/issues/753

Use the Dev version of ESPhome to get working support of Waveshare 7.5inch V2

Willi24c, 2020-08-06:
Hi PaulAntonDeen,
thank you for your comment. This was solving my issue.

closed 2020-08-06 by Willi24c

Thoky81, 2020-08-07:
Hi. Sorry opening the issue again, but I would like to ask you how did you make it work with the 7.5 V2. I have used the Nodemcu-32s and dev branch of esphome where this new model is clearly supported but I dont see anything on the display. Can you please advice? Thanks

Willi24c, 2020-08-07:
Hi Thonky81,
OK, I try to explain:
I am using the Waveshare 7.5 V2 with e-Paper HAT together with LOLIN32 V1.0.0. 

**Cutout from my epaper_75v2.yaml**

# Example configuration entry
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23    <-- on the e-Paper Hat this is DIN (Data in)

display:
  - platform: waveshare_epaper
    cs_pin: GPIO5
    dc_pin: GPIO19
    busy_pin: GPIO16
    reset_pin: GPIO17
    model: 7.50inV2
    rotation: 0°
    update_interval: 90s
You can see the connections, which most likely match with the pin numbers of your Nodemcu-32s. 
On the e-Paper HAT you should switch "Display Config" to B and Interface Config to 0 (4-Line SPI) 
and of course  +3.3V to Vcc (sorry for this eventually stupid statement).

I hope this helps, if you need more input please let me know.
Willi

Thoky81, 2020-08-08:
Hi,

Thanks for great reply. So finally I made it :-). I dont know why but when I change the board from Nodemcu-32s to es32dev its working.. I also used the gpio pins exactly as you - but this should not be a problem.

Thanks again

renjithstalin, 2020-10-23:
Hello Willi, I followed your same pin config, with esp32dev and esphome-dev branch, but can't get the display to work. VCC should be 3.3 or 5? Any help will be highly appreciated.
I got the same hardware as yours, Waveshare 7.5 V2 with e-Paper HAT & LOLIN32 V1.0.0.
Thanks
platform: ESP32
 board: esp32dev

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23
time:
  - platform: homeassistant
    id: esptime  
web_server:
  port: 80
  
display:
  - platform: waveshare_epaper
    cs_pin: GPIO5
    dc_pin: GPIO19
    busy_pin: GPIO16
    reset_pin: GPIO17
    model: 7.50inV2
    rotation: 0°
    update_interval: 90s

phiwa99, 2021-02-28:
> Hi Thonky81,
> OK, I try to explain:
> I am using the Waveshare 7.5 V2 with e-Paper HAT together with LOLIN32 V1.0.0.
>
> **Cutout from my epaper_75v2.yaml**
>
> # Example configuration entry
> spi:
> clk_pin: [...]

I use an olimex esp32-poe with the 7.5in V1 Display and it works,
If I change the yaml to `model: 7.50inV2` and change the  display to the V2 nothing happens. Is a change of the pins nessesary?