esphome/issues #1298: No "oscillation" for fan in HA

fantomnotabene, 2020-06-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, nodemcuv2

**Affected component:**

fan

**Description of problem:**
[image]
No oscillation toggle.

**Problem-relevant YAML-configuration entries:**
```yaml
fan:
  - platform: speed
    output: ballu_fan_speed_output
    oscillation_output: ballu_fan_oscillation_output
    name: "Ballu Fan"
```

**Logs (if applicable):**

```
[speed.fan:047]: Setting oscillation: OFF
```
Shows off on every interaction.
**Additional information and things you've tried:**

glmnet, 2020-06-15:
@blejdfist can you help here?

blejdfist, 2020-06-15:
@glmnet This should already be fixed by https://github.com/esphome/esphome/pull/1048

glmnet, 2020-06-15:
Ahh thanks 

@fantomnotabene would you like to try dev branch?

Let us know if it doesn’t work

closed 2020-06-15 by glmnet

fantomnotabene, 2020-06-15:
Thanks, `dev` works like a charm