esphome/issues #1301: Cannot use ADC sensor with ESP32 Cam

raresserban, 2020-06-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant OS 0.111.3, EspHome addon 1.14.4.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 Cam Ai Thinkerer

**Affected component:**

Analog to Digital Sensor:
https://esphome.io/components/sensor/adc.html?highlight=esp32

**Description of problem:**
I want to use the ADC sensor with an ESP32 Cam, I what to add a photoresistor on pin 13. However, I get an error that for ESP32 only pins 32 to 39 can be used (the documentation already states this).
However, I have checked the pinout of the ESP32 Cam board, and pin 13 should support ADC, according to this:
https://randomnerdtutorials.com/esp32-cam-ai-thinker-pinout/

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: adc
    pin: 13
    name: Hallway Brightness
    update_interval: 10s
    attenuation: 11db
```

**Logs (if applicable):**

```
None
```

**Additional information and things you've tried:**
I believe that for `board: esp32cam`, other pins besides 32 to 39 should also be supported by ADC.

foxacc, 2020-06-22:
I have the same trouble(get an error that for ESP32 only pins 32 to 39 can be used), I can not connect adc

glmnet, 2020-07-01:
While it is possible to use those pins as analog inputs, they cannot be used as analog input while you are using WiFi 

More info here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html

So what will be needed to do is a lot of work, stopping WiFi, reading the analog value and turning WiFiback on, it is not just listing those pins as available for analog input measurements.

glmnet, 2020-07-01:
May be you can try writing you own custom sensor following this guide https://esphome.io/components/sensor/custom.html

Then you can try to use this code (retrieved from the docs above)

```
#include <driver/adc.h>

...

    int read_raw;
    adc2_config_channel_atten( ADC2_CHANNEL_7, ADC_ATTEN_0db );

    esp_err_t r = adc2_get_raw( ADC2_CHANNEL_7, ADC_WIDTH_12Bit, &read_raw);
    if ( r == ESP_OK ) {
        printf("%d\n", read_raw );
    } else if ( r == ESP_ERR_TIMEOUT ) {
        printf("ADC2 used by Wi-Fi.\n");
    }

```

Note that, you should change the `printf()` inside the ESP_OK branch with the `publish_state` function

raresserban, 2020-07-02:
> While it is possible to use those pins as analog inputs, they cannot be used as analog input while you are using WiFi
>
> More info here [...]

That sucks, didn't know that...Will see how I handle this, probs going to get a second normal ESP32. Thank you.

rradar, 2020-07-02:
Going to close because this is a hardware limitation from the esp32

closed 2020-07-02 by rradar

rmit-s3686930-keyu, 2020-11-14:
I ended up solder a wire on GPIO 34, which uses ADC1

glmnet, 2020-11-15:
> I ended up solder a wire on GPIO 34, which uses ADC1 

Which cam?

I see GPIO34 is used as data pin on most cameras

rmit-s3686930-keyu, 2020-11-15:
> > I ended up solder a wire on GPIO 34, which uses ADC1
> 
> Which cam?
> 
> I see GPIO34 is used as data pin on most cameras

I am not using the camera. so GPIO 34 is free

jungle417, 2021-01-27:
> Did anybody solve this? I'm having the same issue.

How about using GPIO33 (which only drives an onboard LED)?  I'm in the process of trying this out myself.

raresserban, 2021-01-28:
I just got an I2C sensor instead:
https://esphome.io/components/sensor/bh1750.html?highlight=bh1750
https://github.com/airijia/esphome-config/blob/master/ESP32/ESP32-CAM-3in1.yaml

emeyedeejay, 2021-01-30:
> I just got an I2C sensor instead:
> https://esphome.io/components/sensor/bh1750.html?highlight=bh1750
> https://github.com/airijia/esphome-config/blob/master/ESP32/ESP32-CAM-3in1.yaml

I'm trying to use a photoresistor to count electricity usage flashing led. Looks like I'll need an additional node in the same location.

emeyedeejay, 2021-01-30:
> > Did anybody solve this? I'm having the same issue.
> 
> How about using GPIO33 (which only drives an onboard LED)? I'm in the process of trying this out myself. Obviously the LED would need to be removed first.

Interesting idea - any success?

jungle417, 2021-01-30:
> > > Did anybody solve this? I'm having the same issue.
> > 
> > 
> > How about using GPIO33 (which only drives an onboard LED)? I'm in the process of trying this out myself. Obviously the LED would need to be removed first.
> 
> Interesting idea - any success?

Yep, it works!  I'm going to use it to monitor the battery level of my lipo battery pack used to power the ESP32-Cam.  The battery is attached to a solar based charger.  When the voltage drops below a certain threshold I put the ESP32S into deep sleep mode to reduce any further battery discharge (I measured the current drain to be ~4mA in this mode vs 250mA+ when actively streaming video).  The hardest part is removing the LED but it CAN be done with some patience.  I then attached a pigtail to the cathode side of the PCB pad for access.  BTW, I've never been completely satisfied with the built-in ADCs of the ESP32 and in the past have used an ADS1115 instead with much greater success.  Unfortunately it uses an I2C interface so it wouldn't work with the Cam module.

jungle417, 2021-01-30:
Yep, it works! I'm going to use it to monitor the battery level of my lipo battery pack used to power the ESP32-Cam. The battery is attached to a solar based charger. When the voltage drops below a certain threshold I put the ESP32S into deep sleep mode to reduce any further battery discharge (I measured the current drain to be ~4mA in this mode vs 250mA+ when actively streaming video). The hardest part is removing the LED but it CAN be done with some patience. I then attached a pigtail to the cathode side of the PCB pad for access. BTW, I've never been completely satisfied with the built-in ADCs of the ESP32 and in the past have used an ADS1115 instead with much greater success. Unfortunately it uses an I2C interface so it wouldn't work with the Cam module.

fredddvargas, 2021-03-29:
Hi @jungle417,
can you share your code and explain a little bit more? I want to do a similar setup.. esp32 cam sending a photo every x min, sending battery voltage (source power will be 18650 + solar panel) and going into deepsleep . 

Thanks in advance,

jungle417, 2021-03-29:
Will do.  Look for my email.

emeyedeejay, 2021-03-29:
Email? Post it here, please - I'm also keen on this! Well done for figuring it out.

jungle417, 2021-04-01:
> Hi @jungle417,
> can you share your code and explain a little bit more? I want to do a similar setup.. esp32 cam sending a photo every x min, sending battery voltage (source power will be 18650 + solar panel) and going into deepsleep .
> 
> Thanks in advance,

For my setup I'm using two 3.7v Lipos connected in series.  These drive an LDO 5v regulator to power the ESP32 Cam.   Whenever the voltage level drops to ~7.2v I go to sleep for a set time. Since I only have a single solar charger, it constantly switches back and forth between the two batteries every 30 minutes throughout the day keeping both batteries charged.  A little unsophisticated but it works.  I accomplish this by controlling two solid state relays.  The length of my sleep time is dependent on the actual time.  For example, if the battery voltage reaches the shutdown threshold at 9:00am then I go to sleep for ~23 hours (I always try to wake up ~8:00am each day).  If the shutdown threshold occurs at 6:00pm then I go to sleep for ~14 hours).  And so on.  I know what time it is because I connect with an NTP server every time I wake up.  I incorporate canned code for streaming to a webserver.  I don’t use the built-in flash or take pictures. To reduce drastic changes with the ADC input I use a moving average function developed by a GitHub contributor (JChristensen/movingAvg).  Email me if you want access to my code.

emeyedeejay, 2021-04-01:
> For my setup I'm using two 3.7v Lipos connected in series. These drive an LDO 5v regulator to power the ESP32 Cam. Whenever the voltage level drops to ~7.2v I go to sleep for a set time. Since I [...]

I would like to take a look at your code if possible, please... Am I blind - where do I find your email address?
m i k e j u n o r AT g m a i l DOT c o m (no spaces) is me. Thanks!

jungle417, 2021-04-01:
> I would like to take a look at your code if possible, please... Am I blind - where do I find your email address?
> m i k e j u n o r AT g m a i l DOT c o m (no spaces) is me. Thanks!

georgejane@bellsouth.net   I tried contacting you thru GitHub but no good.  Use your personal email please.

jungle417, 2021-04-03:
> Hi @jungle417,
> can you share your code and explain a little bit more? I want to do a similar setup.. esp32 cam sending a photo every x min, sending battery voltage (source power will be 18650 + solar panel) and going into deepsleep .
> 
> Thanks in advance,

Here is the sketch I'm using.  It contains video streaming and NTP support code as well.  Strip those out and its pretty simple.  I monitor the Lipo voltage level every minute via GPIO33.  If it drops below my threshold it goes to sleep "x" number of hours depending upon the current time.  I also shut things down in the evening at 7:00pm to conserve batteries and wake up ~8:00am.  Since I need two Lipos in series and I only have one solar panel/charger, I use solid state relays to switch between Lipos.  I charge one for 30 minutes, then switch and charge the other.  This goes on all day until I go to sleep.  email me if you have any questions.

[bird_cam.ino.txt](https://github.com/esphome/issues/files/6253105/bird_cam.ino.txt)