esphome/issues #1302: platform DHT

kktmp1, 2020-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
docker home assistant

**ESP (ESP32/ESP8266, Board/Sonoff):**
M5StickC

**Affected component:**
- platform: dht
DHT Temperature+Humidity

**Description of problem:**
M5Stick crashed if you include platform DHT in your yaml, removing that sensor all works perfect.
I've tried with 3 different sticks, same behaviour.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: m5stickempty
  platform: ESP32
  board: m5stick-c
  platformio_options:
    upload_speed: 1500000

wifi:
  ssid: "MOVISTAR"
  password: "removed"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "kkk"

ota:
  password: "kkk"

sensor:
  - platform: dht
    pin: GPIO26
    temperature:
      name: Temperature
    humidity:
      name: Humidity
    update_interval: 10s

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/m5stickempty.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/m5stickempty
Processing m5stickempty (board: m5stick-c; framework: arduino; platform: espressif32@1.12.1)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
[... 274 lines cut ...]
[16:53:41][I][app:100]: ESPHome version 1.14.4 compiled on Jun 17 2020, 16:53:03
[16:53:41][C][wifi:415]: WiFi:
[16:53:41][C][wifi:283]:   SSID: [redacted]
[16:53:41][C][wifi:284]:   IP Address: 192.168.1.222
[16:53:41][C][wifi:286]:   BSSID: [redacted]
[16:53:41][C][wifi:287]:   Hostname: 'm5stickempty'
[16:53:41][C][wifi:291]:   Signal strength: -36 dB ▂▄▆█
[16:53:41][C][wifi:295]:   Channel: 11
[16:53:41][C][wifi:296]:   Subnet: 255.255.255.0
[16:53:41][C][wifi:297]:   Gateway: 192.168.1.1
[16:53:41][C][wifi:298]:   DNS1: 0.0.0.0
[16:53:41][C][wifi:299]:   DNS2: 0.0.0.0
[16:53:41][C][logger:175]: Logger:
[16:53:41][C][logger:176]:   Level: DEBUG
[16:53:41][C][logger:177]:   Log Baud Rate: 115200
[16:53:41][C][logger:178]:   Hardware UART: UART0
[16:53:41][C][captive_portal:169]: Captive Portal:
[16:53:41][C][ota:029]: Over-The-Air Updates:
[16:53:41][C][ota:030]:   Address: 192.168.1.222:3232
[16:53:41][C][ota:032]:   Using Password.
[16:53:41][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts

```

**Additional information and things you've tried:**

glmnet, 2020-06-17:
There is a known bug in v1.14.4. Which version are you running?

kktmp1, 2020-06-19:
> There is a known bug in v1.14.4. Which version are you running?

Yes, sorry, It's since i've updated to 1.14.4.

I also have another problems with some M5StickCs and wifi updates.  Maybe I should downgraded?

glmnet, 2020-06-25:
Duplicate of #1276 

if you have another issues please open another one. Thanks

closed 2020-06-25 by glmnet