esphome/issues #1308: BLE tracker uses too much RAM it causes other components to fail

baoshi, 2020-06-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

Custom board based on ESP32DevModule (esp32dev)

**Affected component:**

BLE Tracker
https://esphome.io/components/esp32_ble_tracker.html
https://esphome.io/components/sensor/xiaomi_lywsdcgq.html

**Description of problem:**

When my program is getting bigger (uses more memory), BLE tracker seems causes many weird problems:
1. Difficulty in connecting WiFi, giving strange error such as
```
[I][wifi:193]: WiFi Connecting to 'XXXXX...
[W][wifi:470]: Connecting to WiFi network failed. Are the credentials wrong?
```
After power cycle it connects well, but sometimes this problem happens again.

2. Spontaneous "Beacon timeout" error, so Wifi disconnected and reconnects several seconds later

3. OTA gives timeout error (seeing from board side, WiFi disconnects during update).

 ```
[D][ota:243]: OTA in progress: 42.2%
[D][ota:243]: OTA in progress: 45.5%
[D][ota:243]: OTA in progress: 48.8%
[D][ota:243]: OTA in progress: 51.9%
[D][ota:243]: OTA in progress: 55.2%
[D][ota:243]: OTA in progress: 58.1%
[D][ota:243]: OTA in progress: 61.4%
[D][ota:243]: OTA in progress: 64.6%
[D][ota:243]: OTA in progress: 67.9%
[W][ota:319]: Timeout waiting for data!
[W][ota:276]: Update end failed! Error: No Error

[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[E][WiFiClient.cpp:392] write(): fail on fd 55, errno: 11, "No more processes"
[D][BS:017]: Heap Free = 38956
[W][esp32_ble_tracker:061]: Too many BLE events to process. Some devices may not show up.
[D][BS:017]: Heap Free = 46388
[D][BS:017]: Heap Free = 46388
[D][BS:017]: Heap Free = 46388
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bletest
  platform: ESP32
  board: esp32dev
  includes:
    - my_binary_sensor.h

wifi:
  ssid: "AAAAAA"
  password: "BBBBB"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bletest Fallback Hotspot"
    password: "hdZvFjD65M2Y"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:
  id: bletracker

time:
  - platform: sntp
    id: sntp_time

switch:
  - platform: restart
    name: "Restart"
    
  - platform: gpio
    name: "Port A"
    pin: GPIO26
    restore_mode: RESTORE_DEFAULT_OFF
    id: out_port_a
    
  - platform: gpio
    name: "Port B"
    pin: GPIO32
    restore_mode: RESTORE_DEFAULT_OFF
    id: out_port_b
    
  - platform: gpio
    name: "USB"
    pin: GPIO16
    restore_mode: ALWAYS_ON
    id: out_usb
    
sensor:
  - platform: xiaomi_lywsdcgq
    id: xiaomi_sensor1
    mac_address: 4C:65:A8:DD:XX:XX
    temperature:
      id: workshop_temperature
      name: "Workshop Temperature"
    humidity:
      id: workshop_humidity
      name: "Workshop Humidity"
  - platform: xiaomi_lywsdcgq
    id: studyroom_sensor
    mac_address: 4C:65:A8:DD:XX:XX
    temperature:
      id: studyroom_temperature
      name: "Study Room Temperature"
    humidity:
      id: studyroom_humidity
      name: "Study Room Humidity"
  - platform: xiaomi_lywsdcgq
    id: filament_sensor
    mac_address: 4C:65:A8:DD:XX:XX
    humidity:
      id: filament_humidity
      name: "Safebox Humidity"

binary_sensor:
- platform: custom
  lambda: |-
    auto my_custom_sensor = new MyCustomBinarySensor();
    App.register_component(my_custom_sensor);
    return {my_custom_sensor};
  binary_sensors:
      name: "My Custom Binary Sensor"
      

```

```
#include "esphome.h"

class MyCustomBinarySensor : public PollingComponent, public BinarySensor {
 public:
  // constructor
  MyCustomBinarySensor() : PollingComponent(1000) {}

  void setup() override {
    // This will be called by App.setup()
    pinMode(0, INPUT);
    buf = (uint8_t *)malloc(10240);
  }
  void update() override {
    // This will be called every "update_interval" milliseconds.

    // Publish an OFF state
    ESP_LOGD("BS", "Heap Free = %u", ESP.getFreeHeap());
    bool state = digitalRead(0);
    publish_state(state);
  }
  
  uint8_t * buf;
};
```

**Logs (if applicable):**

```
Above
```

**Additional information and things you've tried:**

I inserted a custom components just to check free heap memory. My main program is very similar to this one. It allocates a buffer for display. All above mentioned error starts to occur when I increasingly allocate memory within setup(). Although heap free still give 4xk free memory.

If I allocate 8K -> 51020 heap free  --> OTA is ok,  "Beacon timeout" every 30 minutes or so.
If i allocate 10k -> 46436 heap free --> OTA fail, "Beacon timeout" every 10 minutes

bluelu, 2020-07-16:
Hi,

I have a similar issue for my own component (the ble stack using too much memory). 

There is a drop in replacement library (only minor modifications to the code needed, e.g. see https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/BLE_Beacon_Scanner/BLE_Beacon_Scanner.ino) soon to be released on the arduino library manager which uses the new nimble stack (which is also being used in IDF 4.0).  

I will use this component as a replacement for my BLE code.

closed 2020-11-27 by stale[bot]

lepinkainen, 2021-02-23:
I'm having the same issue, depending on the amount of BLE beacons visible, the board seems to reboot or crash completely.