esphome/issues #1309: Waveshare 5.83 inch E-Paper display prevents wifi connection

lars-1, 2020-06-24:
Hello, I have the following issue:

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- Docker: esphome/esphome:dev

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (waveshare e-Paper ESP8266 Driver Board)

**Affected component:**

https://esphome.io/components/display/waveshare_epaper.html

**Description of problem:**

If I use "model: 5.83in" the wifi connection won't work anymore.

**Problem-relevant YAML-configuration entries:**
```yaml
display:
  - platform: waveshare_epaper
    cs_pin: 15
    dc_pin: 4
    busy_pin: 5
    reset_pin: 2
#    model: 4.20in
    model: 5.83in
```

**Logs (if applicable):**

```
Done! Flashing is complete!

Showing logs:
[21:55:22]
[21:55:22]SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
[21:55:22][I][logger:156]: Log initialized
[21:55:22][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[21:55:22][I][app:028]: Running through setup()...
[21:55:22][V][app:029]: Sorting components by setup priority...
[21:55:22][C][spi:022]: Setting up SPI bus...
[21:55:22][D][text_sensor:015]: 'ESPHome Version': Sending state '1.15.0-dev Jun 24 2020, 19:54:14'
[21:55:23][E][waveshare_epaper:107]: Timeout while displaying image!
[21:55:24][C][wifi:033]: Setting up WiFi...
[21:55:24][V][wifi_esp8266:051]: Disabling AP.
[21:55:24]bcn 0
[21:55:24]del if1
[21:55:24]usl
[21:55:24]mode : null
[21:55:24][V][wifi_esp8266:041]: Enabling STA.
[21:55:24]mode : sta(98:f4:ab:xx:xx:xx)
[21:55:24]Fatal exception[V][wifi_esp8266:433]: Event: Changed Mode old=AP new=OFF
[21:55:24]wifi evt: 8
[21:55:24] 28(LoadProhibitedCause):
[21:55:24]epc1=0x4025549e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
[21:55:24]
[21:55:24]Exception (28):
[21:55:24]epc1=0x4025549e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[21:55:24]
[21:55:24]>>>stack>>>
[21:55:24]
[21:55:24]ctx: cont
[21:55:24]sp: 3ffffc30 end: 3fffffc0 offset: 01a0
[21:55:24]3ffffdd0:  402555fb 00000001 00000001 402555af  
[21:55:24]3ffffde0:  00000008 00000000 00000000 3fffb944  
[21:55:24]3ffffdf0:  402556d2 00000000 3ffea8d4 00000000  
[21:55:24]3ffffe00:  4020d05b 00000000 3ffea8d4 4020d005  
[21:55:24]3ffffe10:  40200101 4bc62900 000006f9 4020d046  
[21:55:24]3ffffe20:  3ffea8d4 3ffea8d4 00000001 4020d614  
[21:55:24]3ffffe30:  01012900 3ffea8d4 3fff2944 4020d6a0  
[21:55:24]3ffffe40:  00010001 30e44323 3fff1250 00000020  
[21:55:24]3ffffe50:  3ffea8d4 3ffea8d4 3fff2944 4020bde4  
[21:55:24]3ffffe60:  3fff2944 00000008 3fff1250 40224b38  
[21:55:24]3ffffe70:  3fff3344 00000005 3fff1250 40224ba5  
[21:55:24]3ffffe80:  3fff28e8 00000007 3fff1250 4020f485  
[21:55:24]3ffffe90:  3fff3344 0000000d 3fff1650 3fff1250  
[21:55:24]3ffffea0:  3fff2e34 402323aa 3fff2e8c 40223986  
[21:55:24]3ffffeb0:  00000000 3fff3304 00000010 3fff1250  
[21:55:24]3ffffec0:  3fff2e34 3fff3324 00000000 40211659  
[21:55:24]3ffffed0:  3fff2b68 00000000 00000000 3fff2ba0  
[21:55:24]3ffffee0:  00000000 00000000 00000000 40226850  
[21:55:24]3ffffef0:  00000000 40226850 00000000 40226850  
[21:55:24]3fffff00:  00000000 40226850 00000000 40226850  
[21:55:24]3fffff10:  00000000 00000000 3fff2a98 00000000  
[21:55:24]3fffff20:  00000000 3fff2ac0 00000000 00000000  
[21:55:24]3fffff30:  00000000 40226850 00000000 40226850  
[21:55:24]3fffff40:  00000000 40226850 00000000 40226850  
[21:55:24]3fffff50:  00000000 40226850 00000000 00000000  
[21:55:24]3fffff60:  3fff2fe4 00000000 40210cc0 40210c4c  
[21:55:24]3fffff70:  3fff3064 3fff2f74 3fff2edc feefeffe  
[21:55:24]3fffff80:  3fff2f74 3fff3064 feefeffe feefeffe  
[21:55:24]3fffff90:  feefeffe feefeffe feefeffe 3fff158c  
[21:55:24]3fffffa0:  3fffdad0 00000000 3fff155c 402219d4  
[21:55:24]3fffffb0:  feefeffe feefeffe 3ffe85d0 40100909  
[21:55:24]<<<stack<<<
[21:55:24]
[21:55:24]last failed alloc call: 40100698(684)
[21:55:24]
[21:55:24] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[21:55:24]
[21:55:24]load 0x4010f000, len 1384, room 16 
[21:55:24]tail 8
[21:55:24]chksum 0x2d
[21:55:24]csum 0x2d
[21:55:24]v8b899c12
[21:55:24]~ld
[21:55:24]
[21:55:24]SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
[21:55:24][I][logger:156]: Log initialized
[21:55:24][C][ota:366]: There have been 1 suspected unsuccessful boot attempts.
[21:55:24][I][app:028]: Running through setup()...
[21:55:24][V][app:029]: Sorting components by setup priority...
[21:55:24][C][spi:022]: Setting up SPI bus...
[21:55:24][D][text_sensor:015]: 'ESPHome Version': Sending state '1.15.0-dev Jun 24 2020, 19:54:14'
[21:55:25][E][waveshare_epaper:107]: Timeout while displaying image!
[21:55:25][C][wifi:033]: Setting up WiFi...
[21:55:25][V][wifi_esp8266:051]: Disabling AP.
[21:55:25]bcn 0
[21:55:25]del if1

```

**Additional information and things you've tried:**

- problem exists even when display is disconnected
- changing the model resolves the issue but obviously won't work

sredfern, 2020-06-25:
Hi @lars-1,

1) The 8266 board works fine with the 4.2in display? 
2) Could you post your entire config? 

Seems like this problem is a little different from:
This issue seems strangely similar to https://github.com/esphome/issues/issues/1190

sredfern, 2020-06-25:
Also, I'd have a look at the SPI definition in this document: 
https://www.waveshare.com/w/upload/a/a7/E-Paper_ESP8266_Driver_Board_EN.pdf

sredfern, 2020-06-25:
Also, does the display work when the wifi crashes out in the error log?

lars-1, 2020-06-25:
> Hi @lars-1,
> 
> 1. The 8266 board works fine with the 4.2in display?
> 2. Could you post your entire config?
> 
> Seems like this problem is a little different from:
> This issue seems strangely similar to #1190

Hi @sredfern , thanks for your fast response.

This is the full config:
```
# ESP8266 EPAPER BOARD

substitutions:
  plug_name: "ESP8266 Display"

esphome:
  name: esp8266_epaper_wz
  platform: ESP8266
  board: esp12e

wifi:
  ssid: "removed"
  password: "removed"
  domain: "removed"

  ap:
    ssid: "${plug_name} Hotspot"
    password: "removed"

captive_portal:

logger:
  level: verbose

api:

ota:

web_server:
  port: 80

time:
  - platform: homeassistant
    timezone: "Europe/Berlin"
    id: homeassistant_time

spi:
  clk_pin: 14
  mosi_pin: 13

display:
  - platform: waveshare_epaper
    cs_pin: 15
    dc_pin: 4
    busy_pin: 5
    reset_pin: 2
#    model: 4.20in
    model: 5.83in
    lambda: |-
      it.line(0, 0, 50, 50); 
      it.line(0, 0, 100, 50);
      it.rectangle(50, 60, 30, 42);
      it.filled_rectangle(50, 60, 30, 42);
      it.circle(25, 25, 10);
      it.filled_circle(25, 25, 10);

text_sensor:
  - platform: version
    name: "${plug_name} - ESPHome Version"
  
sensor:
  - platform: wifi_signal
    name: "${plug_name} - WiFi Signal"
    update_interval: 60s
  
  - platform: uptime
    name: "${plug_name} - Uptime"
    icon: mdi:clock-outline
    update_interval: 60s      

```

I dont have the 4.2" display. I read about a similar issue and then discovered that the ESP boots successfully if I change the model.

lars-1, 2020-06-25:
> Also, does the display work when the wifi crashes out in the error log?

Sometimes it blinks black and white and gets faster with every flash until it stops.

lars-1, 2020-06-25:
> Also, I'd have a look at the SPI definition in this document:
> https://www.waveshare.com/w/upload/a/a7/E-Paper_ESP8266_Driver_Board_EN.pdf

There are multiple sources with different pins so I copied the pins from a working demo program from waveshare

sredfern, 2020-06-25:
Could you try again with a bare bones config? Post the config and output here. 
esphome, wifi, spi, display 

Also, seems like the rest and busy pins don't match the schematic: https://www.waveshare.com/w/upload/7/7e/ESP8266_e-paper.pdf

sredfern, 2020-06-25:
Also, I'd suggest using the direct GPIO pins over the board pins. The waveshare esp8266 board is different from generic 8266 reference boards. You can reference them like this: 

spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13

lars-1, 2020-06-25:
I tried the pins from the schematic too, but these are the pins from a waveshare demo project which I tested on this board. I can test it again with "GPIOxx" later this afternoon. I think the issue must be with the pins or the implementation of the 5.83" display because as I said the ESP boots with other "model" settings.

sredfern, 2020-06-25:
My "implementation" of the 5.83 display was to copy the 7.5in function, update some hex values and calculate the screen resolution in hex. I'm kind of amazed people are still using the 8266 board given that the waveshare esp32 ones are so cheap now days.

lars-1, 2020-06-25:
I bought the display a few days ago and the esp8266 board was suggested by the shop and because I have a few esp8266 around the home I didn't thought about an esp32 board. Can you confirm that the waveshare esp32 board works?

sredfern, 2020-06-25:
Yes, this configuration is working with the ESP32 waveshare board now. 

`esphome:
  name: ink_fridge
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "PandaNet"
  password: "Pandasex!"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Inkfridge Fallback Hotspot"
    password: "1ZmQuh8Lcpdf"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "zRoBMwAPUEF2dCbR"
  services:
    - service: update_display
      then:
        - component.update: my_display

ota:
  password: "zRoBMwAPUEF2dCbR"

font:
  - file: "Sansation-Regular.ttf"
    id: sansation_18
    size: 18
  - file: "Sansation-Regular.ttf"
    id: sansation_45
    size: 45
  - file: "Sansation-Regular.ttf"
    id: sansation_30
    size: 35
  - file: "Sansation-Regular.ttf"
    id: sansation_25
    size: 24
  - file: "WeatherIcons-Regular_Targeted.ttf"
    id: weather_font_105
    size: 105
  - file: "WeatherIcons-Regular_Targeted.ttf"
    id: weather_font_115
    size: 115
  - file: "WeatherIcons-Regular_Targeted.ttf"
    id: weather_font_125
    size: 125

sensor:
  - platform: homeassistant
    entity_id: sensor.felix_bedroom_temperature2
    id: felix_bed
  - platform: homeassistant
    entity_id: sensor.dark_sky_apparent_temperature
    id: apparent_temp
  - platform: homeassistant
    entity_id: sensor.dark_sky_daytime_high_temperature_0d
    id: d0_high_temp
  - platform: homeassistant
    entity_id: sensor.dark_sky_overnight_low_temperature_0d
    id: d0_low_temp
  - platform: homeassistant
    entity_id: sensor.dark_sky_temperature
[... 146 lines cut ...]
          } else if ( id(dark_sky_icon).state == "cloudy") { 
            it.print( 10, 455, id(weather_font_115), "h"); 
            it.print( 155, 555,  id(sansation_30), "Cloudy"); 
          } else if ( id(dark_sky_icon).state == "partly-cloudy-day"  ) { 
            it.print( 3, 455, id(weather_font_105), "i"); 
            it.print( 153, 553,  id(sansation_30), "Partly Cloudly");
          } else if ( id(dark_sky_icon).state == "partly-cloudy-night" ) { 
            it.print(  10, 465, id(weather_font_105), "j"); 
            it.print( 153, 553, id(sansation_30), "Partly Cloudly"); 
          } else if ( id(dark_sky_icon).state  == "hail" ) { 
            it.print( 10, 455, id(weather_font_105), "k"); 
            it.print(  153, 553, id(sansation_30), "Hail"); 
          } else if ( id(dark_sky_icon).state == "thunderstorm" ) { 
            it.print( 10, 455, id(weather_font_105), "l"); 
            it.print(  153, 553, id(sansation_30), "Thunder");
          }

# it.print(0, 200, id(weather), "012345");
# it.print(0, 250, id(weather), "67890-_");
# it.print(0, 300, id(weather), "abcdefghi");
# it.print(0, 350, id(weather), "jklmnopq");
# it.print(0, 400, id(weather), "rstuvwxyz");
# it.print(0, 450, id(weather), "ABCDEFGHI");
# it.print(0, 500, id(weather), "JKLMNOPQ");
# it.print(0, 550, id(weather), "RSTUVWXYZ"); `

sredfern, 2020-06-25:
Yes, it's working with this config. 

```
esphome:
  name: ink_fridge
  platform: ESP32
  board: esp32dev

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Inkfridge Fallback Hotspot"
    password: ""

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""
  services:
    - service: update_display
      then:
        - component.update: my_display

ota:
  password: ""

font:
  - file: "Sansation-Regular.ttf"
    id: sansation_18
    size: 18
  - file: "Sansation-Regular.ttf"
    id: sansation_45
    size: 45
  - file: "Sansation-Regular.ttf"
    id: sansation_30
    size: 35
  - file: "Sansation-Regular.ttf"
    id: sansation_25
    size: 24
  - file: "WeatherIcons-Regular_Targeted.ttf"
    id: weather_font_105
    size: 105
  - file: "WeatherIcons-Regular_Targeted.ttf"
    id: weather_font_115
    size: 115
  - file: "WeatherIcons-Regular_Targeted.ttf"
    id: weather_font_125
    size: 125

sensor:
  - platform: homeassistant
    entity_id: sensor.felix_bedroom_temperature2
    id: felix_bed
  - platform: homeassistant
    entity_id: sensor.dark_sky_apparent_temperature
    id: apparent_temp
  - platform: homeassistant
    entity_id: sensor.dark_sky_daytime_high_temperature_0d
    id: d0_high_temp
  - platform: homeassistant
    entity_id: sensor.dark_sky_overnight_low_temperature_0d
    id: d0_low_temp
  - platform: homeassistant
[... 139 lines cut ...]
            it.print( 10, 455, id(weather_font_115),  "e"); 
            it.print( 153, 553, id(sansation_30), "Sleet"); 
          } else if ( id(dark_sky_icon).state  == "wind" ) { 
            it.print( 15, 455, id(weather_font_115), "f"); 
            it.print( 153, 553, id(sansation_30), "Windy"); 
          } else if ( id(dark_sky_icon).state ==  "fog" ) { 
            it.print( 10, 455, id(weather_font_105), "g"); 
            it.print( 153, 553, id(sansation_30), "Fog"); 
          } else if ( id(dark_sky_icon).state == "cloudy") { 
            it.print( 10, 455, id(weather_font_115), "h"); 
            it.print( 155, 555,  id(sansation_30), "Cloudy"); 
          } else if ( id(dark_sky_icon).state == "partly-cloudy-day"  ) { 
            it.print( 3, 455, id(weather_font_105), "i"); 
            it.print( 153, 553,  id(sansation_30), "Partly Cloudly");
          } else if ( id(dark_sky_icon).state == "partly-cloudy-night" ) { 
            it.print(  10, 465, id(weather_font_105), "j"); 
            it.print( 153, 553, id(sansation_30), "Partly Cloudly"); 
          } else if ( id(dark_sky_icon).state  == "hail" ) { 
            it.print( 10, 455, id(weather_font_105), "k"); 
            it.print(  153, 553, id(sansation_30), "Hail"); 
          } else if ( id(dark_sky_icon).state == "thunderstorm" ) { 
            it.print( 10, 455, id(weather_font_105), "l"); 
            it.print(  153, 553, id(sansation_30), "Thunder");
          }
```

lars-1, 2020-06-26:
Thank you for your config @sredfern . Is it normal that the ESP needs a power cycle after flashing and the display flashes multiple times when updating the screen?

sredfern, 2020-06-26:
Did you get the esp8266 board working? Was it just a config problem?

lars-1, 2020-06-27:
No, I forgot to mention that I switched to the ESP32. It works with your config, but I think it is not normal that the display flashes a few times when refreshing. I found out that my display is a "5.83inch **(B)**". Maybe thats the issue?

sredfern, 2020-06-27:
It's normal for the eink display to flash a couple of times (within 3 seconds) to finish updating the display. 

Did you re-test on the 8266 using GPIO pins?

DJTerentjev, 2020-06-27:
-- Sent from Yandex.Mail for mobile

lars-1, 2020-06-27:
I bought some hardware and this is the result

ESP32 + 5.83inch: ok, but weird flickering when updating: [demo gif](https://gifyu.com/image/uGyu)
ESP32 + 4.20inch: ok
ESP8266 + 5.83inch: does not work (rebooting)
ESP8266 + 4.20inch: ok

closed 2020-12-05 by stale[bot]

thjubeck, 2021-03-12:
> Yes, it's working with this config.
>
> ```
> esphome:
>   name: ink_fridge
>   platform: ESP32
>   board: esp32dev
>
> wifi:
>   ssid: ""
>   password: ""
>
>   # Enable fallback hotspot (captive portal) in case wifi [...]

Do you still have the font files? I am trying to set up something similar.