esphome/issues #1310: Sonoff touch TX touch backlight turn off

MeloT88, 2020-06-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio in docker container
**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Touch TX Series 3 gangs
**Affected component:**

https://esphome-configs.io/devices/sonoff-t1-r2/

**Description of problem:**
I would like to separate the backlight of the touch button from the relay and be able to turn off the back lights of each single touch. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: masterwallswitch
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ""
    password: ""

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Master Touch1"
    on_press:
      - switch.toggle: relay_1
  - platform: gpio
    pin:
      number: GPIO9
      mode: INPUT_PULLUP
      inverted: True
    name: "Master Touch2"
    # on_press:
    #   - switch.toggle: relay_2
    on_multi_click:
      - timing:
          - ON for at most 1s
          - OFF for at most 1s
          - ON for at most 1s
          - OFF for at least 0.2s
        then:
          - homeassistant.service:
              service: switch.toggle
              data:
                entity_id: switch.bedside_lamps2_2
          - logger.log: "Dining Lamp"  
  - platform: gpio
    pin:
      number: GPIO10
      mode: INPUT_PULLUP
      inverted: True
    name: "Master Touch3"
    # on_press:
    #   - switch.toggle: relay_3
    on_multi_click:
      - timing:
          - ON for at most 1s
          - OFF for at most 1s
          - ON for at most 1s
          - OFF for at least 0.2s
        then:
          - homeassistant.service:
              service: switch.toggle
              data:
                entity_id: switch.bedside_lamps_2
          - logger.log: "Dining Lamp"
      - timing:
          - ON for 3s to 5s
          - OFF for at least 0.5s
        then:
          - homeassistant.service:
              service: homeassistant.turn_off
              data:
                entity_id: group.common_area_lights
          - logger.log: "Single Long Clicked"
  - platform: status
    name: "Sonoff TX 3 Gang Status"

switch:
  - platform: gpio
    name: "Master Relay1"
    pin: GPIO12
    id: relay_1
  - platform: gpio
    name: "Master Relay2"
    pin: GPIO5
    id: relay_2
  - platform: gpio
    name: "Master Relay3"
    pin: GPIO4
    id: relay_3

output:
  # Register the blue LED as a dimmable output ....
  - platform: esp8266_pwm
    id: blue_led
    pin: GPIO13
    inverted: True

light:
  # ... and then make a light out of it.
  - platform: monochromatic
    name: "Sonoff TX 3 Gang Blue LED"
    output: blue_led
    internal: true
    id: blue_led_light
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

zewelor, 2020-06-25:
From what I've understand, its hardwired, so its not possible without hardware modifications.

MeloT88, 2020-06-25:
That is what I though but, if I was flashing it with Tasmota there is a option to disable the LED (ledstate). How is it possible that?

zewelor, 2020-06-25:
But was it working ?

MeloT88, 2020-06-25:
I only tested on Sonoff Touch 1 gang old series. Still working with Tasmota, I just inverted the LED status to be off when the light is off.
I have not tried with the TX series unfortunately. I have flashed them using ESPhome right away

zewelor, 2020-06-25:
Some time ago I've desoldered that leds, from T1 2 gang, because it was not possible to control them. I could assume it was done same way on TX series. Old touch was 1 gang only, and it could have spare pin just for that led control.

MeloT88, 2020-06-25:
ok thanks for that. I will try later

lukaszkac, 2020-07-01:
I have a few sonoff TX and it also annoys me that I can't turn off the button illumination from software. It is not possible because was used capacitive touch toggle switch with LED such as HCSWIT0063. We also have HCSWIT0063 alone and I know it is not posible to disable button illumination because that's how they were built. In Hi state they shine more intensively than in LOW. The option to turn off the LED (ledstate) rather applies to the LED on GPIO13, i.e. this additional diode. It's not possible without hardware modifications. I'm sorry but it's not possible without hardware modifications.

MeloT88, 2020-07-02:
Thanks @lukaszkac I don't want to modify the hardware so I think I will cover that switch with a 3d printed box during the night

closed 2020-11-13 by stale[bot]