esphome/issues #1311: Web Server API Basic Authorization not working correctly

IQNeoXen, 2020-06-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker, but this only affects the Web Server API on esp-devices so this shouldn't be relevant

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (ESP-01S)

**Affected component:**

Web Server Api

**Description of problem:**

I've activated the Web Server with username/password according to the documentation. When using the Web-Interface, I can log in using my credentials and everything works as expected. When trying to access the REST-Api, I get an 401 unauthorized every time. The relevant code from my python-script looks like this:
```python
    url = "http://my-ip-address/switch/myid/turn_on"

    headers = {
        'Authorization': 'Basic mytoken'
    }
    
    response = requests.post(url=url, headers=headers)
```

The token has been generated like this: `username:password` in base64.

**Additional information and things you've tried:**
I've tried using other programs like Postman & Insomnia using basic-auth, I'm still getting the same error.

JT089, 2020-06-26:
Got exactly the same problem (but thought, it was a problem within my environment). Thanks for reporting @IQNeoXen! Hope it will be fixed soon.

beatkind, 2020-06-26:
Same here! Thanks @IQNeoXen for the details

closed 2020-12-05 by stale[bot]

5tan, 2020-12-09:
same issue here.

I own Sonoff 4CH PRO and Sonoff SV. Only SV seems to be affected