esphome/issues #1314: MQTT automation conditions are not

algirdasc, 2020-06-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Esphome 1.14.4 on Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (LEDENET Smart WiFi LED Controller 5 Channel)

**Affected component:**

https://esphome.io/components/mqtt.html?highlight=mqtt%20json#mqtt-publish-json-action

**Description of problem:**
After update from 1.14.3 to 1.14.4 version MQTT JSON message condition seems broken. When `{"occupancy": true}` both of conditions are evaluated to true. It looks like I cannot re-use `x["occupancy"]`, because after first use it is evaluated to `false` regardless it is set to `true`. I've tried reordering conditions which solved the problem - putting `return x["occupancy"] == false;` in front of the condition list solves the issue. It can be reproduced in 3 same devices.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  discovery: False
  on_json_message:
    topic: zigbee2mqtt/motion1_sensor
    then:
      - if:
          condition: 
            and:
              - lambda: 'return x["occupancy"] == true;'
              - light.is_on: ${esp_id}_strip
          then:
            - light.turn_on: 
                id: ${esp_id}_strip
                brightness: 65%
                white: 50%
      - if:
          condition: 
            and:
              - lambda: 'return x["occupancy"] == false;'
              - light.is_on: ${esp_id}_strip
          then:
            - light.turn_on: 
                id: ${esp_id}_strip
                brightness: 15%      
                white: 15%
                transition_length: 60s      
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

definitio, 2020-10-26:
I have bootloops or wifi reconnects triggered by MQTT json messages since 1.14.4:
```
[20:41:38]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[20:41:38]
[20:41:38]Exception (28):
WARNING Exception type: unknown
[20:41:38]epc1=0x4000bdcb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[20:41:38]
[20:41:38]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[20:41:38]
[20:41:38]ctx: cont
[20:41:38]sp: 3ffffb10 end: 3fffffc0 offset: 0190
[20:41:38]3ffffca0:  3ffffd20 3fff1dbc 3ffffcc8 4021211e  
WARNING Decoded 0x4021211e: ArduinoJson::Internals::EnableIf<ArduinoJson::Internals::StringTraits<char const*, void>::has_equals, bool>::type ArduinoJson::Internals::JsonVariantComparisons<ArduinoJson::Internals::JsonObjectSubscript<char const*> >::equals<char const*>(char const* const&) const
[20:41:38]3ffffcb0:  3fff0628 3fff05f4 3fff2e68 402121cc  
WARNING Decoded 0x402121cc: std::_Function_handler<bool (ArduinoJson::JsonObject const&), setup::{lambda(ArduinoJson::JsonObject const&)#14}>::_M_invoke(std::_Any_data const&, ArduinoJson::JsonObject const&) at main.cpp
[20:41:38]3ffffcc0:  3fff47fc 3ffe942e 3ffe944d 3f800000  
[20:41:38]3ffffcd0:  3f800000 3f800000 3f800000 40100119  
WARNING Decoded 0x40100119: std::function<bool (ArduinoJson::JsonObject const&)>::operator()(ArduinoJson::JsonObject const&) const
[20:41:38]3ffffce0:  3f800000 3f800000 3f800000 40213e61  
WARNING Decoded 0x40213e61: esphome::LambdaCondition<ArduinoJson::JsonObject const&>::check(ArduinoJson::JsonObject const&)
[20:41:38]3ffffcf0:  3f800000 3f800000 3f800000 402285d5  
WARNING Decoded 0x402285d5: esphome::AndCondition<ArduinoJson::JsonObject const&>::check(ArduinoJson::JsonObject const&)
[20:41:38]3ffffd00:  3ffffeb4 3fff47fc 3fff2e6c 40211c62  
WARNING Decoded 0x40211c62: esphome::IfAction<ArduinoJson::JsonObject const&>::play_complex(ArduinoJson::JsonObject const&)
[20:41:38]3ffffd10:  00000001 3fff47fc 3fff2e0c 402287bd  
WARNING Decoded 0x402287bd: esphome::Action<ArduinoJson::JsonObject const&>::play_next_(ArduinoJson::JsonObject const&)
[20:41:38]3ffffd20:  3fff1d64 00000101 ffffff00 40211c08  
WARNING Decoded 0x40211c08: std::_Function_handler<void (ArduinoJson::JsonObject const&), esphome::IfAction<ArduinoJson::JsonObject const&>::add_then(std::vector<esphome::Action<ArduinoJson::JsonObject const&>*, std::allocator<esphome::Action<ArduinoJson::JsonObject const&>*> > const&)::{lambda(ArduinoJson::JsonObject const&)#1}>::_M_invoke(std::_Any_data const&, ArduinoJson::JsonObject const&)
[20:41:38]3ffffd30:  3ffe8300 00000002 00000d01 40210ac4  
WARNING Decoded 0x40210ac4: esphome::LambdaAction<ArduinoJson::JsonObject const&>::play(ArduinoJson::JsonObject const&)
[20:41:38]3ffffd40:  00000000 3fff482e 00000000 40211c2f  
WARNING Decoded 0x40211c2f: esphome::Action<ArduinoJson::JsonObject const&>::play_complex(ArduinoJson::JsonObject const&)
[20:41:38]3ffffd50:  00000000 3fff47fc 3fff2cb4 402287bd  
WARNING Decoded 0x402287bd: esphome::Action<ArduinoJson::JsonObject const&>::play_next_(ArduinoJson::JsonObject const&)
[20:41:38]3ffffd60:  3fff4700 3fff2c38 3fff2c00 40211c08  
WARNING Decoded 0x40211c08: std::_Function_handler<void (ArduinoJson::JsonObject const&), esphome::IfAction<ArduinoJson::JsonObject const&>::add_then(std::vector<esphome::Action<ArduinoJson::JsonObject const&>*, std::allocator<esphome::Action<ArduinoJson::JsonObject const&>*> > const&)::{lambda(ArduinoJson::JsonObject const&)#1}>::_M_invoke(std::_Any_data const&, ArduinoJson::JsonObject const&)
[20:41:38]3ffffd70:  3fff0101 00000000 00000000 40210ac4  
WARNING Decoded 0x40210ac4: esphome::LambdaAction<ArduinoJson::JsonObject const&>::play(ArduinoJson::JsonObject const&)
[20:41:38]3ffffd80:  3f800000 3ffefd2c 3fff2c38 40211c2f  
WARNING Decoded 0x40211c2f: esphome::Action<ArduinoJson::JsonObject const&>::play_complex(ArduinoJson::JsonObject const&)
[20:41:38]3ffffd90:  3fff05f4 3fff47fc 3fff2c64 402287bd  
WARNING Decoded 0x402287bd: esphome::Action<ArduinoJson::JsonObject const&>::play_next_(ArduinoJson::JsonObject const&)
[20:41:38]3ffffda0:  3ffffeb4 3fff47fc 3fff2c64 40211c36  
WARNING Decoded 0x40211c36: esphome::Action<ArduinoJson::JsonObject const&>::play_complex(ArduinoJson::JsonObject const&)
[20:41:38]3ffffdb0:  3ffffe3c 3fff47fc 3fff2c3c 402287d4  
WARNING Decoded 0x402287d4: esphome::ActionList<ArduinoJson::JsonObject const&>::play(ArduinoJson::JsonObject const&)
[20:41:38]3ffffdc0:  3ffffeb4 3fff47fc 3fff2c3c 40211c95  
WARNING Decoded 0x40211c95: esphome::IfAction<ArduinoJson::JsonObject const&>::play_complex(ArduinoJson::JsonObject const&)
[20:41:38]3ffffdd0:  3fff47fc 3fff47fc 3fff2bd4 402287d4  
WARNING Decoded 0x402287d4: esphome::ActionList<ArduinoJson::JsonObject const&>::play(ArduinoJson::JsonObject const&)
[20:41:38]3ffffde0:  3ffffeb4 3fff47fc 3fff2bd4 40211c95  
WARNING Decoded 0x40211c95: esphome::IfAction<ArduinoJson::JsonObject const&>::play_complex(ArduinoJson::JsonObject const&)
[20:41:38]3ffffdf0:  3ffefa90 3fff47fc 3fff2b6c 402287d4  
WARNING Decoded 0x402287d4: esphome::ActionList<ArduinoJson::JsonObject const&>::play(ArduinoJson::JsonObject const&)
[20:41:38]3ffffe00:  3ffffeb4 3fff47fc 3fff2b6c 40211c95  
WARNING Decoded 0x40211c95: esphome::IfAction<ArduinoJson::JsonObject const&>::play_complex(ArduinoJson::JsonObject const&)
[20:41:38]3ffffe10:  3ffffe80 3fff47fc 3fff2b0c 402287d4  
[... cut ...]
[20:41:38][C][ota:366]: There have been 4 suspected unsuccessful boot\xe0\xd1\xff[C][light:097]: Setting up light 'test'...
[20:41:38][D][light:265]: 'test' Setting:
[20:41:38][D][light:274]:   State: ON
[20:41:38][D][light:278]:   Brightness: 100%
[20:41:38][C][wifi:033]: Setting up WiFi...
[20:41:38][I][wifi:194]: WiFi Connecting to 'iot_net'...
[20:41:41][W][wifi_esp8266:392]: Event: Disconnected ssid='iot_net' reason='Probe Request Unsuccessful'
[20:41:41][W][wifi:485]: Error while connecting to network.
[20:41:41][W][wifi:522]: Restarting WiFi adapter...
```

definitio, 2021-02-23:
The issue still exists.

closed 2021-07-07 by stale[bot]