esphome/issues #1315: Possible Merge Problems with Branches for Releases

herostrat, 2020-06-29:
Hey guys, it seems (to me) that there were some problems during the last release.

See for example this component:
https://github.com/esphome/esphome/pull/896

It was merged into dev, but did not find its way into master for the last release.

Another strange point is the nomenclature for the branches in general.
It seems there are multiple different structures:

- esphome
  - master
  - dev
  - beta
- esphome-docs
  - current
  - beta
  - next

Is there any docu on how the release cycle works, e.g. dev => beta => master?
I only found infos in https://github.com/esphome/feature-requests/issues/658 .

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-07-01:
Latest release is 1.14, the 1.14.4 and 1.14.5 are point releases and don't include new features, just bug fixes. So esphome/esphome#896 is expected to be released in 1.15

Branch names correlates 
master -> current
dev -> next
beta -> beta

this might look confusing, but I don't think it's a good idea changing it now, as a bonus when I see current branch I know I'm in docs repo.

Not much info on release process available yet. There are some scripts here https://github.com/esphome/esphome-release but no docs on how to use them.

herostrat, 2020-07-06:
Ok, thanks for the answer and sorry for the noise!

closed 2020-07-06 by herostrat

fjufju, 2020-07-22:
@glmnet any news about 1.15 release date? I hope it will be released soon.