esphome/issues #1317: Bang Bang climate control does not switch from "Off" to "Idle" until it has cycled through "Cool"

jptrsn, 2020-07-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini

**Affected component:**

[Bang Bang Climate](https://esphome.io/components/climate/bang_bang.html)

**Description of problem:**
When changing the state of the bang-bang climate from off to cool, the thermostat remains in the "Off" state unless it transitions into the "Cool" state. For example, if I set my air conditioner to 25 degrees, and the temperature sensor is at 23 degrees, then I change the thermostat state from "Off" to "Cool" it remains in the off state. This doesn't trigger the `idle_action` as expected.

**Problem-relevant YAML-configuration entries:**
```yaml
climate:
  - platform: bang_bang
    name: ${DEVICE_NAME} Thermostat
    sensor: temp
    default_target_temperature_high: ${TARGET_TEMP}
    id: therm

    cool_action:
      - script.execute: on_cool
    idle_action:
      - script.execute: on_idle
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

I'm currently using the dev branch of esphome, as it supports single-set-point in the bang bang climate platform. This may not be a visible problem yet, but could possibly affect the next release.

glmnet, 2020-07-01:
We are having some bad weather regarding climate devices

bang_bang changed the behavior a few days ago but we rolled back that change and we are soon to introduce a new climate called thermostat.

maybe @kbx81 can follow this issue better.

kbx81, 2020-07-02:
@jptrsn the new `thermostat` component was just merged in today -- since you are wanting to control an AC system, I'd suggest giving this a try and let us know if the behavior is more in line with what you are expecting. Still, I believe this is an issue with the `bang_bang` component and we will work on a fix.

jptrsn, 2020-07-03:
I'm now using the new `thermostat` component, and it's working as I'd expected. However, I've run into an edge-case that seems to require an event that I can't hook into.

Steps to reproduce:
1. Set the climate mode to `COOL`, and the fan mode to `ON`. Allow the thermostat (climate action) to go from `COOLING` to `OFF`. This will result in the fan running, but no active cooling (i.e. the compressor is not running).
> [23:39:33][D][climate:010]: 'AC 01 Thermostat' - Setting
[23:39:33][D][climate:014]:   Mode: COOL
[23:39:33][D][climate:262]: 'AC 01 Thermostat' - Sending state:
[23:39:33][D][climate:265]:   Mode: COOL
[23:39:33][D][climate:267]:   Action: OFF
[23:39:33][D][climate:270]:   Fan Mode: ON
[23:39:33][D][climate:276]:   Current Temperature: 25.35°C
[23:39:33][D][climate:282]:   Target Temperature: 25.50°C
2. Change the climate mode to `OFF`. Since there is no event handler that I can hook into for this event (the change from idle to off), I can't turn off the fan in this scenario.
> [23:45:42][D][climate:010]: 'AC 01 Thermostat' - Setting
[23:45:42][D][climate:014]:   Mode: OFF
[23:45:42][D][climate:262]: 'AC 01 Thermostat' - Sending state:
[23:45:42][D][climate:265]:   Mode: OFF
[23:45:42][D][climate:267]:   Action: OFF
[23:45:42][D][climate:270]:   Fan Mode: ON
[23:45:42][D][climate:276]:   Current Temperature: 25.43°C
[23:45:42][D][climate:282]:   Target Temperature: 25.50°C
3. This is the inverse of the original issue - when changing `MODE` from `OFF` to `COOL` but the current temperature does not trigger the device to go into active cooling mode, no event handler can be set.

I think there needs to be an event handler that we can use to trigger actions for mode changes that do not trigger action or fan state changes.

kbx81, 2020-07-03:
@jptrsn if I am understanding you correctly, you want the fan mode to change to off when you set the climate mode to off. Is that correct? Could you use the `off_mode` trigger for this purpose? See https://next.esphome.io/components/climate/thermostat.html#heating-and-cooling-modes

jptrsn, 2020-07-03:
THIS IS EXACTLY WHAT I FAILED TO READ BEFORE POSTING! 😄 

Thanks so much for pointing out those triggers. It's the piece of the puzzle I couldn't figure find.

jptrsn, 2020-07-03:
I think the original issue persists, though.

closed 2020-12-05 by stale[bot]