esphome/issues #1318: gate_pin option on ac_dimmer component doesn't honor pin schema

tracestep, 2020-07-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass OS 4.10 (Home Assistant 0.111.4) on Virtualbox x86_64, python 3.7.7, using ESPHome dev 1.14.5

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP01 1MB flash board

**Affected component:**

`ac_dimmer` component

**Description of problem:**
ac_dimmer does not honor `inverted: yes` option for `gate_pin`: pin is high when dimmer is off and low when at 100% birghtness, which is expected behavior with `inverted: yes` option. But when dimmer is between 1 and 254 the pulse is active high (same behavior as with `inverted: no`). This result in the load being 100% on independent of the brightness adjustment.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP8266
  board: esp01_1m

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO2
    method: leading_pulse
    min_power: 0%
    init_with_half_cycle: false
    inverted: yes
    zero_cross_pin:
      number: GPIO1
      mode: INPUT_PULLUP
      inverted: no

light:
  - platform: monochromatic
    output: dimmer1
    name: Dimmer Light
    gamma_correct: 2
```

**Logs (if applicable):**

not applicable

**Additional information and things you've tried:**
Scope capture attached - brightness at ~80 (32%); zero-cross circuit with SFH620 (emitter to GND, collector on GPIO1). MOC3021 triac driver (anode to +3.3V via 330R resistor, cathode to GPIO2).

[image]

glmnet, 2020-07-01:
if you're trying to invert gate pin you should write something like:

```yaml
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: 
      number: GPIO2
      inverted: yes
    method: leading_pulse
    min_power: 0%
    init_with_half_cycle: false
    zero_cross_pin:
      number: GPIO1
      mode: INPUT_PULLUP
      inverted: no
````

tracestep, 2020-07-01:
Well, I feel silly... It works now. I was under the impression I had already tried that at some point and that it didn't work. Thanks again for the answer! Closing.

closed 2020-07-01 by tracestep

glmnet, 2020-07-01:
Could you share your project details and performance? Like if you’re using a commercial dimmer hardware or what kind of device (led light?) are you testing?

tracestep, 2020-07-01:
So far I'm still in a very early testing stage. Everything is simply on a breadboard, no commercial modules (apart from the ESP01 or other ESP8266 / ESP32 modules), and I am testing on a plain old incandescent bulb for now. The circuit is as simple as can be with opto-isolation, not even snubbers for now, and it's something like that (just pretend the PIC is the ESP module):
[image]
I am having better results without the R3 resistor, since ESP has pullups, and now with the "inverted" logic the connections on the MOC3021 will be inverted (and on 3.3V I have reduced R9 and R10 to 330R).

glmnet, 2020-07-01:
I decided to stop playing with mains for a while, my latest test where with an halogen 12V lamp which had an electronic dimmable transformer (Input dimming on AC 220V). It blew up, the transformer and the triac. I would like to guess that it was not a software issue, but who knows.

Well, may be you already checked the thread https://github.com/esphome/esphome/pull/880 we went thru several iterations of the code, adding features which are arguably useful.

Yellow probe seems to be ticking every 8.33 ms which is right for a 60Hz mains
I didn't know of a device like the SFH6206, 50% of the problems seems to be the ZC detector and 50% the triac / driver.
I don't understand how the device works, the antiparallel diodes there seems magical.

What is the blue probe showing?

tracestep, 2020-07-01:
I also use a 12V halogen with an electromagnetic (conventional) 110-12V transformer in my tests, no problem so far. It is possible that the electronic transformer, even though dimmable, didn't like the chopped waveform (maybe already defective) and shorted something, blowing the triac. 

Yes, I went through the evolution of the component; last year when I got started with Home Assistant there was none, I don't know why it took so long, it is such a perfect fit for a ESP-01 :-).

The blue waveform is the output (collector) of the SFH620 (it is not a SFH6206, I used that part in the schematic because it had the right footprint on Eagle). The antiparallel diodes are neede to catch both positive and negative halves of the waveform; also one clamps the other on each half-cycle, so excess reverse voltage will not kill them. If you used an optocoupler with a single diode you would need a bridge rectifier on the input.

If the `ac_dimmer` component had an option to use a simulated open-collector output, low on active, hi-Z (input) on idle, one could potentially even forego the optocouplers and drive the triac directly, like on the following circuit:

[image]

I am reluctant to submit a feature request at this point, because since the current drawn by the ESP chip is too much for a transformerless design like that, the use case is limited as you would be only saving a few optocouplers. It might be more advantageous because of potential board space saving. Once I learn more about the ESP component building I might even submit a patch.

This circuit was inspired by Microchip Application Notes [AN958](https://www.microchip.com/wwwAppNotes/AppNotes.aspx?appnote=en021266) and [AN954A](https://ww1.microchip.com/downloads/en/appnotes/00954a.pdf). Notice how VDD is connected to the AC live, and zero cross pin (GP2) is connected to AC neutral through a 10M resistor. This is not be a problem on PIC microcontrollers because they have clamping diodes on the IO pins, and provides for a sharper and shorter zero cross inverted-pulse. I guess ESPs have similar protection. And not having isolation from mains is not a problem on wireless circuits (as long as you connect nothing else to them). Anyway, if anyone were to use that, the usual disclaimers apply; proceed at your own risk.