esphome/issues #1320: Ultrasonic Distance Sensor issues

royasulin, 2020-07-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 0.111.4
esphome: 1.14.5

**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP8266
 board: nodemcuv2 

**Affected component:**

https://esphome.io/components/sensor/ultrasonic.html

**Description of problem:**
the ultrasonic ( hc-sr04)  sensor is reporting a fixed value and not updating even if we change the update_interval parameter 
Tested on two different boards and two different HC-SR04. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

- platform: ultrasonic
    trigger_pin: D1
    echo_pin: D2
    name: "level in cm"
    update_interval: 3s
    filters:
    # Replace the 0.42 by the height of your container. From the sensor to the bottom.
    # I multiplied by 100 in order to get CM since the sensor works in meters
    - lambda: return (0.42-x)*100;
    unit_of_measurement: "cm"

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

royasulin, 2020-07-02:
https://github.com/esphome/issues/issues/147  This is a similar issue but the solution provided here was a fix that should be already in the later release of ESPhome

rwifall, 2020-07-03:
How is your hc-sr04 sensor wired?  (Because it works for me, although I'm using D5 as trigger and D6 as echo)

closed 2020-07-06 by royasulin

jcallaghan, 2020-07-31:
I'm seeing the exact same issue with m hc-sr04. I've tried the pins as suggested above but I get no value back from the sensor. I have one working perfectly with 1.14.3 but using latest build and dev I can't seem to get a value from the sensor. I've tried different ESP32 boards as well as different HC-SR04 sensors too.

```
[08:53:30][D][ultrasonic.sensor:030]: 'esph_height_distance' - Got distance: 0.00 m
[08:53:30][D][sensor:092]: 'esph_height_distance': Sending state 0.00034 m with 2 decimals of accuracy
```

```
  - platform: ultrasonic
    name: "esph_${system_name}_distance"
    trigger_pin: GPIO4
    echo_pin: GPIO5
    update_interval: 30s
    timeout: 2m
```

Any ideas?

cho0p, 2020-08-17:
I am facing the same issue as jcallaghan. I've read a lot of different forum and blog posts but I cannot figure out what's wrong. I use a AZDelivery ESP32 Dev Kit C V4 NodeMCU and the Ultrasonic HC-SR04 sensor (5V input but 3,3V Signal Output to the ESP32) but I just see values of 0.000XX cm.

Home Assistant 0.113.3 (Docker)
Tested ESPHome Addon 1.14.4 as well as 1.15.0b3 (same behavious)

My configuration:

esphome:
  name: esp32_wassertank_fuellstand
  platform: ESP32
  board: nodemcu-32s

sensor:
  - platform: ultrasonic
    trigger_pin: GPIO5
    echo_pin: GPIO18
    name: "Ultrasonic Sensor"
    update_interval: 1s
    timeout: 3.0m
    accuracy_decimals: 2    
    unit_of_measurement: "cm"

In Home Assistant the sensor sticks to 0.00 cm. If I listen to the log of the ESP32 using a USB connection and the ESPHome Flasher Tool I can see the following all the time (no matter if something is directly in front of the sensor):

[23:13:26][D][sensor:092]: 'Ultrasonic Sensor': Sending state 0.00017 cm with 2 decimals of accuracy
[23:13:27][D][ultrasonic.sensor:030]: 'Ultrasonic Sensor' - Got distance: 0.00 m
[23:13:27][D][sensor:092]: 'Ultrasonic Sensor': Sending state 0.00017 cm with 2 decimals of accuracy
[23:13:28][D][ultrasonic.sensor:030]: 'Ultrasonic Sensor' - Got distance: 0.00 m
[23:13:28][D][sensor:092]: 'Ultrasonic Sensor': Sending state 0.00017 cm with 2 decimals of accuracy
[23:13:29][D][ultrasonic.sensor:030]: 'Ultrasonic Sensor' - Got distance: 0.00 m
[23:13:29][D][sensor:092]: 'Ultrasonic Sensor': Sending state 0.00017 cm with 2 decimals of accuracy
[23:13:30][D][ultrasonic.sensor:030]: 'Ultrasonic Sensor' - Got distance: 0.00 m
[23:13:30][D][sensor:092]: 'Ultrasonic Sensor': Sending state 0.00017 cm with 2 decimals of accuracy
[23:13:31][D][ultrasonic.sensor:030]: 'Ultrasonic Sensor' - Got distance: 0.00 m

It also randomly switches the value but it is always 0.000xx cm. 

I have tested different GPIOs - currently using 5 and 18. The Ultrasonic sensor itself works - if I switch to a Arduino Uno V3 I can perfectly measure the correct distance with a simple sketch. 

Any help or ideas are highly appreciated. I can immediately test it.

cho0p, 2020-08-17:
@jcallaghan I probably found a solution just after finishing my comment -.-

I just changed the power source/supply for the sensor. I used a USB port from my laptop (5V!) for the sensor and I saw the 0.000xx in the log. I switched to a USB port from my Desktop PC and suddenly it works:

[23:29:14][D][sensor:092]: 'Ultrasonic Sensor': Sending state 1.79269 cm with 2 decimals of accuracy
[23:29:15][D][ultrasonic.sensor:030]: 'Ultrasonic Sensor' - Got distance: 1.81 m
[23:29:15][D][sensor:092]: 'Ultrasonic Sensor': Sending state 1.80538 cm with 2 decimals of accuracy
[23:29:16][D][ultrasonic.sensor:030]: 'Ultrasonic Sensor' - Got distance: 1.79 m

I can reproduce the issue by just switching the USB ports (for the sensor power suppley). Both ports deliver 5V so I don't really know what the problem is but at least it seems like it solves my issue. Maybe it's worth to give it a try.

multipolygon, 2021-02-13:
I also have this exact issue

nickizgr8, 2021-07-29:
I am having the exact same issue and my sensor is actually reporting the same value too, ".00034...". I am using an ESP32-Cam board, GPIO13 for Echo and GPIO12 for Trigger. I have the sensor connected to the 3.3v pin on the esp32 cam and the cam board itself is powered by the 5v pin. The camera is working and viewable from home assistant but the distance values are reporting a fixed value of .00034m. I have tried 2 different esp-32 cam boards and 3 separate HC-SR04 sensors. what's frustrating is I had this working on breadboard. In testing I initially connected sensor to 5v and was getting timeout error and switched to 3.3 after reading to do so online. This resulted in accurate readings. I soldered everything into my final project and have been getting these bad values ever since. Of course the first time I connected everything to the sensor I held the connector backwards (3.3v=gnd, echo=trig, trig=echo, gnd=3.3v) so when I started receiving these results I figured I cooked the esp32-cam somehow so I changed the pin scheme to GPIO14 for Echo and GPIO15 for Trigger, same results. Frustrated I unsoldered everything and wired up to a fresh-unopened esp32-cam board. I re-flashed with the original code using GPIO13=Echo and GPIO12=Trigger, 5v into the esp-32 cam board, sensor powered via the esp32's 3.3v pin, SAME RESULT. I had checked my wiring 20x's by this point and finally searched the web for answers which brought me here and low-and-behold @jcallaghan and @cho0p were getting the exact same values. Did you guys find a solution to this? I did try connecting everything to the same power supply I initially had it all working on when I was breadboarding, same result. Any insight would be much appreciated, Tonight I might try powering the sensor from 5v and then putting a 1k resistor from the echo to pin 13 and see if that makes a difference, but everything I've read says that powering with 3.3v shouldn't be a problem for this sensor (and I had it working with 3.3v previously!).

davet2001, 2021-08-17:
I am having very similar problems.  My doit_esp32_devkit_v1 always reports no response from sensor.
On an oscilloscope I see trigger (about 20us) and echo (several 100us), reaching the pins of the ESP32 board, but no signal recognised by esphome.

I tried re-assigning the echo pin, then setting up the original echo pin as a GPIO switch - worked totally fine detecting on off transitions when I shorted it to 3.3V (so confident that the GPIO is not broken).

My current approach is to try tasmota - if it works on the same hardware then it suggests an esphome code problem.