esphome/issues #1323: xiaomi_hhccjcy01 missing battery_level

CODeRUS, 2020-07-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** esphome-1.15.0-dev 

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32

**Affected component:** https://next.esphome.io/components/sensor/xiaomi_ble.html?highlight=ble#hhccjcy01

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: xiaomi_hhccjcy01
    battery_level:
      name: "Xiaomi HHCCJCY01 Battery Level"
```

**Logs (if applicable):**

```
[battery_level] is an invalid option for [sensor.xiaomi_hhccjcy01]. Please check the indentation.

```

**Additional information and things you've tried:**
https://github.com/esphome/esphome/commit/3fba3a5e2e812485a49dd7cf38aed7e1ff403c94

JacobD10, 2020-08-08:
I'm also affected by this. It looks like it got removed in this commit: https://github.com/esphome/esphome/pull/1027/commits/b2723830f4ed1e29611e3688a2c9323c1071bc23

SteveEdson, 2020-09-16:
Same issue here. Worked fine on 1.14.5, but on 1.15.1 I get the validation error. It's still showing as valid in the docs though https://esphome.io/components/sensor/xiaomi_ble.html?highlight=xiaomi_hhccjcy01

dckiller51, 2020-11-01:
Hello, for now I am using this gateway. I am in firmware 3.2.2. I have the battery level with this link. If that can help.
[https://github.com/zewelor/bt-mqtt-gateway/blob/master/workers/miflora.py](url)

`2020-11-01 23:51:35,268 DEBUG bt-mqtt-gw.workers.miflora miflora.py:109:status_update - Updating miflora device 'anthurium_salon' (c4:7c:8d:6a:ca:fb)
2020-11-01 23:51:38,883 DEBUG bt-mqtt-gw.workers_manager workers_manager.py:68:execute - Execution result of command MifloraWorker.status_update: [{'topic': 'miflora/anthurium_salon/temperature', 'payload': '22.0'}, {'topic': 'miflora/anthurium_salon/moisture', 'payload': '60'}, {'topic': 'miflora/anthurium_salon/light', 'payload': '147'}, {'topic': 'miflora/anthurium_salon/conductivity', 'payload': '1556'}, {'topic': 'miflora/anthurium_salon/battery', 'payload': '100'}, {'topic': 'miflora/anthurium_salon/low_battery', 'payload': 'OFF'}]`

javicalle, 2020-12-14:
Fixed in 1.15.2:
https://github.com/esphome/esphome/pull/1288

glmnet, 2020-12-14:
Thanks @javicalle, closing

closed 2020-12-14 by glmnet