esphome/issues #1325: UART rx does not work on GPIO3 on ESP32 (with serial logger turned off)

nkitanov, 2020-07-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 wemos d1 mini compatible board like [these](https://de.aliexpress.com/item/4000195928480.html?spm=a2g0s.13010208.0.0.4e4byz7Ryz7RxP&gps-id=pay_success&scm=1007.13409.84631.0&scm_id=1007.13409.84631.0&scm-url=1007.13409.84631.0&pvid=24a0a936-3aee-4b1a-a122-15a7a25f029d).

**Affected component:**
https://esphome.io/components/uart.html

**Description of problem:**
Not getting UART rx data on the default GPIO3 with uart logger disabled. GPIO3 is where the onboard usb-uart converter is connected. Exactly the same yaml configuration works on esp8266 wemos d1 mini. If the rx pin changes to another it works. 

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: esp32_test
  platform: ESP32
  board: wemos_d1_mini32

wifi:
  networks: !include wifi/wifi.yaml

logger:
  baud_rate: 0

uart:
  rx_pin: GPIO3
  baud_rate: 9600
  id: pms

sensor:
    - platform: pmsx003
      type: PMSX003
      uart_id: pms
      pm_1_0:
        name: "PM 1 Concentration"
        id: pm1
      pm_2_5:
        name: "PM 2.5 Concentration"
        id: pm2_5
      pm_10_0:
        name: "PM 10 Concentration"
        id: pm10
```

**Logs (if applicable):**

```
none

```

**Additional information and things you've tried:**
Tried with different board, checking physical connection between the pin and board pin to rule out PCB issues - the same.
The same configuration on esp8266 works without issues, I get data from the sensor.
If I reconfigure uart rx to gpio5 - no issues, it works, so somehow it's related only to gpio3.
To rule out issues from serial converter connected directly gpio3 with gpio5 when rx is gpio5 and it continues to receive data so it's not the from cp210x serial converter. 

Checking documentation of ESP32 and could not find a reason why gpio3 is not receiving data. Theoretically there should be no issues. 

I wanted to test it with a simple sketch in platformio if gpio3 receives but need to change uart pins for the logging and I do not have right now with me UART to USB converter. Can do that after 2 weeks when I'm back home.

glmnet, 2020-07-03:
which ESPHome version are you using?

nkitanov, 2020-07-03:
> which ESPHome version are you using?

It's 1.14.4 - the latest one.

glmnet, 2020-07-05:
I tested with a DFPlayer module and found no issues, tested 1.14.5 and dev versions.

you could try adding a pull up or pull down on the line

nkitanov, 2020-07-05:
> I tested with a DFPlayer module and found no issues, tested 1.14.5 and dev versions.
> 
> you could try adding a pull up or pull down on the line

So you tried using UART RX pin GPIO3 and it works? It's not pull/up-down as I connected directly GPIO3 with 5 and on 5 it's working. So if there is any issue with pulling up-down before the pin it won't work on gpio5 either.

glmnet, 2020-07-06:
https://github.com/glmnet/esphome_devices/blob/master/test.issue_1325.yaml

I got this working

The DF player is powered by 3.3v of the node MCU regulator

glmnet, 2020-07-06:
[image]

nkitanov, 2020-07-06:
Thanks for testing. I am not familiar with dfplayer but as far as I see from the config (correct me if I'm wrong) you are using only TX pin to control the module and not RX to get any data from the module. 
```For communication you can connect only the tx_pin of the uart bus to the module’s RX but if you need feedback of playback active you will also need to connect the rx_pin to the module’s TX.```

P.S. I see now you have binary sensor, this should be using RX pin. Do you get this sensor correctly showing when it's playing?
```binary_sensor:
- platform: template
  name: "dfplayer_is_playing"
  lambda: 'return id(my_df_player).is_playing();'```

I can test with nodemcu when I'm back home also, but still I do not beleive the circuit is too much different since it's the same module ESP32-WROOM32.

glmnet, 2020-07-06:
Yes I wrote the dfplayer component.  

It’s receiving data properly

glmnet, 2020-07-06:
I guess I can test with the GPS too, that one sends data so I can better mimic your config without TX pin

glmnet, 2020-07-06:
GPS Works too

```yaml

uart:
  rx_pin: RX
  baud_rate: 9600
  id: pms

# Declare GPS module
gps:

# GPS as time source
time:
  - platform: gps
    id: gps_time

interval:
  - interval: 5s
    then:
      - lambda: |-
          char buffer [80];
          tm gps_time_tm = id(gps_time).now().to_c_tm();
          strftime (buffer,80,"Time is %c", &gps_time_tm);
          ESP_LOGI("TEST", buffer);
```

nkitanov, 2020-07-11:
Thank you. I will do some tests when I have time with nodemcu and my board and update the issue.

nkitanov, 2020-07-14:
Hi again. Issue solved, it's not related to esphome. I tested first with nodemcu32 board and indeed it's working. Then I decided that something is wrong with the board, I remembered that I have a simple multimeter in the car, pick it up, and found that the pinout diagram of these "WEMOS D1 Mini ESP32 Minikit" is wrong! They reversed the uart pins - TXD goes to GPIO3 and RXD to GPIO1, not like it's described on the pinout picture. Luckily it's ESP32 so no issues - I just selected the correct GPIO1 in the yaml instead of GPIO3 and it's working now. :) 

Thanks one again @glmnet for your help! Unfortunately I lost some hours of my time and yours, but at least I know now these pinout pictures are not always correct.

closed 2020-07-14 by nkitanov

glmnet, 2020-07-14:
Ahh yes. I have an ESP32 that says GND in one of the SPI flash pins. 🤦‍♂️