esphome/issues #1326: Cant send SMS with sim800l with esp32

delta010, 2020-07-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

 Esp32 (TTGO T-Call ESP32 with SIM800L)

**ESPHome version (latest production, beta, dev branch)**

1.14.4 

**Affected component:**

https://esphome.io/components/sim800l.html

**Description of problem:**
I am  using TTGO T-Call ESP32 with SIM800L for sending and receiving SMS . when I send an SMS to my phone via home assistant service nothing happens the ESphome log says "Sending to +XXXXXXXXXx: Hello World!" but I don't receive any SMS

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: esp32gsm
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "xXXXXXXXXX"
  password: "xxxxxxxxxx"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.217
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
    dns1: 192.168.1.1
ota:

# Enable logging
logger:

# Enable Home Assistant API
web_server:
  port: 80
  
i2c:  
  sda: GPIO21
  scl: GPIO22
  scan: True
  
# Enable Home Assistant API

api:
  services:
  - service: send_sms
    variables:
      recipient: string
      message: string
    then:
    - sim800l.send_sms:
        recipient: !lambda 'return recipient;'
        message: !lambda 'return message;'

uart:
  baud_rate: 9600
  tx_pin: TX
  rx_pin: RX
sim800l:
  on_sms_received:
    - lambda: |-
        id(sms_sender).publish_state(sender);
        id(sms_message).publish_state(message);
    - logger.log:
        format: "Received '%s' from %s"
        args: [ 'message.c_str()', 'sender.c_str()' ]

text_sensor:
- platform: template
  id: sms_sender
  name: "Sms Sender"
- platform: template
  id: sms_message
  name: "Sms Message"
    
time:
  - platform: sntp
    id: timer
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 2.pool.ntp.org

switch:
  - platform: gpio
    name: "SIM800_PWKEY"
    pin: 4
    restore_mode: ALWAYS_OFF
  - platform: gpio
    name: "SIM800_RST"
    pin: 5
    restore_mode: ALWAYS_ON
  - platform: gpio
    name: "SIM800_POWER"
    pin: 23
    restore_mode: ALWAYS_ON
```

**Logs (if applicable):**

```
[13:05:36][C][wifi:415]: WiFi:
[13:05:36][C][wifi:283]:   SSID: 

[redacted]
[13:05:36][C][wifi:284]:   IP Address: 192.168.1.217
[13:05:36][C][wifi:286]:   BSSID: 

[redacted]
[13:05:36][C][wifi:287]:   Hostname: 'ttgo_t_call_esp32'
[13:05:36][C][wifi:291]:   Signal strength: -17 dB ▂▄▆█
[13:05:36][C][wifi:295]:   Channel: 11
[13:05:36][C][wifi:296]:   Subnet: 255.255.255.0
[13:05:36][C][wifi:297]:   Gateway: 192.168.1.1
[13:05:36][C][wifi:298]:   DNS1: 192.168.1.1
[13:05:36][C][wifi:299]:   DNS2: 0.0.0.0
[13:05:36][C][uart:040]: UART Bus:
[13:05:36][C][uart:042]:   TX Pin: GPIO1
[13:05:36][C][uart:045]:   RX Pin: GPIO3
[13:05:36][C][uart:047]:   Baud Rate: 9800 baud
[13:05:36][C][uart:048]:   Stop bits: 1
[13:05:36][C][template.text_sensor:020]: Template Sensor 'Sms Sender'
[13:05:36][C][template.text_sensor:020]: Template Sensor 'Sms Message'
[13:05:36][C][switch.gpio:042]: GPIO Switch 'SIM800_PWKEY'
[13:05:36][C][switch.gpio:043]:   Pin: GPIO4 (Mode: OUTPUT)
[13:05:36][C][switch.gpio:059]:   Restore Mode: Always OFF
[13:05:36][C][switch.gpio:042]: GPIO Switch 'SIM800_RST'
[13:05:36][C][switch.gpio:043]:   Pin: GPIO5 (Mode: OUTPUT)
[13:05:36][C][switch.gpio:059]:   Restore Mode: Always ON
[13:05:36][C][switch.gpio:042]: GPIO Switch 'SIM800_POWER'
[13:05:36][C][switch.gpio:043]:   Pin: GPIO23 (Mode: OUTPUT)
[13:05:36][C][switch.gpio:059]:   Restore Mode: Always ON
[13:05:36][C][logger:175]: Logger:
[13:05:36][C][logger:176]:   Level: DEBUG
[13:05:36][C][logger:177]:   Log Baud Rate: 0
[13:05:36][C][logger:178]:   Hardware UART: UART0
[13:05:36][C][sim800l:259]: SIM800L:
[13:05:36][C][sim800l:260]:   RSSI: 0 dB
[13:05:36][C][web_server:123]: Web Server:
[13:05:36][C][web_server:124]:   Address: 192.168.1.217:80
[13:05:36][C][ota:029]: Over-The-Air Updates:
[13:05:36][C][ota:030]:   Address: 192.168.1.217:3232
[13:05:36][C][api:095]: API Server:
[13:05:36][C][api:096]:   Address: 192.168.1.217:6053
[13:05:36][C][sntp:043]: SNTP Time:
[13:05:36][C][sntp:044]:   Server 1: '0.pool.ntp.org'
[13:05:36][C][sntp:045]:   Server 2: '1.pool.ntp.org'
[13:05:36][C][sntp:046]:   Server 3: '2.pool.ntp.org'
[13:05:36][C][sntp:047]:   Timezone: 'IST-5:30'
[13:05:38][D][api.connection:583]: Client 'Home Assistant 0.111.0 (192.168.1.5)' connected successfully!
[13:07:31][I][ota:046]: Boot seems successful, resetting boot loop counter.
[13:07:59][D][switch:021]: 'SIM800_PWKEY' Turning ON.
[13:07:59][D][switch:045]: 'SIM800_PWKEY': Sending state ON
[13:08:07][D][sim800l:252]: Sending to +919744019818: Hello World!

```

**Additional information and things you've tried:**

rradar, 2020-07-04:
Could you please provide a log with `very_verbose` @delta010 ?

Can you conform the the same sim800l works with a esp8266 and doesn't work with another esp32?

delta010, 2020-07-04:
Log with very_verbos
`
INFO Reading configuration /config/esphome/ttgo_t_call_esp32.yaml...
INFO Detected timezone 'IST' with UTC offset 5:30
INFO Starting log output from 192.168.1.217 using esphome API
INFO Connecting to 192.168.1.217:6053 (192.168.1.217)
INFO Successfully connected to 192.168.1.217
[21:38:39][I][app:100]: ESPHome version 1.14.4 compiled on Jul  4 2020, 21:35:31
[21:38:39][C][wifi:415]: WiFi: xxxxxxx
[21:38:39][C][wifi:283]:   SSID: xxxxxxxx

[redacted]
[21:38:39][C][wifi:284]:   IP Address: 192.168.1.217
[21:38:39][C][wifi:286]:   BSSID: xxxxxxxxxx

[redacted]
[21:38:39][C][wifi:287]:   Hostname: 'ttgo_t_call_esp32'
[21:38:39][C][wifi:291]:   Signal strength: -18 dB ▂▄▆█
[21:38:39][V][wifi:293]:   Priority: 0.0
[21:38:39][C][wifi:295]:   Channel: 11
[21:38:39][C][wifi:296]:   Subnet: 255.255.255.0
[21:38:39][C][wifi:297]:   Gateway: 192.168.1.1
[21:38:39][C][wifi:298]:   DNS1: 192.168.1.1
[21:38:39][C][wifi:299]:   DNS2: 0.0.0.0
[21:38:39][C][uart:040]: UART Bus:
[21:38:39][C][uart:042]:   TX Pin: GPIO26
[21:38:39][C][uart:045]:   RX Pin: GPIO27
[21:38:39][C][uart:047]:   Baud Rate: 9800 baud
[21:38:39][C][uart:048]:   Stop bits: 1
[21:38:39][C][template.text_sensor:020]: Template Sensor 'Sms Sender'
[21:38:39][C][template.text_sensor:020]: Template Sensor 'Sms Message'
[21:38:39][C][switch.gpio:042]: GPIO Switch 'SIM800_PWKEY'
[21:38:39][C][switch.gpio:043]:   Pin: GPIO4 (Mode: OUTPUT)
[21:38:39][C][switch.gpio:059]:   Restore Mode: Always OFF
[21:38:39][C][switch.gpio:042]: GPIO Switch 'SIM800_RST'
[21:38:39][C][switch.gpio:043]:   Pin: GPIO5 (Mode: OUTPUT)
[21:38:39][C][switch.gpio:059]:   Restore Mode: Always ON
[21:38:39][VV][uart:070]:     Read 0b00000000 (0x00)
[21:38:39][VV][sim800l:230]: Buffer pos: 1 0
[21:38:39][C][switch.gpio:042]: GPIO Switch 'SIM800_POWER'
[21:38:39][C][switch.gpio:043]:   Pin: GPIO23 (Mode: OUTPUT)
[21:38:39][C][switch.gpio:059]:   Restore Mode: Always ON
[21:38:39][C][logger:175]: Logger:
[21:38:39][C][logger:176]:   Level: VERY_VERBOSE
[21:38:39][C][logger:177]:   Log Baud Rate: 115200
[21:38:39][C][logger:178]:   Hardware UART: UART0
[21:38:39][C][sim800l:259]: SIM800L:
[21:38:39][C][sim800l:260]:   RSSI: 0 dB
[21:38:39][C][web_server:123]: Web Server:
[21:38:39][C][web_server:124]:   Address: 192.168.1.217:80
[21:38:39][C][ota:029]: Over-The-Air Updates:
[21:38:39][C][ota:030]:   Address: 192.168.1.217:3232
[21:38:39][C][api:095]: API Server:
[21:38:39][C][api:096]:   Address: 192.168.1.217:6053
[21:38:39][C][sntp:043]: SNTP Time:
[21:38:39][C][sntp:044]:   Server 1: '0.pool.ntp.org'
[21:38:39][C][sntp:045]:   Server 2: '1.pool.ntp.org'
[21:38:39][C][sntp:046]:   Server 3: '2.pool.ntp.org'
[21:38:39][C][sntp:047]:   Timezone: 'IST-5:30'
[21:38:40][VV][scheduler:131]: Running interval 'update' with interval=5000 last_execution=128296 (now=133296)
[21:38:42][VV][uart:070]:     Read 0b00000000 (0x00)
[21:38:42][VV][sim800l:230]: Buffer pos: 2 0
[21:38:44][VV][api.service:250]: on_ping_request: PingRequest {}
[21:38:44][VV][api.service:035]: send_ping_response: PingResponse {}
[21:38:44][VV][scheduler:131]: Running interval 'update' with interval=5000 last_execution=133296 (now=138296)
[21:38:45][VV][uart:070]:     Read 0b00000000 (0x00)
[21:38:45][VV][sim800l:230]: Buffer pos: 3 0
[21:38:45][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=129289 (now=139289)
[21:38:48][VV][uart:070]:     Read 0b00000000 (0x00)
[21:38:48][VV][sim800l:230]: Buffer pos: 4 0
[... 80 lines cut ...]
[21:39:19][VV][uart:070]:     Read 0b01000001 (0x41)
[21:39:19][VV][sim800l:230]: Buffer pos: 6 65
[21:39:19][VV][uart:070]:     Read 0b01010100 (0x54)
[21:39:19][VV][sim800l:230]: Buffer pos: 7 84
[21:39:19][VV][uart:070]:     Read 0b00001010 (0x0A)
[21:39:19][VV][sim800l:230]: Buffer pos: 8 10
[21:39:19][V][sim800l:048]: R:  - 2
[21:39:20][VV][api.service:250]: on_ping_request: PingRequest {}
[21:39:20][VV][api.service:035]: send_ping_response: PingResponse {}
[21:39:20][VV][api.service:250]: on_ping_request: PingRequest {}
[21:39:20][VV][api.service:035]: send_ping_response: PingResponse {}
[21:39:22][VV][uart:070]:     Read 0b00000000 (0x00)
[21:39:22][VV][sim800l:230]: Buffer pos: 0 0
[21:39:24][VV][scheduler:131]: Running interval 'update' with interval=5000 last_execution=173296 (now=178296)
[21:39:25][VV][uart:070]:     Read 0b00000000 (0x00)
[21:39:25][VV][sim800l:230]: Buffer pos: 1 0
[21:39:25][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=169289 (now=179290)
[21:39:25][VV][api.service:250]: on_ping_request: PingRequest {}
[21:39:25][VV][api.service:035]: send_ping_response: PingResponse {}
[21:39:28][VV][uart:070]:     Read 0b00000000 (0x00)
[21:39:28][VV][sim800l:230]: Buffer pos: 2 0
[21:39:29][VV][scheduler:131]: Running interval 'update' with interval=5000 last_execution=178296 (now=183296) `

and this is my module
[image]

glmnet, 2020-07-04:
Ahh good thing you posted the module picture. 
Looks like serial and setup is not done properly

Try following this thread 

https://community.home-assistant.io/t/esp32-sim800l-ttgo-t-call-not-working/149781/15

delta010, 2020-07-04:
I already followed that thread but nothing changed ;

I also found something strange about the sim800l on the log

`[21:38:39][C][sim800l:259]: SIM800L:
[21:38:39][C][sim800l:260]: RSSI: 0 dB`

glmnet, 2020-07-04:
Per your verbose logs, the receiver is not getting valid data via serial.

Try changing this

````
uart:
  baud_rate: 9600
  tx_pin: TX
  rx_pin: RX
````

to this as in the thread:

````

Copy to clipboard
uart:
  baud_rate: 9600
  tx_pin: 27
  rx_pin: 26
````

delta010, 2020-07-06:
after I change the UART and I2C I got  "Registration Fail" error
`[09:12:19][W][sim800l:107]: Registration Fail
[09:12:19][VV][uart:070]:     Read 0b00001101 (0x0D)
[09:12:19][VV][sim800l:230]: Buffer pos: 0 13
[09:12:19][VV][uart:070]:     Read 0b00001010 (0x0A)
[09:12:19][VV][sim800l:230]: Buffer pos: 0 10
[09:12:19][V][sim800l:048]: R:  - 3
[09:12:19][VV][uart:070]:     Read 0b01001111 (0x4F)
[09:12:19][VV][sim800l:230]: Buffer pos: 0 79
[09:12:19][VV][uart:070]:     Read 0b01001011 (0x4B)
[09:12:19][VV][sim800l:230]: Buffer pos: 1 75
[09:12:19][VV][uart:070]:     Read 0b00001101 (0x0D)
[09:12:19][VV][sim800l:230]: Buffer pos: 2 13
[09:12:19][VV][uart:070]:     Read 0b00001010 (0x0A)
[09:12:19][VV][sim800l:230]: Buffer pos: 2 10
[09:12:19][V][sim800l:048]: R: OK - 3
[09:12:19][V][sim800l:041]: S: AT+CREG? - 3
[09:12:19][VV][uart:064]:     Wrote "AT+CREG?"
[09:12:19][VV][uart:054]:     Wrote 0b00001010 (0x0A)
[09:12:19][V][app:076]: A component took a long time in a loop() cycle (0.20 s).
[09:12:19][V][app:077]: Components should block for at most 20-30ms in loop().
[09:12:19][VV][uart:070]:     Read 0b00001101 (0x0D)
[09:12:19][VV][sim800l:230]: Buffer pos: 0 13
[09:12:19][VV][uart:070]:     Read 0b00001010 (0x0A)
[09:12:19][VV][sim800l:230]: Buffer pos: 0 10
[09:12:19][V][sim800l:048]: R:  - 4
[09:12:19][VV][uart:070]:     Read 0b00101011 (0x2B)
[09:12:19][VV][sim800l:230]: Buffer pos: 0 43
[09:12:19][VV][uart:070]:     Read 0b01000011 (0x43)`

glmnet, 2020-07-06:
Great. 
You can reduce log level now, comm with the module is working. 

Usually that error is because of some chips which are roaming. 

Can you try the development branch?

delta010, 2020-07-07:
changing to dev branch solves all the problems

closed 2020-07-11 by glmnet

delta010, 2020-08-07:
after changing to dev branch it works fine for a couple of weeks, but now it not working and it producing same error please help me

`
INFO Reading configuration /config/esphome/sms_gateway.yaml...
INFO Detected timezone 'UTC' with UTC offset 0
INFO Starting log output from 192.168.1.217 using esphome API
INFO Connecting to 192.168.1.217:6053 (192.168.1.217)
INFO Successfully connected to 192.168.1.217
[14:11:32][I][app:105]: ESPHome version 1.16.0-dev compiled on Aug  7 2020, 14:04:33
[14:11:32][C][wifi:443]: WiFi:
[14:11:32][C][wifi:303]:   SSID: 

[redacted]
[14:11:32][C][wifi:304]:   IP Address: 192.168.1.217
[14:11:32][C][wifi:306]:   BSSID: 

[redacted]
[14:11:32][C][wifi:307]:   Hostname: 'sms_gateway'
[14:11:32][C][wifi:311]:   Signal strength: -19 dB ▂▄▆█
[14:11:32][V][wifi:313]:   Priority: 0.0
[14:11:32][C][wifi:315]:   Channel: 9
[14:11:32][C][wifi:316]:   Subnet: 255.255.255.0
[14:11:32][C][wifi:317]:   Gateway: 192.168.1.1
[14:11:32][C][wifi:318]:   DNS1: 192.168.1.1
[14:11:32][C][wifi:319]:   DNS2: 0.0.0.0
[14:11:32][C][i2c:028]: I2C Bus:
[14:11:32][C][i2c:029]:   SDA Pin: GPIO21
[14:11:32][C][i2c:030]:   SCL Pin: GPIO22
[14:11:32][C][i2c:031]:   Frequency: 50000 Hz
[14:11:32][I][i2c:033]: Scanning i2c bus for active devices...
[14:11:32][I][i2c:040]: Found i2c device at address 0x75
[14:11:32][C][uart_esp32:088]: UART Bus:
[14:11:32][C][uart_esp32:090]:   TX Pin: GPIO26
[14:11:32][C][uart_esp32:093]:   RX Pin: GPIO27
[14:11:32][C][uart_esp32:094]:   RX Buffer Size: 256
[14:11:32][C][uart_esp32:096]:   Baud Rate: 9600 baud
[14:11:32][C][uart_esp32:097]:   Data Bits: 8
[14:11:32][C][uart_esp32:098]:   Parity: NONE
[14:11:32][C][uart_esp32:099]:   Stop bits: 1
[14:11:32][C][template.text_sensor:020]: Template Sensor 'Sms Sender'
[14:11:32][C][template.text_sensor:020]: Template Sensor 'Sms Message'
[14:11:32][C][switch.gpio:042]: GPIO Switch 'SIM800_PWKEY'
[14:11:32][C][switch.gpio:043]:   Pin: GPIO4 (Mode: OUTPUT)
[14:11:32][C][switch.gpio:059]:   Restore Mode: Always OFF
[14:11:32][C][switch.gpio:042]: GPIO Switch 'SIM800_RST'
[14:11:32][C][switch.gpio:043]:   Pin: GPIO5 (Mode: OUTPUT)
[14:11:32][C][switch.gpio:059]:   Restore Mode: Always ON
[14:11:32][C][switch.gpio:042]: GPIO Switch 'SIM800_POWER'
[14:11:32][C][switch.gpio:043]:   Pin: GPIO23 (Mode: OUTPUT)
[14:11:32][C][switch.gpio:059]:   Restore Mode: Always ON
[14:11:32][C][logger:185]: Logger:
[14:11:32][C][logger:186]:   Level: VERY_VERBOSE
[14:11:32][C][logger:187]:   Log Baud Rate: 115200
[14:11:32][C][logger:188]:   Hardware UART: UART0
[14:11:32][C][sim800l:259]: SIM800L:
[14:11:32][C][sim800l:260]:   RSSI: 0 dB
[14:11:32][C][web_server:131]: Web Server:
[14:11:32][C][web_server:132]:   Address: 192.168.1.217:80
[14:11:32][C][ota:029]: Over-The-Air Updates:
[14:11:32][C][ota:030]:   Address: 192.168.1.217:3232
[14:11:32][C][api:095]: API Server:
[14:11:32][C][api:096]:   Address: 192.168.1.217:6053
[14:11:32][C][sntp:043]: SNTP Time:
[14:11:32][C][sntp:044]:   Server 1: '0.pool.ntp.org'
[14:11:32][C][sntp:045]:   Server 2: '1.pool.ntp.org'
[14:11:32][C][sntp:046]:   Server 3: '2.pool.ntp.org'
[14:11:32][C][sntp:047]:   Timezone: 'UTC0'
[14:11:33][VV][uart_esp32:121]:     Read 0b00000000 (0x00)
[14:11:33][VV][sim800l:230]: Buffer pos: 3 0
[14:11:33][VV][api.service:220]: on_ping_request: PingRequest {}
[... 50 lines cut ...]
[14:11:52][VV][uart_esp32:121]:     Read 0b01010100 (0x54)
[14:11:52][VV][sim800l:230]: Buffer pos: 7 84
[14:11:52][VV][uart_esp32:121]:     Read 0b00001010 (0x0A)
[14:11:52][VV][sim800l:230]: Buffer pos: 8 10
[14:11:52][V][sim800l:048]: R:  - 2
[14:11:52][VV][api.service:220]: on_ping_request: PingRequest {}
[14:11:52][VV][api.service:032]: send_ping_response: PingResponse {}
[14:11:55][VV][uart_esp32:121]:     Read 0b00000000 (0x00)
[14:11:55][VV][sim800l:230]: Buffer pos: 0 0
[14:11:57][VV][scheduler:131]: Running interval 'update' with interval=5000 last_execution=352839 (now=357839)
[14:11:58][VV][api.service:220]: on_ping_request: PingRequest {}
[14:11:58][VV][api.service:032]: send_ping_response: PingResponse {}
[14:11:58][VV][uart_esp32:121]:     Read 0b00000000 (0x00)
[14:11:58][VV][sim800l:230]: Buffer pos: 1 0
[14:12:00][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=351425 (now=361426)
[14:12:01][VV][uart_esp32:121]:     Read 0b00000000 (0x00)
[14:12:01][VV][sim800l:230]: Buffer pos: 2 0
[14:12:02][VV][scheduler:131]: Running interval 'update' with interval=5000 last_execution=357839 (now=362839)
[14:12:03][VV][api.service:220]: on_ping_request: PingRequest {}
[14:12:03][VV][api.service:032]: send_ping_response: PingResponse {}
[14:12:04][VV][api.service:220]: on_ping_request: PingRequest {}
[14:12:04][VV][api.service:032]: send_ping_response: PingResponse {}
[14:12:04][VV][uart_esp32:121]:     Read 0b00000000 (0x00)
[14:12:04][VV][sim800l:230]: Buffer pos: 3 0
`

delta010, 2020-08-08:
while I send an SMS it producing this error
`
[04:13:12][D][sim800l:216]: Unhandled: OK - 0
[04:13:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:13:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:13:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:14:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:14:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:14:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:14:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:15:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:15:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:15:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:15:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:16:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:16:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:16:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:16:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:17:27][W][sim800l:062]: Not ack. 8  
[04:17:27][D][sim800l:216]: Unhandled: ERROR - 0
[04:17:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:17:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:18:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:18:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:18:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:18:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:19:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:19:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:19:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:19:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:20:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:20:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:20:55][W][sim800l:062]: Not ack. 8  
[04:20:55][D][sim800l:216]: Unhandled: ERROR - 0
[04:21:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:21:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:21:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:21:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:22:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:22:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:22:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:22:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:23:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:23:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:23:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:23:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:24:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:24:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:24:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:24:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:25:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:25:25][D][sim800l:196]: Sending message: 'Hello World!'
[04:25:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:25:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:26:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:26:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:26:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:26:54][D][sim800l:196]: Sending message: 'Hello World!'
[04:27:09][D][sim800l:196]: Sending message: 'Hello World!'
[04:27:24][D][sim800l:196]: Sending message: 'Hello World!'
[04:27:39][D][sim800l:196]: Sending message: 'Hello World!'
[04:27:54][D][sim800l:196]: Sending message: 'Hello World!'
`

glmnet, 2020-08-08:
Post full yaml again please.

delta010, 2020-08-08:
esphome yaml:

esphome:
  name: esp32gsm
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "WiFi"
  password: "*****"
  manual_ip:
    static_ip: 192.168.1.75
    gateway: 192.168.1.1
    subnet: 255.255.255.0

logger:

ota:

web_server:
  port: 80
  
i2c:  
  sda: GPIO21
  scl: GPIO22
  scan: True
  
api:
  services:
  - service: send_sms
    variables:
      recipient: string
      message: string
    then:
    - sim800l.send_sms:
        recipient: !lambda 'return recipient;'
        message: !lambda 'return message;'

uart:
  baud_rate: 9600
  tx_pin: 27
  rx_pin: 26

sim800l:
  on_sms_received:
    - lambda: |-
        id(sms_sender).publish_state(sender);
        id(sms_message).publish_state(message);
    - logger.log:
        format: "Received '%s' from %s"
        args: [ 'message.c_str()', 'sender.c_str()' ]

text_sensor:
- platform: template
  id: sms_sender
  name: "Sms Sender"
- platform: template
  id: sms_message
  name: "Sms Message"
    
time:
  - platform: sntp
    id: timer
    servers:
      - 0.pool.ntp.org
      - 1.pool.ntp.org
      - 2.pool.ntp.org

switch:
  - platform: gpio
    name: "SIM800_PWKEY"
    pin: 4
    restore_mode: ALWAYS_OFF
  - platform: gpio
    name: "SIM800_RST"
    pin: 5
    restore_mode: ALWAYS_ON
  - platform: gpio
    name: "SIM800_POWER"
    pin: 23
    restore_mode: `ALWAYS_ON`

delta010, 2020-08-10:
It receiving incoming messages without any problem but it cant send outgoing SMS