esphome/issues #1331: esp32_ble_tracker brings in xiaomi_ble Component, should it?

mdframe, 2020-07-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 20.x Server with Hass.io in a Docker Instance

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
I am working on a custom component to read bluetooth values for Govee temp/humidity devices.  I created a simple test project with nothing but the base YAML component, see below,  and then added esp32_ble_tracker to see what source and information would be generated automatically.  I found it odd that a specific device, xiaomi_ble, was generated with this default espHome.io component.  Is the current ble tracker component built specific for xiaomi ble devices?  Should I be basing my ble device reader on a different component?  

This will be my first attempt at an espHome custom component and using ble is a large learning curve so I am looking for others insight how I could build a component for espHome, future to add to the component list, that is not based on a current device that most likely will have no meaning for the manufacturer I am coding for.   If anyone can advise me with additional content beyond the current espHome custom component documentation please provide details.  I wish to develop a product that is usable beyond my personal scope and may be be a base for many Govee devices in the future for others. 

Thank you.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xxxSSIDxxx"
  password: "xxxpaswordxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "lnDlYus1CmN6"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

IsaacBreuer, 2020-07-26:
There is a homeassistant custom component that can read the govee temperature sensors, I wish we can port it to esphome
https://community.home-assistant.io/t/govee-ble-thermometer-hygrometer-sensor/166696

jimbo19922003, 2020-08-15:
Was any progress made on this? I would also be interested in a solution and cannot seem to get the Govee bluetooth to play nice with ESPHome.

thrasher204, 2020-08-16:
Same issue here. The Govee sensors are detected are recognized by name but i cannot find a way to get them connected to ESPHome

WhichWayWazzit, 2020-10-15:
I found a workaround without a requirement for new component in ESPhome. Posted about it in https://community.home-assistant.io/t/read-govee-temperature-humidity-ble-advertisements-with-esp32-and-esphome/230449/2

closed 2021-02-26 by stale[bot]