esphome/issues #1334: Problem using a custom switch

marianomd, 2020-07-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP-01s

**ESPHome version (latest production, beta, dev branch)**
1.15-dev

**Affected component:**
https://esphome.io/components/switch/custom.html
https://esphome.io/components/sensor/custom.html

**Description of problem:**
I'm trying to create a fading led switch, it is accepted well until I add it to on_press for example.

**Problem-relevant YAML-configuration entries:**
```yaml
switch:
  - platform: custom
    id: ledring
    lambda: |-
      auto fading_led = new FadingLed();
      App.register_component(fading_led);
      return {fading_led};
    switches:
      name: "Led Ring"

binary_sensor:
  - platform: gpio
    name: "Bot.n"
    id: button
    pin:
      number: 0
      inverted: False
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.turn_on: ledring
        - http_request.post:
            url: http://192.168.1.110/switch/ding_dong/turn_on
        - delay: 10s
        - switch.turn_off: ledring
    on_release:
      then:
        - http_request.post:
            url: http://192.168.1.110/switch/ding_dong/turn_off
```

**Logs (if applicable):**

```
λ esphome doorbutton.yaml run
INFO Reading configuration doorbutton.yaml...
Failed config

binary_sensor.gpio:
  platform: gpio
  name: Botón
  id: button
  pin:
    number: 0
    inverted: False
    mode: INPUT_PULLUP
  on_press:
    - then:
        - switch.turn_on:

            ID 'ledring' of type custom::CustomSwitchConstructor doesn't inherit from switch_::Switch. Please double check your ID is pointing to the correct value.
            id: ledring
        - http_request.post:
            url: http://192.168.1.110/switch/ding_dong/turn_on
            method: POST
            verify_ssl: True
        - delay: 10s
        - switch.turn_off:

            ID 'ledring' of type custom::CustomSwitchConstructor doesn't inherit from switch_::Switch. Please double check your ID is pointing to the correct value.
            id: ledring
  on_release:
    - then:
        - http_request.post:
            url: http://192.168.1.110/switch/ding_dong/turn_off
            method: POST
            verify_ssl: True
```

**Additional information and things you've tried:**

This is the component definition (fading_led.h)

```c++
#include "esphome.h"

class FadingLed : public PollingComponent, public Switch
{
public:
    // constructor
    FadingLed() : PollingComponent(50) {}
    int LED = 2;
    bool state = false;

    void setup() override
    {
        // LED - ROJO
        pinMode(LED, OUTPUT);
        publish_state(state);
    }

    void update() override
    {
        if (state)
        {
            float something = millis() / 3000.0;
            int value = 512.0 + 512 * sin(something * 2.0 * PI);
            analogWrite(LED, value);
        }
    }

    void write_state(bool state) override
    {
        this->state = state;

        digitalWrite(LED, 0);

        // Acknowledge new state by publishing it
        publish_state(state);
    }
};
```

glmnet, 2020-07-08:
You are giving the `ledring` id to the custom component, not the single switch, move the `id: ledring` line below the `name: "Led Ring" ` ...

```
switch:
  - platform: custom
    lambda: |-
      auto fading_led = new FadingLed();
      App.register_component(fading_led);
      return {fading_led};
    switches:
      name: "Led Ring"
      id: ledring
```

marianomd, 2020-07-08:
Thank you so much Guillermo. I'm closing, sorry for the pollution.

closed 2020-07-08 by marianomd