esphome/issues #1335: st7789v Display Resolution Hardcoded To 135x240

DACOFFEY, 2020-07-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10 + pip3 install https://github.com/esphome/esphome/archive/dev.zip

**ESP (ESP32/ESP8266, Board/Sonoff):**

TTGO TWatch 2020

https://www.tindie.com/products/ttgo/lilygor-ttgo-t-watch-2020/

**ESPHome version (latest production, beta, dev branch)**

[app:101]: ESPHome version 1.15.0-dev compiled on Jul  7 2020, 13:36:05

**Affected component:**

st7789v

**Description of problem:**

Watch has a display of 240x240 but the st7789v display seems hardcoded to 135x240 ?

**Problem-relevant YAML-configuration entries:**
```

spi:
  clk_pin: GPIO18
  mosi_pin: GPIO19

display:
  - platform: st7789v
    backlight_pin: GPIO12
    cs_pin: GPIO5
    dc_pin: GPIO27
    reset_pin: 0
    rotation: 0
    lambda: |-
      it.fill(id(my_blue));

```

**Logs (if applicable):**

```

LOG

```

**Additional information and things you've tried:**

I changed the following cpp to return 240 instead of 135 from the ST7789V::get_width_internal() function but that didn't help.

I think it might have to do with the offsets in ST7789V::write_display_data() but I am not sure how to find new ones.

C:\Python38\Lib\site-packages\esphome\components\st7789v\st7789v.cpp

fre69, 2020-11-13:
Hi, somebody as a solution, I have a display 240*320 and I can display only in the middle.

Electronlibre2012, 2020-12-16:
Hello @fre69 and @DACOFFEY 

have you found a solution please, my screen is 240x320 too...

thanks!

fre69, 2020-12-17:
Hello,
Sorry, I spent few hours to check with the display documentation to change in the code the display initialisation but  didn't found the good parameter and I didn't want to spend too much time.

jfle, 2021-04-18:
Hello i have the same issue. Please look the bellow results:
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from esp_tdisplay.local using esphome API
INFO Connecting to esp_tdisplay.local:6053 (192.168.1.74)
INFO Successfully connected to esp_tdisplay.local
[23:48:43][I][app:105]: ESPHome version 1.16.2 compiled on Apr 14 2021, 23:46:54
[23:48:44][C][wifi:443]: WiFi:
[23:48:44][C][wifi:303]:   SSID: [redacted]
[23:48:44][C][wifi:304]:   IP Address: 192.168.1.74
[23:48:44][C][wifi:306]:   BSSID: [redacted]
[23:48:44][C][wifi:307]:   Hostname: 'esp_tdisplay'
[23:48:44][C][wifi:311]:   Signal strength: -44 dB ▂▄▆█
[23:48:44][C][wifi:315]:   Channel: 11
[23:48:44][C][wifi:316]:   Subnet: 255.255.255.0
[23:48:44][C][wifi:317]:   Gateway: 192.168.1.254
[23:48:44][C][wifi:318]:   DNS1: 192.168.1.254
[23:48:44][C][wifi:319]:   DNS2: 0.0.0.0
[23:48:44][C][spi:081]: SPI bus:
[23:48:44][C][spi:082]:   CLK Pin: GPIO18 (Mode: OUTPUT)
[23:48:44][C][spi:084]:   MOSI Pin: GPIO19 (Mode: OUTPUT)
[23:48:44][C][spi:085]:   Using HW SPI: YES
[23:48:44][C][logger:185]: Logger:
[23:48:44][C][logger:186]:   Level: DEBUG
[23:48:44][C][logger:187]:   Log Baud Rate: 115200
[23:48:44][C][logger:188]:   Hardware UART: UART0
[23:48:44][C][st7789v:242]: SPI ST7789V
[23:48:44][C][st7789v:242]:   Rotations: 0 °
**[23:48:44][C][st7789v:242]:   Dimensions: 135px x 240px**
[23:48:44][C][st7789v:243]:   CS Pin: GPIO5 (Mode: OUTPUT)
[23:48:44][C][st7789v:244]:   DC Pin: GPIO16 (Mode: OUTPUT)
[23:48:44][C][st7789v:245]:   Reset Pin: GPIO23 (Mode: OUTPUT)
[23:48:44][C][st7789v:246]:   B/L Pin: GPIO4 (Mode: OUTPUT)
[23:48:44][C][st7789v:247]:   Update Interval: 1.0s
[23:48:44][C][captive_portal:169]: Captive Portal:
[23:48:46][C][web_server:132]: Web Server:
[23:48:46][C][web_server:133]:   Address: esp_tdisplay.local:80
[23:48:46][C][ota:029]: Over-The-Air Updates:
[23:48:46][C][ota:030]:   Address: esp_tdisplay.local:3232
[23:48:46][C][ota:032]:   Using Password.
[23:48:46][C][api:095]: API Server:
[23:48:46][C][api:096]:   Address: esp_tdisplay.local:6053

[](https://community.home-assistant.io/t/st7789v-esp32/297837/3)

fraxinas, 2021-05-03:
I ran into the issue with the hardcoded resolution too, so i hacked a fix for this.
It will allow for a `dimension` parameter in the yaml file.
unfortunately the implementation is a little trickier than just setting the width and height because the required buffer size is greater than the largest available heap chunk. therefore i needed to implement a class which abstracts allocation and access to the display buffer into logical segments.
 
Notice: this will probably break other resolutions, I only have a 240x240 to test.
you need this:
* https://github.com/fraxinas/Geckocontroller/tree/main/software/custom_components/st7789v
* https://github.com/fraxinas/esphome/tree/fix_st7789v/esphome/components/display

it would be cool, if somebody with a different sized display could test this modification (or provide one for me).
if it works allright, then i can file a pull request and we can get the fix upstreamed.

fraxinas, 2021-05-03:
btw, my branch also fixes issue https://github.com/esphome/issues/issues/1944

fre69, 2021-05-13:
Hi,
I tested your code, but I think esphome version was not the same so I have few error to compile.
src/esphome/components/st7789v/st7789v.cpp: In member function 'virtual void esphome::st7789v::ST7789V::draw_absolute_pixel_internal(int, int, esphome::Color)':
src/esphome/components/st7789v/st7789v.cpp:283:25: error: 'struct esphome::Color' has no member named 'to_rgb_565'
   auto color565 = color.to_rgb_565();
                         ^
*** [/data/esp32_base/.pioenvs/esp32_base/src/esphome/components/st7789v/st7789v.cpp.o] Error 1

fraxinas, 2021-05-13:
then you're probably on the dev branch rather than the master / release branch... 
there are a few more things that changed in between

fre69, 2021-05-14:
To be compatible with 1.17.2 I haved to change line 270 in st7789v.cpp, 
auto color565 = color.to_rgb_565(); by this  auto color565 = display::ColorUtil::color_to_565(color);

fre69, 2021-07-15:
Hi,
I'm sorry I tried few time with display buffer but I have problems with glyphs, I think in the last few version of esphome there to much change on display to be compatible.

djtoast, 2022-04-15:
Hi group,   can someone point me to good documentation or give me some direction on how to import this code change onto my setup.  Im attempting to program a 240x240 and cannot seem to find anything .   I tried the ESPHOME Libraries.  Is it even possible to pull these file only or you have to move esphome to your branch completely?   Sorry if this is a noob question i did spend several hours trying to figure this out :(

fraxinas, 2022-04-15:
You need to clone my repo:
`git clone git@github.com:fraxinas/esphome.git -b fix_st7789v`
and then proceed to build and install it from source

dougmaitelli, 2022-06-10:
Is there any plans of merging this fix to esphome? I just got the display and found out about the issue

hjalmarlr, 2022-08-15:
Got it working this way in ESPHOME. Add at the top:
```
external_component:
  - source: github://fraxinas/esphome@fix_st7789v
    components:
      - display
      - st7789v
```
  
Add dimensions under display:
```
display:
  - platform: st7789v
    dimensions: 240x240
    …
```

Works great! Cheers @fraxinas

kbx81, 2022-08-18:
Fixed in https://github.com/esphome/esphome/pull/3651

closed 2022-08-18 by kbx81