esphome/issues #1340: EspHome Dev addon don't install

MiguelAngelLV, 2020-07-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**

All
**ESPHome version (latest production, beta, dev branch)**

1.15-dev

**Affected component:**

Hassio addons

**Description of problem:**

When I try install Hassio dev addont, I get error. I try cleaning cache, reboot... All.
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
20-07-10 12:20:28 ERROR (SyncWorker_19) [supervisor.docker.addon] Can't build 15ef4d2f/armv7-addon-esphome-dev:dev: The command '/bin/bash -o pipefail -c pip3 install --no-cache-dir https://github.com/esphome/esphome/archive/dev.zip' returned a non-zero code: 
```

**Additional information and things you've tried:**

Hassio in Raspberry Pi 4

jesserockz, 2020-07-10:
The recent merge of PR esphome/esphome#1080 and subseequently esphome/esphome#1116 has broken the dev build because it also introduces the need to build the `cffi` module natively for the `arm` variants.

There is no compiler or python headers installed in the hassio base image so this fails.

glmnet, 2020-07-10:
reverting https://github.com/esphome/esphome/pull/1080 fixed this.
The issue happens only on hassio docker images on raspberry pi3 and 4
It did work on raspberry pi 3 - no hassio (running raspbian)

We will need to add the `cffi` to the docker base hassio image before adding back ESP32 support for WPA2-EAP Enterprise WiFi authentication

cc @tomtom5152

tomtom5152, 2020-07-10:
Hello! Well thats somewhat annoying, I don't have any RPi's spare. How would one go about arguing with the containers, I haven't used the hassio at all so which Dockerfile am I actually looking at. I can poke it on a BeagleBone black to see if I can get the container working for ARM which should solve it.

glmnet, 2020-07-10:
Yes annoying, but right now master version v1.14.5 is quite behind of current dev, so many people are using dev in the HomeAssistant add on, that's why I did the revert.

tomtom5152, 2020-07-10:
Oh I fully support the revert if its breaking things, but still irritating that it didn't just work because of some ARM weirdness :P

Anyway, as I said, which Dockerfile am I poking to try and get it working?

glmnet, 2020-07-10:
I really cannot help much as I did not design this and Otto is somewhat idle lately. However this seems an issue with the docker stuff mostly, as we have successfully installed ESPHome with the crypto stuff on the raspberry pi 3. Just not in the home assistant docker.

The docker configurations are all here on this ESPHome organization, repositories relevant are 

- https://github.com/esphome/hassio somewhat, this is read by Home Assistant supervisor to know which add ons are available, we have 3, one for Production, a beta and the dev one.

- https://github.com/esphome/esphome-docker-base this is the repo which has the base docker image, it has some code generation going on to support all the platforms.

jesserockz, 2020-07-10:
From within the [esphome/hassio](https://github.com/esphome/hassio) repo and inside the `esphome-dev` folder I ran
```
docker build --build-arg BUILD_FROM=esphome/esphome-hassio-base-armv7:2.1.1 -t esphome-dev .
```

Output is here:

```
Sending build context to Docker daemon  495.6kB
Step 1/6 : ARG BUILD_FROM=esphome/esphome-hassio-base-amd64:2.0.0
Step 2/6 : FROM ${BUILD_FROM}
2.1.1: Pulling from esphome/esphome-hassio-base-armv7
bfb2710e7a49: Pull complete
c78424bd0db4: Pull complete
60330e98a48f: Pull complete
5fc5290c7dcf: Pull complete
bd27f1bfc347: Pull complete
f8699bba9956: Pull complete
0c195482fb5d: Pull complete
e958e62eb6d1: Pull complete
3e8ea69c0ce2: Pull complete
a183a8e383dd: Pull complete
Digest: sha256:3efc1919a55b4d01a4ba499fa36d3bafc1c9412c2a4dfd81eb2770fa41932384
Status: Downloaded newer image for esphome/esphome-hassio-base-armv7:2.1.1
 ---> 7e9ade0dc143
Step 3/6 : COPY rootfs /
COPY failed: stat /var/lib/docker/tmp/docker-builder388823307/rootfs: no such file or directory
pi@ichigo:~/esphome/hassio $ cd esphome-dev/
pi@ichigo:~/esphome/hassio/esphome-dev $ ls
build.json  config.json  Dockerfile  icon.png  images  logo.png  README.md  rootfs
pi@ichigo:~/esphome/hassio/esphome-dev $ docker build --build-arg BUILD_FROM=esphome/esphome-hassio-base-armv7:2.1.1 -f esphome-dev/Dockerfile -t esphome-dev .
unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/pi/esphome/hassio/esphome-dev/esphome-dev: no such file or directory
pi@ichigo:~/esphome/hassio/esphome-dev $ docker build --build-arg BUILD_FROM=esphome/esphome-hassio-base-armv7:2.1.1 -t esphome-dev .
Sending build context to Docker daemon  140.8kB
Step 1/6 : ARG BUILD_FROM=esphome/esphome-hassio-base-amd64:2.0.0
Step 2/6 : FROM ${BUILD_FROM}
 ---> 7e9ade0dc143
Step 3/6 : COPY rootfs /
 ---> ce06632f6ca8
Step 4/6 : RUN pip3 install --no-cache-dir https://github.com/esphome/esphome/archive/dev.zip
 ---> Running in 3c8a59992a01
Collecting https://github.com/esphome/esphome/archive/dev.zip
  Downloading https://github.com/esphome/esphome/archive/dev.zip (1.7MB)
Collecting PyYAML==5.3.1 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/64/c2/b80047c7ac2478f9501676c988a5411ed5572f35d1beff9cae07d321512c/PyYAML-5.3.1.tar.gz (269kB)
Collecting colorlog==4.1.0 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/00/0d/22c73c2eccb21dd3498df7d22c0b1d4a30f5a5fb3feb64e1ce06bc247747/colorlog-4.1.0-py2.py3-none-any.whl
Collecting cryptography==2.9.2 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/56/3b/78c6816918fdf2405d62c98e48589112669f36711e50158a0c15d804c30d/cryptography-2.9.2.tar.gz (517kB)
Collecting esptool==2.8 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/68/91/08c182f66fa3f12a96e754ae8ec7762abb2d778429834638f5746f81977a/esptool-2.8.tar.gz (84kB)
Collecting ifaddr==0.1.6 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/9f/54/d92bda685093ebc70e2057abfa83ef1b3fb0ae2b6357262a3e19dfe96bb8/ifaddr-0.1.6.tar.gz
Collecting paho-mqtt==1.5.0 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/59/11/1dd5c70f0f27a88a3a05772cd95f6087ac479fac66d9c7752ee5e16ddbbc/paho-mqtt-1.5.0.tar.gz (99kB)
Collecting platformio==4.3.3 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/84/30/fbeab6f6269df25dda9d25c3ab999511c0aedfb30ea76a817b74f36644aa/platformio-4.3.3.tar.gz (187kB)
Collecting protobuf==3.11.3 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/27/9c/ef816295b4b40298fd0a17bf8f0ba6cf3e0c44cb2ce72257168e09996b8b/protobuf-3.11.3-py2.py3-none-any.whl (434kB)
Requirement already satisfied: pyserial==3.4 in /usr/local/lib/python3.6/dist-packages (from esphome==1.15.0.dev0)
Collecting pytz==2020.1 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/4f/a4/879454d49688e2fad93e59d7d4efda580b783c745fd2ec2a3adf87b0808d/pytz-2020.1-py2.py3-none-any.whl (510kB)
Collecting tornado==6.0.4 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/95/84/119a46d494f008969bf0c775cb2c6b3579d3c4cc1bb1b41a022aa93ee242/tornado-6.0.4.tar.gz (496kB)
Collecting tzlocal==2.0.0 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/ef/99/53bd1ac9349262f59c1c421d8fcc2559ae8a5eeffed9202684756b648d33/tzlocal-2.0.0-py2.py3-none-any.whl
Collecting voluptuous==0.11.7 (from esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/24/3b/fe531688c0d9e057fccc0bc9430c0a3d4b90e0d2f015326e659c2944e328/voluptuous-0.11.7.tar.gz (45kB)
Collecting cffi!=1.11.3,>=1.8 (from cryptography==2.9.2->esphome==1.15.0.dev0)
  Downloading https://files.pythonhosted.org/packages/05/54/3324b0c46340c31b909fcec598696aaec7ddc8c18a63f2db352562d3354c/cffi-1.14.0.tar.gz (463kB)
[... 55 lines cut ...]
    copying cffi/vengine_gen.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/verifier.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/setuptools_ext.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/pkgconfig.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/model.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/error.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/__init__.py -> build/lib.linux-armv7l-3.6/cffi
    copying cffi/api.py -> build/lib.linux-armv7l-3.6/cffi
[... cut ...]
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-5rix4cqo/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-eq_m92jl-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-5rix4cqo/cffi/
The command '/bin/bash -o pipefail -c pip3 install --no-cache-dir https://github.com/esphome/esphome/archive/dev.zip' returned a non-zero code: 1
```

tomtom5152, 2020-07-11:
So a bit of further digging reveals that `cryptography` wheels are not available for ARM, and no amount of arguing with APT appears to be getting me anywhere as the chances of the repo version and the `setup.py` version being the same are remote at best, so it has to be compiled from scratch, and as such requires both a compiler and all `-dev` dependancies, namely `build-essential libssl-dev libffi-dev python3-dev`. 
As I see it, this gives three options:
1. Add the build dependancies to the base image. Increases base image size.
2. Add the build dependancies to the hassio image. Keeps base image size the same, but increases dev image
3. Add dependancies in the hassio image, build, then purge the dependancies and compiler to maintain image size.

I'm not sure which is best for these containers, as I don't use them myself so don't really know how much a problem container size is. Option 1 is obviously the most portable of them, option 3 the most space efficient. Unfortunately all of these will increase the build time for ARM targets, but it isn't horrific in my testing. Comments appreciated before I submit a pull.

glmnet, 2020-07-12:
Adding crypto to wheels is not an option?
Just trying to think out of the box.

tomtom5152, 2020-07-13:
As best as I can tell, pyca/cryptography#5292 is attempting to add for 64 bit ARM as it is support by [pypa/manylinux](https://github.com/pypa/manylinux) as of manylinux2014, but ARMv7 support has not been added. Due to the relatively high prevalence of ARMv7 in RPis and similar, I can only assume that this is due to some technical difficult or performance issue with builds, which would likely put it well beyond the scope of ESPHome to fix and beyond my skill set as well.

I therefore suggest option 3 above which I will submit as a PR later this week unless theres any objections.

OttoWinter, 2020-07-14:
@tomtom5152 Interesting, I just tried to build a wheel for `cryptography` on the esphome `aarch64` base image and that worked in ~5 minutes. I'll try armv7 next but in my experience qemu armv7 is faster than aarch64.

OttoWinter, 2020-07-14:
Ah :facepalm: it's even easier: we use ubuntu in the base image, and ubuntu has `python3-cryptography` already. So we just need to install that there.

OttoWinter, 2020-07-25:
This issue has been fixed - cryptography is no longer a hard requirement for installing ESPHome, and the base images have added the package.

closed 2020-07-25 by OttoWinter