esphome/issues #1346: USB to Serial not showing in the dropdown

xrado, 2020-07-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP 01

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.14.5

**Affected component:**

**Description of problem:**
Hass detects usb to serial as /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 and is not showing in a dropdown, therefore can't be selected. Used to show up in previous versions, a few months ago. Don't know if this is esphome or hassio issue.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-07-13:
Did you try restarting everything with the thing plugged in?

xrado, 2020-07-13:
Ok I've tried restarting with USB plugged in and now I also see /dev/ttyUSB0 under hass.io supervisor hardware, but esphome dropdown still not showing it.

[image]

CC2531 is the zigbee dongle and I also have SSD over USB connected
[image]

OttoWinter, 2020-07-15:
We use PySerial internally for listing the serial ports.

If you have SSH/terminal access to the device, could you run

```bash
pip3 install pyserial
python3 -c "from serial.tools.list_ports import comports; print([x[0] for x in comports(include_links=True)])"
```

and post the output?

xrado, 2020-07-15:
@OttoWinter here is the output

    ['/dev/ttyACM0','/dev/ttyAMA0']

if I run `ha hardware info`:

    ...
    serial:
    /dev/ttyAMA0
    /dev/ttyACM0
    /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001949D626-if00
    /dev/ttyUSB0

OttoWinter, 2020-07-15:
Hmm since pyserial doesn't seem to list them we probably have to extend those methods ourselves.

xrado, 2020-07-20:
@OttoWinter I've tested dev version ..works great

closed 2020-07-24 by OttoWinter