esphome/issues #1347: Dev: Detaching servo causes unintended motion

ZetaPhoenix, 2020-07-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 D1 Mini

**ESPHome version (latest production, beta, dev branch)**

v1.15.0-dev

**Affected component:**

servo
https://next.esphome.io/components/servo.html

**Description of problem:**

I order to prevent the cover from moving on reset of the D1 (https://github.com/esphome/issues/issues/964) I switched to the dev branch. After switching to `dev`, when ever I issue a detach for a servo the servo will make an sudden jerk in the CW rotation (negative direction, IE short pulse).

Using detach in the non dev branch works just fine (servo stops driving with out noise or motion) but then the servo will always go to 0 on startup.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
cover:
  - platform: template
    device_class: blind
    name: "Living Room Blinds"

    optimistic: true
    open_action:
    - servo.write:
         id: blind_tilt_servo
         level: 0.0%
    - delay: 4s
    - servo.detach: blind_tilt_servo
    close_action:
    - servo.write:
         id: blind_tilt_servo
         level: -100.0%
    - delay: 4s
    - servo.detach: blind_tilt_servo
    stop_action:
    - servo.detach: blind_tilt_servo

output:
  - platform: esp8266_pwm
    id: pwm_output
    pin: D3
    frequency: 50 Hz

servo:
  - id: blind_tilt_servo
    output: pwm_output
    min_level: 3.0%
    idle_level: 7.5%
    max_level: 12.0%
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ZetaPhoenix, 2020-07-13:
After looking at this with an oscilloscope, I can confirm that the last pulse sent to the servo is shorter that the programed position most of the time.

Debug: I keep opening the cover in HA and can see the 1.5ms pulse for 0% until the detach.

After detach I used a oscilloscope to capture the last pulse sent and measured the duration:
1.1ms
1.22ms
1.08ms
1.5ms

Is detach not waiting for the end of a pulse to stop? That would explain the only shorter pulse duration.

glmnet, 2020-07-13:
Actually it happens on my servo driven courtain too, it doesn't happen always so it makes sense. 

The servos are driven using PWM, software PWM on the ESP8266 

Looks like it is chopped https://github.com/esphome/esphome/blob/351ecf9bd4f3322c110347fabb80cea672fbd1a3/esphome/components/esp8266_pwm/esp8266_pwm.cpp#L41

If we "read" the pin and dont turn it off it will be held high as the stopWaveForm function doesn't seem to stop bring it down at the end of the cycle (because it is stopped).

Not sure how to fix this, certainly could be fixed by implementing servo like the AcDimmer

ZetaPhoenix, 2020-07-13:
Would it make sense for the servo.detach() function to check the status of the waveform and stop it when the pin is low? IE wait for duty_off==0 and then detach?

ZetaPhoenix, 2020-07-14:
I just checked and found out that I also see the issue in 0.14

OttoWinter, 2020-07-15:
@ZetaPhoenix Servo doesn't know about the internals (like `waveform`), it only writes to a float output.

So this is more of an issue with the ESP8266 software PWM. And that's just a proxy for the software PWM from arduino, so the issue is probably not in this codebase.

ZetaPhoenix, 2020-07-15:
@OttoWinter,

I do not have this unintended motion on the MQTT listener I wrote using the Arduino servo library. I am trying to convert the devices that run on that code to ESPHome to allow easier discovery in HA and make it much easier to add additional sensors.

The issue may not be directly in this code base, but how the components are stacked together.

ZetaPhoenix, 2020-11-25:
Still see this on 1.15.3

ZetaPhoenix, 2020-11-25:
Logic Analyzer capture of the last pulse: 
[image]

ZetaPhoenix, 2021-03-27:
I'll need to test again with my logic analyzer but last I did it was still there.

ZetaPhoenix, 2021-03-27:
I am still able to catch the short pulse occasionally.

glmnet, 2021-03-29:
@ZetaPhoenix  I've sent a fix for this, it's not straightforward so merging to codebase will need review and test (specially no other stuff gets broken)

closed 2021-04-08 by OttoWinter