esphome/issues #1348: Library conflicts in LibraryManager

coder6423, 2020-07-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.15.0-dev

**Affected component:**

https://esphome.io/components/esphome.html

**Description of problem:**
When LibraryManager finds two libraries with the same name (eg: "SD"), it includes the first one. That does not support ESP32. I use a library that uses that library, so I cannot edit the include in the source.

I tried to include the correct library into esphome/libraries with its numerical value.
I also tried to manually install with LibraryManager using --interactive and picking the correct one.

However, when I run `esphome test.yaml run`, the wrong library gets installed and used.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
Looking for SD library in registry
Conflict: More than one library has been found by request {"name": "SD", "frameworks": ["arduino"]}:
SD
Automatically chose the first available library (use `--interactive` option to make a choice)
==
#ID: 161
This library aims to expose a subset of SD card functionality in the form of a higher level 'wrapper' object

Keywords: sd, card, file, system
Compatible frameworks: Arduino
Compatible platforms: Atmel AVR
Authors: Adafruit Industries

SD
==
#ID: 868
Enables reading and writing on SD cards. Once an SD memory card is connected to the SPI interface of the Arduino or Genuino board you can create files and read/write on them. You can also move through directories on the SD card.

Keywords: data, storage
Compatible frameworks: Arduino
Compatible platforms: Infineon XMC, Kendryte K210, GigaDevice GD32V, ASR Microelectronics ASR605x, Atmel AVR, Atmel SAM, Espressif 8266, Intel ARC32, Microchip PIC32, Nordic nRF51, ST STM32, Teensy, TI MSP430, TI TIVA, Espressif 32, Nordic nRF52, ST STM8, Atmel megaAVR
Authors: Arduino, SparkFun

LibraryManager: Installing id=161
SD @ 0.0.0-alpha+sha.041f788250 has been successfully installed!

```

**Additional information and things you've tried:**

OttoWinter, 2020-07-15:
ESPHome just puts the libraries you specify directly into the `platformio.ini` file and PIO takes care of installing them.

Indeed, platformio behaves this way when installing libraries (and we've had to deal with that too already).

But this is an issue with PIO, not esphome.

closed 2020-07-15 by OttoWinter

coder6423, 2020-07-15:
Thank you for your response, Otto! 

PIOs solution to this is to install the libraries in --interactive mode. How do I achieve that with esphome, without getting an autoinstall run on top?

OttoWinter, 2020-07-15:
Now I understand what you mean.

Never heard of the interactive option before. I tried to go into the generated platformio project and manually run `pio run --interactive`, but that returns that interactive is not a valid argument. Not sure how interactive is supposed to be used.

coder6423, 2020-07-15:
The --interactive option is on "pio lib install", but you were correct actually - pio run doesn't respect it. I also tried various LDF modes, but it didn't seem to work.

Still, I could find a combination of parameters to make it work, so here is the working approach for anyone in the future with a similar problem: We ignore the library based on its name, and include it based on its id. 

```
esphome:
  libraries:
      - "868" #SD
  platformio_options:
    lib_ignore: SD
```

It still gives the message about the library conflict and that it will select the first one - but actually it links the correct one. Feels a little hacky, but it works for now.

Thanks for your help!

coder6423, 2020-07-15:
The --interactive option is on pip lib install, but you are right, pio run doesn't respect it. I'll keep looking into it and add something if I figure something out. Thanks a lot for your help!