esphome/issues #1349: Pulse counter count does not match binary_sensor input values -> measurements wrong.

neographikal, 2020-07-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker esphome 1.14.5
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

Pulse counter

**Description of problem:**
Pulse counter pulses/min not matching input. See attached log, the binary sensor counts 63 pulses which corresponds to my manual count. The pulse counter component reports:

[12:27:29][D][pulse_counter:159]: 'AEG inverter pulse count': Retrieved counter: 93.00 pulses/min
[12:27:29][D][sensor:092]: 'AEG inverter pulse count': Sending state 5.58000 Kw with 2 decimals of accuracy
[12:27:29][D][sensor:092]: 'AEG inverter daily energy': Sending state 4.21761 Kwh with 4 decimals of accuracy

[sensorlog.txt](https://github.com/esphome/issues/files/4912420/sensorlog.txt)

93 != 63 and therefor my kw measurement is off.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

[config.txt](https://github.com/esphome/issues/files/4912418/config.txt)

```

**Logs (if applicable):**

```
Attached already

```

**Additional information and things you've tried:**

OttoWinter, 2020-07-15:
The binary sensor operates in a polling mode (it checks the state of the pin every few milliseconds).

The pulse counter on the other hand registers an interrupt on the pin, and is immediately notified of any changes and is therefore technically more accurate.

If there's some noise or if the signal bounces a bit then you will see higher values from the pulse_counter. This can be mitigated with the `internal_filter` option

I'd recommend increasing the internal_filter until the values match

neographikal, 2020-07-16:
Crap, thanks for your extensive reply, you were right. The amount of noise surprises me somewhat, but I was wrong on the units (us vs ms...) and had to bump it heavily to 100ms. Will monitor it, but it seems okay now.

closed 2020-07-16 by neographikal