esphome/issues #1351: notify data with group gets error

nathan951, 2020-07-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**
1.15-dev

**Affected component:**

**Description of problem:**
cant save/upload yaml

**Problem-relevant YAML-configuration entries:**
```
  - platform: gpio
    pin: GPIO4
    id: pool_heater
    name: Pool Heater
    inverted: True
    on_turn_on:
      - switch.turn_on: pool_pump
      - switch.turn_off: spa_heater
      - homeassistant.service:
          service: notify.nate
          data:
            title: POOL Heater is ON
            message: The pool heater was just turned on.
            data:
              group: 'pool'
```
 Adding: `group: 'pool' ` flags error

**Logs (if applicable):**

```
flags error: `must be string, got <class 'esphome.helper.OrderedDict'>. did you forget putting quotes around the value?.`

```

**Additional information and things you've tried:**

tried with and without quotes

AlexDanault, 2020-07-14:
I think the `data` attribute of `homeassistant.service` only supports basic name-value mappings, not complex structures. Are you sure you don't have an extra `data` structure (so instead of data.data.group you'd have data.group) ?

OttoWinter, 2020-07-15:
Correct, `data` only supports basic name-value mappings (and adding support for that is not possible due to how the call works internally). If you need to use nested mappings you can declare a `script` on the HA side and call that from ESPHome.

closed 2020-07-15 by OttoWinter