esphome/issues #1354: AM2320 sensor component is unreliable with AM2315

ei-ke, 2020-07-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Arch Linux / PIP / HomeAssistant

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU ESP8266

**ESPHome version (latest production, beta, dev branch)**

Version: 1.14.5

**Affected component:**

https://esphome.io/components/sensor/am2320.html

**Description of problem:**

I'm using the AM2315 sensor with the AM2320 sensor component. It is not working 100% of the time which you can see in the following logs. If I unplug the sensor and replug it to the I2C bus while the rest is alive it works. Sometimes it was also working after a cold boot of the entire setup which is powering the NodeMCU throught the VIN pin but also not everytime.
Maybe this should rather become a feature request as I'm using the AM2315, but I thought it might just be a tiny change to the existing code.

**Problem-relevant YAML-configuration entries:**
```yaml
# Example configuration entry
i2c:
    sda: D2
    scl: D1
    scan: True

sensor:  
  - platform: am2320
    temperature:
      name: "Lufttemperatur"
    humidity:
      name: "Luftfeuchtigkeit"
    update_interval: 10s
```

**Logs (if applicable):**

```
[11:36:45][W][i2c:070]: Received NACK on transmit of address 0x5C
[11:36:45][W][i2c:070]: Received NACK on transmit of address 0x5C
[11:36:45][W][am2320:074]: Writing bytes for AM2320 failed!
[11:36:45][W][am2320:089]: Updating AM2320 failed!
[11:36:48][V][sensor:013]: 'Zerfälle pro Minute': Received new state 18.000000
[11:36:48][D][sensor:092]: 'Zerfälle pro Minute': Sending state 18.00000 CPM with 0 decimals of accuracy
[11:36:48][V][sensor:013]: 'Dosisleistung': Received new state 0.102600
[11:36:48][D][sensor:092]: 'Dosisleistung': Sending state 0.10260 µSv/h with 4 decimals of accuracy
[11:36:51][D][pulse_counter:159]: 'Niederschlag aktuell': Retrieved counter: 1.00 pulses/min
[11:36:51][V][sensor:013]: 'Niederschlag aktuell': Received new state 1.000000
[11:36:51][D][sensor:092]: 'Niederschlag aktuell': Sending state 0.27731 mm/min with 2 decimals of accuracy
[11:36:55][V][sensor:013]: 'Wetterstation Betriebszeit': Received new state 38.993000
[11:36:55][D][sensor:092]: 'Wetterstation Betriebszeit': Sending state 38.99300 s with 0 decimals of accuracy
[11:36:55][V][bmp280.sensor:125]: Sending conversion request...
[11:36:55][D][bmp280.sensor:149]: Got temperature=23.2°C pressure=1009.0hPa
[11:36:55][V][sensor:013]: 'Lufttemperatur (BMP280)': Received new state 23.200001
[11:36:55][D][sensor:092]: 'Lufttemperatur (BMP280)': Sending state 23.20000 °C with 1 decimals of accuracy
[11:36:55][V][sensor:013]: 'Luftdruck': Received new state 1008.975464
[11:36:55][D][sensor:092]: 'Luftdruck': Sending state 1008.97546 hPa with 1 decimals of accuracy
[11:36:55][W][i2c:070]: Received NACK on transmit of address 0x5C
[11:36:56][D][am2320:044]: Got temperature=20.6°C humidity=64.8%
[11:36:56][V][sensor:013]: 'Lufttemperatur': Received new state 20.600000
[11:36:56][D][sensor:092]: 'Lufttemperatur': Sending state 20.60000 °C with 1 decimals of accuracy
[11:36:56][V][sensor:013]: 'Luftfeuchtigkeit': Received new state 64.800003
[11:36:56][D][sensor:092]: 'Luftfeuchtigkeit': Sending state 64.80000 % with 1 decimals of accuracy

```

**Additional information and things you've tried:**
I've created a custom component based on [this library](https://github.com/asukiaaa/AM2320_asukiaaa) which works flawlessly.
Further I tried using `setup_priority: -100` for the AM2320 sensor component which was mentioned [here](https://github.com/esphome/issues/issues/192#issuecomment-481171704) and once had a success around `-50` but would rather call it a glitch as I couldn't reproduce it consistently. Also changing the `update_interval` for the component or using other I2C bus frequencys changed nothing relating to this issue.
I also tried to compare your implemantation with the one of asukiaaa but that's beyond my "programming knowledge".

The YAML with the custom AM2315 component is this:
```yaml
i2c:
    sda: D2
    scl: D1
    scan: True

sensor:
  - platform: custom
    lambda: |-
      auto AM2315sensor = new AM2315();
      App.register_component(AM2315sensor);
      return {AM2315sensor->temperature, AM2315sensor->humidity};

    sensors:
      - name: "Lufttemperatur"
        unit_of_measurement: °C
        accuracy_decimals: 1
      - name: "Luftfeuchtigkeit"
        unit_of_measurement: "%"
        accuracy_decimals: 1

```

AM2315.h
```C++
#include "esphome.h"
#include "AM2320_asukiaaa.h"

class AM2315 : public PollingComponent, public Sensor {
 public:
  AM2320_asukiaaa mySensor;

  Sensor *temperature = new Sensor();
  Sensor *humidity = new Sensor();

  AM2315() : PollingComponent (10000) {}

  void setup() override {
    // Initialize the device here. Usually Wire.begin() will be called in here,
    // though that call is unnecessary if you have an 'i2c:' entry in your config

    Wire.begin();
    mySensor.setWire(&Wire);
  }
  void update() override {
    if (mySensor.update() == 0) {
      float temp = mySensor.temperatureC;
      temperature->publish_state(temp);
      float humid = mySensor.humidity;
      humidity->publish_state(humid);
    }
  }
};
```

OttoWinter, 2020-07-15:
CC @T3m3z (https://github.com/esphome/esphome/pull/554)

ei-ke, 2020-07-15:
In case one of you needs an AM2315 I can ship and donate one or you let me know what it costs in your region and I'l send you the ammount and you can order one yourself.

T3m3z, 2020-07-18:
I tried to check if I could find a significant difference from these two implementations but they seem to do essentially the same. Both wake up the sensor (needed to get the sensor ready for the actual read and this causes those NACK-errors). After that they send commands to read data (0x3, 0x0, 0x4) and then read 8 bytes (2 preamble, 2 temperature, 2 humidity, 2 CRC). Only difference seems to be that the library you are using has added 1.5ms delay between wakeup and the actual read.

This is just a wild guess but could it be that the AM2315 takes a bit longer to be operational after the wakeup call? It might be that on some boots the component gets marked as "failed" (row 57 in ```esphome/components/am2320/am2320.cpp```) as it cannot read the data during the boot. Im not sure but ESPHome might not try to update the sensor after this has happened ([source](https://esphome.io/api/classesphome_1_1_component#a03418b5e70b9076e02e7066a3c42bc70)).

You seem to be quite capable with coding so you could try this with your custom component by commenting out row 42 (```delay(2); // >1.5ms```) in ```AM2320_asukiaaa.cpp``` . This test might reveal if the culprit is the absence of this delay.

ei-ke, 2020-07-18:
Thanks for helping me T3m3z.
Commenting out row 42 breaks my custom component silently (as I have no test included to see if the sensor is reading something or not).
I've just tried modifying the am2320.cpp by adding a delay but I have no clue where exactly or it won't fix it. Not sure :/

T3m3z, 2020-07-19:
So it seems that my wild guess was correct :)

Method ```AM2320Component::read_bytes_``` has already support for that conversion delay - see lines 72 and 78-79. in  ```esphome/components/am2320/am2320.cpp```. You need to edit row 88 ```if (!this->read_bytes_(3, data, 8, 2)) {``` and change it to ```if (!this->read_bytes_(3, data, 8, 2, 2)) {```. This will add 2ms delay between wakeup call and the actual read.

ei-ke, 2020-07-19:
I already noticed this and changed the initial value of `conversion` in the header file to 2 (as I don't know what I'm doing :) )
You recommended change throws the following error

```
Compiling .pioenvs/wetterstation/src/esphome/components/am2320/am2320.cpp.o
src/esphome/components/am2320/am2320.cpp: In member function 'bool esphome::am2320::AM2320Component::read_data_(uint8_t*)':
src/esphome/components/am2320/am2320.cpp:88:42: error: no matching function for call to 'esphome::am2320::AM2320Component::read_bytes_(int, uint8_t*&, int, int, int)'
   if (!this->read_bytes_(3, data, 8, 2, 2)) {
                                          ^
src/esphome/components/am2320/am2320.cpp:88:42: note: candidate is:
src/esphome/components/am2320/am2320.cpp:72:6: note: bool esphome::am2320::AM2320Component::read_bytes_(uint8_t, uint8_t*, uint8_t, uint32_t)
 bool AM2320Component::read_bytes_(uint8_t a_register, uint8_t *data, uint8_t len, uint32_t conversion) {
      ^
src/esphome/components/am2320/am2320.cpp:72:6: note:   candidate expects 4 arguments, 5 provided
*** [.pioenvs/wetterstation/src/esphome/components/am2320/am2320.cpp.o] Error 1
```

Isn't the original call already handing over 2ms delay?
`3, data, 8, 2`
`uint8_t a_register, uint8_t *data, uint8_t len, uint32_t conversion`

T3m3z, 2020-07-19:
Oh sorry. Yes it seems that you are right. Therefore let me withdraw that
wild guess as it was completely wrong. You could try to modify the setup
method and comment out the "if" in which the component is marked as failed
if it does not respond correctly. Again, this is just a wild guess but it
might be that the component gets marked as failed during bootup.

Or was it that after booting, the sensor does not work and you could get it
to work by unplugging and replugging it?(You might have mentioned this but
I am replying by email so I cannot see the original issue report.)

ei-ke, 2020-07-19:
I could get it to work after hot unplug and replug. That's correct.
I played around (fumbled around in the dark) today and tried various things. Waking it up with begin_transmission, write_raw, end_transmission using the values from the working library (though the existing ones look like thise in the data sheet).
Changed the wakeup to just one byte / no byte - at least the datasheets looks to me like it needs just a I2C start and stop. 
Adding delays here and there, raising the existing delay from 2 to 5,10,20,1000.
I also commented out the mark as failed section which results in the two new errors: 1) write fail 2) read fail (which is in the right order as the write fail is the wakeup write).

I'll use the second AM2315 (thought maybe the first one has a defect) tomorrow and do all thise changes again and leave them as a comment in my repo and notify you so you can have a look what doesn't help.
I'll also hook up my el cheapo logic analyser just in case it shows something interesting.

ei-ke, 2020-07-27:
Sorry for the delay. I've added my logic analyser recordings to [this branch](https://github.com/ei-ke/esphome-esphome/tree/AM2320-issue-data). The noticable difference to me is that with the custom library on cold boots it beginning with 4x `Setup Write to ['184'] + NAK` on the bus and with the internal implementation there are only 2x `Setup Write to ['184'] + NAK`.
Further I purchased an AM2320 an can confirm that this component doesn't have these issues.

ei-ke, 2020-07-27:
As promised [this](https://github.com/ei-ke/esphome-esphome/commits/AM2320-with-AM2315) is what I've tried so far.

T3m3z, 2020-07-28:
Interesting results (and nice logic analyzer logs!). I tried to check and see if I could see something significant in those but the only major difference seems to be that the custom library makes 3 of those ```Setup Write to ['184'] + NAK``` followed by one ```Setup Write to ['184'] + ACK``` (you mentioned that there were 4x ```Setup Write to ['184'] + NAK``` but I didn't see that). Anyway, this means that with the asukiaa library, the first wakeup and read attempt fails as the sensor does not respond. But the next attempt after 5 seconds seems to be successful. This makes me wonder if the sensor is somehow "slower" to wakeup during the cold boot. Possibly is needs some time before it can be read? Asuakiaa library gets over this by just repeating the reading process.

Was it that you tested removing the part where component is marked as failed during setup? Does the esphome continue trying to read the sensor every now and then or does it stop polling the sensor completely? In you logic analyzer dumps it seems that after cold boot it does try to read the sensor only once and the sensor does not respond (the same behaviour as with the asuakiaa library but that library just tries to read again and succeeds the second time it tries to read the data.

ei-ke, 2020-07-28:
That was my fault. The working library has shown 3x NAK and 1x ACK as the log also shows.

Yes, I just removed the part of the code that marks the sensor as failed in ESPHome in [this](https://github.com/ei-ke/esphome-esphome/commit/432b4c087023f73d36a64fa73eb1fcd1f7522fc7) commit.
Than it's trying to poll the sensor but fails with communication error in the ESPHome log and  [this](https://github.com/ei-ke/esphome-esphome/blob/AM2320-with-AM2315/logic_analyser_logs/432b4c08.csv) is the logic analyser log.
Further when I boot the original ESPHome code with the AM2320 sensor attached and just swap it with the AM2315 the measurements continue without any issue.

ei-ke, 2020-07-28:
I've just tried it with delays in the setup section without success. Changes can be found [here](https://github.com/ei-ke/esphome-esphome/blob/1a2f3a8d18060fb8c48c0ce5a1dbbf61082b2b8b/esphome/components/am2320/am2320.cpp#L51). Also going up to 5000ms or using the original code like the following doesn't work:

```cpp
void AM2320Component::setup() {
  ESP_LOGCONFIG(TAG, "Setting up AM2320...");
  uint8_t data[8];
  data[0] = 0;
  data[1] = 4;
  this->read_data_(data);
  delay(5000);
  if (!this->read_data_(data)) {
    this->mark_failed();
    return;
  }
}
```

1994rstefan, 2020-08-08:
I came accross this issue because I have problems using the AM2320 sensors. Maybe a lot of AM2320 work without a significant delay between wake and real read, but mine seem to have a problem with that. I also checked the datasheet which states that there must be a delay of at least 800µs. (https://datasheetspdf.com/pdf-file/952504/Aosong/AM2320/1 - Page 16)

I have also another device on the same I²C-Bus, so I checked if the second device is messing something up. But it seems, that the second request comes after 200µs, which is far below the minimum of 800µs. Maybe this issue also hardly depends on the used hardware because a slower microcontroller reaches the 800µs and mine does not? But regardles of that, a delay should get implemented to make sure the timing requirements are met.

ei-ke, 2020-08-08:
Moin Stefan,

if you have an idea where this delay should be added or rather how please let me know and I'll try it and also add the logic analyser logs.
As you can see [here](https://github.com/ei-ke/esphome-esphome/tree/AM2320-with-AM2315) I have more or less no clue what to do.

1994rstefan, 2020-08-08:
@ei-ke use the [original am2320 code](https://github.com/esphome/esphome/blob/dev/esphome/components/am2320/am2320.cpp) and try to add the delay [on line 86](
https://github.com/esphome/esphome/blob/ccb6fc3010008f4ab57665852c11d1a57b0ff9ce/esphome/components/am2320/am2320.cpp#L86)

I need some sleep now - but I would check it in detail tomorrow if this does not fix it.

ei-ke, 2020-08-08:
That magic moment - or WTF... You name it.

I have reset my branch to be even with dev to create a fresh logic analyser log while running the dev version. While doing so I changed
```yaml
i2c:
    sda: D2
    scl: D1
    scan: False
```
to
```yaml
i2c:
    sda: D1
    scl: D2
    scan: False
```
because... OCD... Just because it matched the jumper cables better (yellow -> SDA)... Works out of the box.
Interchanging D1/D2 in hardware as well as the config breaks the setup which is still the same as few weeks ago: NodeMCU, AM2315 on 3.3V, nothing else.
Running version 1.14.5 installed via PIP the entire setup is broken. No matter to which of D1/D2 SDA/SCL is linked.

closed 2020-12-21 by stale[bot]

loujdev, 2021-06-09:
I added the am2315 library to my project and used the function to read the temp and humidity at the same time and that works fine

loujdev, 2021-06-09:
Yaml example of esphome:

```  Yaml
esphome:
  includes:
    - AM2315.h
  libraries:
    - "Adafruit AM2315"
  name: koeling
  platform: ESP32
  board: nodemcu-32s #esp-wrover-kit
  
wifi:
  ssid: "wifiexample"
  password: "pass example"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "passsss"

ota:
  password: "passssss"

i2c:
  sda: 21
  scl: 22
  scan: false
  
sensor:

  - platform: custom
    lambda: |-
      auto AM2315sensor = new AM2315();
      App.register_component(AM2315sensor);
      return {AM2315sensor->temperature_sensor, AM2315sensor->humidity_sensor};

    sensors:
      - name: "Temperature"
        unit_of_measurement: °C
        accuracy_decimals: 1
      - name: "Humidity"
        unit_of_measurement: "%"
        accuracy_decimals: 1
```
The file i place in /config/esphome of my home assistant
``` C#
#include "esphome.h"

#include "Adafruit_AM2315.h"

class AM2315 : public PollingComponent, public Sensor {
 public:
  Adafruit_AM2315 mySensor;

  Sensor *temperature_sensor = new Sensor();
  Sensor *humidity_sensor = new Sensor();

  AM2315() : PollingComponent (10000) {}
  float get_setup_priority() const override { return esphome::setup_priority::BUS; }

  void setup() override {
    // Initialize the device here. Usually Wire.begin() will be called in here,
    // though that call is unnecessary if you have an 'i2c:' entry in your config

    mySensor.begin();
    
  }
  void update() override {
    // This is the actual sensor reading logic.
    float temperature;
    float humidity;
    
    mySensor.readTemperatureAndHumidity(&temperature, &humidity);
    
    temperature_sensor->publish_state(temperature);
    humidity_sensor->publish_state(humidity);
  }
};
```
and used the library here: https://github.com/adafruit/Adafruit_AM2315/blob/master/Adafruit_AM2315.h

snailbusa, 2021-09-25:
hey!! -->> : https://github.com/snailbusa/AM2315-For-ESPHome <<-- @ei-ke

snailbusa, 2021-09-25:
> hey!! -->> : https://github.com/snailbusa/AM2315-For-ESPHome <<-- @ei-ke
[image]