esphome/issues #1356: remote_transmitter with ESP8266 and FS1000A

bretez, 2020-07-14:
**Raspberry Pi 4 (2gb) with Home Assistant 0.112.4, Receiver : RXB6, transmitter FS1000A**

**ESP8266**

**ESPHome 1.14.5**

**remote_transmitter and remote_receiver : https://esphome.io/components/remote_transmitter.html#remote-transmitter-actions**

**Hi everybody,

I'm using Home Assistant on a Raspberry pi 4 since 1week, I'm little Noob :D
I have download the installation for Raspberry pi4 (32 recommended) and burn the image on Micro SD Card.

I'm using for control clim, switches via Zigbe2mqtt and the solution work perfectly.

In my home, i have roller shutters that work with RF modules in 433mhz.

I would like to be able to control them via the home assistant application. So I decided to go through ESPHOME via an ESP8266 (HiLetgo 2pcs NodeMCU LUA ESP8266 CP2102 ESP-12E).
I followed the procedure on the page: https://esphome.io/components/remote_transmitter.html#remote-transmitter-actions (SETUP RF DEVICES section).
For this operation I have : RXB6 Receiver and FS1000a Transmitter.

So I attacked by the procedure to receive the codes by pressing on the remote controls that I already have. (Tx Pocket 4)

Here is my code: 

**

**Problem-relevant YAML-configuration entries:**
```name: esp
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "BRETEZ"
  password: "My Secret SSID Password ;)"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp Fallback Hotspot"
    password: "bH2avbNBgDBw"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

remote_receiver:
  pin:
    number: D5
    inverted: yes
    mode: INPUT_PULLUP
  dump: all

```

**[13:43:41][D][api.connection:583]: Client 'Home Assistant 0.112.4 (192.168.1.233)' connected successfully!
[13:43:46][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='10010000'
[13:43:46][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=6 data='101101110'
[13:44:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=2 data='011001011'
[13:45:27][I][ota:046]: Boot seems successful, resetting boot loop counter.
[13:46:02][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=6 data='10100100'
[13:46:17][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=6 data='11100110'
[13:46:20][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=6 data='110011111'
[13:46:49][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=8 data='11111111111'
[13:47:04][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=8 data='11111111'
[13:47:13][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=1 data='1111111111110000010111111111110111111111111100000101111111111101'
[13:47:37][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='1111011011101'
[13:47:38][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=6 data='101111000011111101'**

```
PASTE DEBUG LOG HERE

```

**My code must be among all those ...
I'm in the good way, maybe :)

__
But now, I want send code with ESPHOME.
When I past the code in ESPHOME in my Device, I have error on "LINE 24", you can show on the image

Do you have an idea for me ?

Thanks so Much. (From France) ;)
[image]
**

glmnet, 2020-07-14:
Line 28 is too much indented switch shouldn't be indented at all

bretez, 2020-07-14:
What I can do ? :S

bretez, 2020-07-14:
Cool, I have remove the space and it's word (I have no "red" in line 24)
But, when I show log, nothing happens...

[15:00:37][C][wifi:287]:   Hostname: 'esp'
[15:00:37][C][wifi:291]:   Signal strength: -41 dB ▂▄▆█
[15:00:37][C][wifi:295]:   Channel: 1
[15:00:37][C][wifi:296]:   Subnet: 255.255.255.0
[15:00:37][C][wifi:297]:   Gateway: 192.168.1.1
[15:00:37][C][wifi:298]:   DNS1: 192.168.1.1
[15:00:37][C][wifi:299]:   DNS2: (IP unset)
[15:00:37][C][logger:175]: Logger:
[15:00:37][C][logger:176]:   Level: DEBUG
[15:00:37][C][logger:177]:   Log Baud Rate: 115200
[15:00:37][C][logger:178]:   Hardware UART: UART0
[15:00:37][C][remote_transmitter:018]: Remote Transmitter...
[15:00:37][C][remote_transmitter:019]:   Carrier Duty: 100%
[15:00:37][C][remote_transmitter:020]:   Pin: GPIO14 (Mode: OUTPUT)
[15:00:37][C][captive_portal:169]: Captive Portal:
[15:00:37][C][ota:029]: Over-The-Air Updates:
[15:00:37][C][ota:030]:   Address: esp.local:8266
[15:00:37][C][api:095]: API Server:
[15:00:37][C][api:096]:   Address: esp.local:6053
[15:01:38][I][ota:046]: Boot seems successful, resetting boot loop counter.

glmnet, 2020-07-14:
you should see some messages that it is transmitting something, if nothing happens when you actuate the switch try increasing logger level:

```yaml
logger:
  - level: VERBOSE
```

bretez, 2020-07-14:
Thanks so much for the reply.

I have Add Swich in the homepage and when I press it, I have this message on log : 

INFO Reading configuration /config/esphome/esp.yaml...
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:19:54][D][switch:021]: 'RF Power Button' Turning ON.
[15:19:54][D][remote_transmitter:066]: Sending remote code...

It's work but nothing happens in my roller Shutters.
Maybe I send wrong code ?

glmnet, 2020-07-14:
Alright, may be the code wrong, or may be your transmitter is not working. Most likely first option.
If you have another ESP8266 you can setup as a receiver only to monitor what happens when you press the button on the handheld remote unit vs when you click the switch on the Home assistant UI you'll start finding answers yourself 😀

bretez, 2020-07-14:
Hi,

I have test with remote for low-cost switches and I receive the signal and I can send him without problem. 
It's work.

But with my Roller shutters I don't see the code.. It's not Somfy RTS...

bretez, 2020-07-14:
This is my Receiver :

https://www.elima.cz/obchod/3299-15500-modul-prijimace-rf-zaluziovy-vestavny-abb-p-8551.html

I have bought in France but.. I find only on cz website. :/

closed 2020-12-05 by stale[bot]