esphome/issues #1359: Lambda commands turn_on(), turn_off() do not work for fan

epiniguin, 2020-07-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini
**ESPHome version (latest production, beta, dev branch)**

v1.14.4
**Affected component:**

Fan

**Description of problem:**
I want to turn_on and turn_off fans from lambda.
But commands like id(fan1).turn_on() do not work.
There is not anything in the debug.

**Problem-relevant YAML-configuration entries:**
```yaml
        - lambda: |-
              if (x > 70) {
                id(bathroom1_fan).turn_on(); //this does not work
                id(bathroom1_fan_is_on_by_humidity).turn_on(); //this works - I see it in the debug
              }
              else if ((x < 67) && (id(bathroom1_fan_is_on_by_humidity).state)){ 
                id(bathroom1_fan).turn_off(); //this does not work
                id(bathroom1_fan_is_on_by_humidity).turn_off(); //this works - I see it in the debug
              }

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
By the way the commands for switch id(bathroom1_fan_is_on_by_humidity).turn_on() and id(bathroom1_fan_is_on_by_humidity).turn_off() work in this lambda code.

OttoWinter, 2020-07-15:
Correct, the action works as `id(the_fan).turn_on().perform()` (like with lights).

The reason for that is you can set additional parameters in the object returned by `turn_on()`, for example `set_oscillating()` etc.

epiniguin, 2020-07-15:
Thanks a lot!

closed 2020-07-15 by epiniguin