esphome/issues #1360: ESPHome 315Mhz RF Reciever and Transmitter Issue after working test with Arduino Uno and RC-Switch Library

CJOWood, 2020-07-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Attempted with D1 Mini and NodeMCU board.

**ESPHome version (latest production, beta, dev branch)**

1.14.5

**Affected component:**

[remote_receiver](https://esphome.io/components/remote_receiver.html)

**Description of problem:**
I recently picked up a 315Mhz transmitter and receiver. To verify they worked I set them up using an Arduino Uno and the rc-switch library. Both the transmitter and receiver worked really well from both ends of my house using the example scripts from the rc-switch library.

I then set them up with a D1 Mini and ESPHome… and I couldn’t receive a signal unless the remote was extremely close to the antenna (1-2cm away). The received code was different from what rc-switch's output showed.  Nothing changed except switching out the board. Output from ESPHome Log:
```
[19:27:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=6 data='001001101111100011010111' //This is the only one that matches and is inconsistent.
[19:27:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:33][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:34][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:34][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
[19:27:34][D][remote.rc_switch:243]: Received RCSwitch Raw: protocol=3 data='111111011000011110111100'
```

I bought the “**LR33B** 315MHz ASK Superheterodyne RF Module Wireless Remote Control Receiver Module” from BangGood.

**Problem-relevant YAML-configuration entries:**
```yaml
remote_receiver:
  pin: D2
  dump:
    - rc_switch
  tolerance: 25%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb

```

**Additional information and things you've tried:**
This is the output I get from the rc-switch library's ReceiveDemo_Advanced example code:
```
Decimal: 2554071 (24Bit) Binary: 001001101111100011010111 Tri-State: not applicable PulseLength: 480 microseconds Protocol: 6
Raw data: 11044,456,416,896,424,892,872,436,428,888,432,888,876,428,876,432,432,884,880,428,880,432,876,432,876,436,872,440,424,892,428,884,440,876,884,428,880,428,436,884,880,428,436,880,884,428,880,428,880,

Decimal: 2554071 (24Bit) Binary: 001001101111100011010111 Tri-State: not applicable PulseLength: 480 microseconds Protocol: 6
Raw data: 11044,456,416,900,1356,292,740,96,3860,376,52,88,688,132,88,96,832,188,92,332,464,192,108,596,68,296,116,736,120,368,48,344,228,1248,80,3684,700,1276,224,276,304,168,168,152,888,156,260,156,532,
```

glmnet, 2020-07-17:
You are mixing too much here. 
To really point that esphome is trouble here you should do all your testing without changing any hardware. If you're using arduino uno chances are you're powering the receiver with 5V with it and and esp runs on 3.3v just to tell one difference.

CJOWood, 2020-07-17:
> 
> 
> You are mixing too much here.
> To really point that esphome is trouble here you should do all your testing without changing any hardware. If you're using arduino uno chances are you're powering the receiver with 5V with it and and esp runs on 3.3v just to tell one difference.

Hello, Thanks for the reply. The receiver can run on 5v or 3.3v and the D1 Mini can provide 5v and 3.3v. I've tried using a variety of combinations.

I can also confirm it works perfectly with the D1 Mini and the rc-switch library. Output from D1 Mini running rc-switch example script:
```
Decimal: 2554071 (24Bit) Binary: 001001101111100011010111 Tri-State: not applicable PulseLength: 480 microseconds Protocol: 6
Raw data: 11051,444,430,883,437,878,884,426,440,877,442,873,890,420,887,423,442,873,890,420,887,423,886,424,884,426,881,429,437,877,442,874,446,869,894,416,892,418,446,871,892,418,446,869,893,417,890,420,887,

Decimal: 2554071 (24Bit) Binary: 001001101111100011010111 Tri-State: not applicable PulseLength: 480 microseconds Protocol: 6
Raw data: 11049,446,426,887,433,882,882,428,436,879,441,874,888,422,887,423,441,874,888,422,885,425,884,424,882,428,881,429,436,879,110,286,88,114,113,982,201,98,566,116,177,402,227,1119,152,1556,137,54,178,
```

As soon as I load ESPHome it can't read the signal.

glmnet, 2020-07-17:
Ok. No arduino better.

So are you loading just the rc switch library? 

Are you enabling / disabling WiFi in both Devices? 

Wifi “should not” interfere with 315mhz but may be 

Now your device seems to transmit 430us and 880ms pulses. Switching between protocols 6 and 3

I don’t understand much of this but seems like needs some more thinking. 

You can experiment replacing by your values on protocol 1 here 

esphome/components/remote_base/rc_switch_protocol.cpp Line 10

That should match better your timings

CJOWood, 2020-07-17:
The remote control that I am sniffing does not switch between protocols or codes. ESPHome is inconsistent in reading the signal. The output I've posted from the rc-switch library is correct and can be used to control the device (ceiling fan). The ones received from ESPHome are often too short or only have parts of the full signal read.

CJOWood, 2020-07-20:
I haven't gotten anywhere with this problem yet. Are there any suggestions that could help me out?

sergeolkhovik, 2020-11-09:
Gentlemen, let me join your discussion. I don't know if I should create another issue, or stay here. I have very similar situation - I'm trying to use RF 433MHz reciever and ESP8266 to receive some window sensor states. Sensor works OK with RF Link but the distance is too far, so I decided to reuse one of my ESPHome/ESP8266 devices located not so far from that sensor. I'd been suprised that my device sees nothing from the sensor, so I decided to do more tests to figure out what's wrong.
So then I tested my SYN480R reciever with Arduino Uno and any RF device - all is fine. Tested with another receiver - all is fine. Then I set up my testing Wemos D1 mini and set up such configuration:

```
esphome:
  name: d3
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password

remote_receiver:
  pin:
    number: D5
    #inverted: true
  #dump: all
  dump: rc_switch

sensor:
  # Sensors with general information.
  # Uptime sensor.
  - platform: uptime
    name: D3 Uptime
    # WiFi Signal sensor.
  - platform: wifi_signal
    name: D3 WiFi Signal
    update_interval: 60s

text_sensor:
  # Text sensors with general information.
  # Expose ESPHome version as sensor.
  - platform: version
    name: D3 ESPHome Version
    # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: D3 IP address
  - platform: wifi_info
    bssid:
      name: D3 BSSID
  - platform: template
    name: D3 Channel
    lambda: |-
      std::string out;
      char buffer[64];
      snprintf(buffer, sizeof(buffer), "%u", wifi_get_channel());
      out.append(buffer);
      return out;
    update_interval: 60s

switch:
  # Switch to restart the doorbell.
  - platform: restart
    name: D3 Restart
```
Pretty simple, right? And I have to say that I see nothing in logs. Then I simply uploaded the following Arduino sketch into that Wemos (and changed no wiring! +5V, GND and DATA connected to D5) and got everything working!

```
GOT A MESSAGE!
Decimal: 14992654 (24Bit) Binary: 111001001100010100001110 Tri-State: not applicable PulseLength: 455 microseconds Protocol: 1
Raw data: 14070,1367,466,1339,469,1382,469,437,1369,484,1368,1340,470,478,1371,435,1398,1361,468,1388,468,480,1370,435,1369,438,1369,1386,470,484,1368,1387,496,455,1369,436,1369,437,1368,487,1367,1341,467,1384,468,1339,469,476,1371,

GOT A MESSAGE!
Decimal: 14992650 (24Bit) Binary: 111001001100010100001010 Tri-State: not applicable PulseLength: 455 microseconds Protocol: 1
Raw data: 14071,1362,470,1386,469,1383,469,437,1369,438,1369,1339,471,482,1366,436,1398,1362,467,1387,469,484,1371,482,1367,439,1406,1349,469,485,1367,1389,495,455,1368,439,1366,440,1367,486,1367,1390,466,487,1366,1342,466,483,1367,
```

The sketch is really simple from rc-switch library (https://github.com/sui77/rc-switch/tree/master/examples/ReceiveDemo_Advanced):

```
/*
  Example for receiving
  
  https://github.com/sui77/rc-switch/
  
  If you want to visualize a telegram copy the raw data and 
  paste it into http://test.sui.li/oszi/
*/

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();

void setup() {
  Serial.begin(9600);
  mySwitch.enableReceive(D5);  // Receiver on interrupt 0 => that is pin #2
  Serial.println("OK!");
}

void loop() {
  if (mySwitch.available()) {
    Serial.println("GOT A MESSAGE!");
    output(mySwitch.getReceivedValue(), mySwitch.getReceivedBitlength(), mySwitch.getReceivedDelay(), mySwitch.getReceivedRawdata(),mySwitch.getReceivedProtocol());
    mySwitch.resetAvailable();
  }
}
```

I don't understand what's wrong with this? I had Sonoff RF (hacked) previously that worked with some other my RF sensors (like water leakage) and that worked fine! So I doubt this is rc-switch library itself.
Also I tried those settings mentioned above:
```
  tolerance: 25%
  filter: 250us
  idle: 4ms
  buffer_size: 2kb
```
And still so no dump from my GS-WDS07 sensor. I have no idea what to do next with all this. Any thounghts? Thanks!

sergeolkhovik, 2020-11-09:
Hey! I was able to get it working with this configuration!

```
remote_receiver:
  pin:
    number: D5
  #dump: all
  dump: rc_switch
  tolerance: 60%
  #filter: 250us
  #idle: 4ms
  #buffer_size: 2kb
```

And if you take a look at https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp, line 105:

```
int RCSwitch::nReceiveTolerance = 60;
```

Now I see in the logs for that my sensor:

```
[22:30:12][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=1 data='111001001100010100001110'
[22:30:12][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=1 data='111001001100010100001110'
[22:30:12][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=1 data='111001001100010100001110'
[22:30:12][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=1 data='11100100110001010000111'
[22:30:13][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=1 data='111001001100010100001010'
[22:30:13][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=1 data='111001001100010100001010'
[22:30:13][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=1 data='111001001100010100001010'
[22:30:13][D][remote.rc_switch:256]: Received RCSwitch Raw: protocol=1 data='111001001100010100001010'
```
First 3 lines is the closing event, last 4 - opening event.
Great!

CJOWood, 2020-11-10:
@sergeolkhovik I just ended up using MQTTGateway. It works but it's not as fun to use at ESPHome.

sergeolkhovik, 2020-11-11:
@CJOWood I'd say entire RF solution isn't reliable at all, it might be good (but completely non-secure) to receive some data like thermometers etc, may be acceptable for door/window or PIR sensors for some automations, but it's completely non-good to manage devices like relays because you don't have any feedback and don't know what's current/real state of the device.
Yeah, I have Sonoff RF flashed with OMG and it runs at my parents' house but personally I think that RF sensors are good becuase of low power consumption but for switches/relays I'd better look at Zigbee/WiFi solution.

Regarding my success with syn480r, once I connected it to my real device (ESP-07) it started working very unstable and reconnecting to HA every 2 minutes or something like. Surprising! Will check today once more, have no idea what's the reason.

closed 2021-03-26 by stale[bot]