esphome/issues #1361: BLE UUID Tracking Non-Functional

tjharman, 2020-07-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Running the dev version on a Linux Debian 10 platform.

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sparkfun ESP32 Thing
**ESPHome version (latest production, beta, dev branch)**

v1.15.0-dev
**Affected component:**

ble_rssi sensor

**Description of problem:**
A patch was introduced [here](https://github.com/esphome/esphome/pull/800) by @lumpusz which supports UUID tracking for ble, it was accepted but it doesn't appear to work.  The beacon is never found by my device, even though a ble scanner on my mobile phone is able to clearly see the ibeacon.  The ble scanner does seen the MAC address of my beacon, however as it's mac that gets randomised, this can't be used.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: loungeesp
  platform: ESP32
  board: esp32thing

wifi:
  ssid: "X"
  password: "X"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Loungeesp Fallback Hotspot"
    password: "zWtCRsTgnQ1p"

captive_portal:

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:
sensor:
  - platform: ble_rssi
    service_uuid: '08885DD0-111B-11E4-9191-0900200C9A66'
    name: "Dog Tracker"
  - platform: ble_rssi
    mac_address: 'F0:99:19:2B:F0:43'
    name: "Tim's Garmin"

```

**Logs (if applicable):**

```
INFO Reading configuration esp/loungeesp.yaml...
INFO Starting log output from loungeesp.local using esphome API
INFO Connecting to loungeesp.local:6053 (192.168.0.217)
INFO Successfully connected to loungeesp.local
[08:35:42][I][app:105]: ESPHome version 1.15.0-dev compiled on Jul 18 2020, 08:25:44
[08:35:42][C][wifi:416]: WiFi:
[08:35:42][C][wifi:284]:   SSID: 

[redacted]
[08:35:42][C][wifi:285]:   IP Address: 192.168.0.217
[08:35:42][C][wifi:287]:   BSSID: 

[... 117 lines cut ...]
[08:37:56][D][esp32_ble_tracker:544]: Found device 4C:98:35:08:38:8D RSSI=-88
[08:37:56][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[08:38:21][D][esp32_ble_tracker:544]: Found device 00:EA:24:05:03:27 RSSI=-84
[08:38:21][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[08:38:21][D][esp32_ble_tracker:567]:   Name: 'XY-0327-59'
[08:38:45][D][esp32_ble_tracker:544]: Found device 00:EA:24:06:DC:67 RSSI=-29
[08:38:46][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[08:38:46][D][esp32_ble_tracker:567]:   Name: 'XY-DC67-5F'
[08:38:58][D][esp32_ble_tracker:544]: Found device 7B:19:B9:90:44:9B RSSI=-88
[08:38:58][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[08:38:58][D][esp32_ble_tracker:567]:   Name: 'SENSE  '
[08:38:58][V][esp32_ble_tracker:515]: Unhandled type: advType: 0x08
[08:38:58][D][esp32_ble_tracker:544]: Found device 88:C6:26:9F:76:F5 RSSI=-87
[08:38:58][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[08:38:59][V][esp32_ble_tracker:515]: Unhandled type: advType: 0x08
[08:39:01][V][esp32_ble_tracker:515]: Unhandled type: advType: 0x08
[08:39:12][D][esp32_ble_tracker:544]: Found device 62:93:F7:13:0A:12 RSSI=-85
[08:39:12][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[08:39:12][D][esp32_ble_tracker:569]:   TX Power: 2

```

**Additional information and things you've tried:**

I've checked and double checked I have copied the UUID correctly!  The only other interesting bit of information is that I have three of these XY dongles and they all broadcast the **same UUID** however the have a different major/minor number.  I don't know if this is supposed to be taken into account somehow (I can't see so in the docs)

tjharman, 2020-07-17:
Managed to grab a "VERY_VERBOSE" too.
I also setup a ibeacon advertisement on my other mobile phone to see if it would work, but it isn't found either.

Logs:

```
[09:10:13][I][app:105]: ESPHome version 1.15.0-dev compiled on Jul 18 2020, 09:07:28
[09:10:13][C][wifi:416]: WiFi:
[09:10:13][C][wifi:284]:   SSID: 

[redacted]
[09:10:13][C][wifi:285]:   IP Address: 192.168.0.217
[09:10:13][C][wifi:287]:   BSSID: 

[redacted]
[09:10:13][C][wifi:288]:   Hostname: 'loungeesp'
[09:10:13][C][wifi:292]:   Signal strength: -31 dB ▂▄▆█
[09:10:13][V][wifi:294]:   Priority: 0.0
[09:10:13][C][wifi:296]:   Channel: 6
[09:10:13][C][wifi:297]:   Subnet: 255.255.255.0
[09:10:13][C][wifi:298]:   Gateway: 192.168.0.1
[09:10:13][C][wifi:299]:   DNS1: 192.168.0.1
[09:10:13][C][wifi:300]:   DNS2: 0.0.0.0
[09:10:13][C][logger:185]: Logger:
[09:10:13][C][logger:186]:   Level: VERY_VERBOSE
[09:10:13][C][logger:187]:   Log Baud Rate: 115200
[09:10:13][C][logger:188]:   Hardware UART: UART0
[09:10:13][C][esp32_ble_tracker:530]: BLE Tracker:
[09:10:13][C][esp32_ble_tracker:531]:   Scan Duration: 300 s
[09:10:13][C][esp32_ble_tracker:532]:   Scan Interval: 320.0 ms
[09:10:13][C][esp32_ble_tracker:533]:   Scan Window: 30.0 ms
[09:10:13][C][esp32_ble_tracker:534]:   Scan Type: ACTIVE
[09:10:13][C][ble_rssi:011]: BLE RSSI Sensor 'Dog Tracker'
[09:10:13][C][ble_rssi:011]:   Unit of Measurement: 'dB'
[09:10:13][C][ble_rssi:011]:   Accuracy Decimals: 0
[09:10:13][C][ble_rssi:011]:   Icon: 'mdi:signal-distance-variant'
[09:10:13][C][ble_rssi:011]: BLE RSSI Sensor 'Tims Phone'
[09:10:13][C][ble_rssi:011]:   Unit of Measurement: 'dB'
[09:10:13][C][ble_rssi:011]:   Accuracy Decimals: 0
[09:10:13][C][ble_rssi:011]:   Icon: 'mdi:signal-distance-variant'
[09:10:13][C][ble_rssi:011]: BLE RSSI Sensor 'Tim's Garmin'
[09:10:13][C][ble_rssi:011]:   Unit of Measurement: 'dB'
[09:10:13][C][ble_rssi:011]:   Accuracy Decimals: 0
[09:10:13][C][ble_rssi:011]:   Icon: 'mdi:signal-distance-variant'
[09:10:13][VV][esp32_ble_tracker:326]: Parse Result:
[09:10:13][VV][esp32_ble_tracker:343]:   Address: 1F:14:1A:1D:7F:E4 (RANDOM)
[09:10:13][VV][esp32_ble_tracker:345]:   RSSI: -81
[09:10:13][VV][esp32_ble_tracker:346]:   Name: ''
[09:10:13][VV][esp32_ble_tracker:354]:   Ad Flag: 26
[09:10:13][VV][esp32_ble_tracker:360]:   Manufacturer data: 09.06.03.2E.C0.A8.00.65 (8)
[09:10:13][VV][esp32_ble_tracker:368]: Adv data: 02.01.1A.0B.FF.4C.00.09.06.03.2E.C0.A8.00.65 (15)
[09:10:13][C][captive_portal:169]: Captive Portal:
[09:10:13][C][ota:029]: Over-The-Air Updates:
[09:10:13][C][ota:030]:   Address: loungeesp.local:3232
[09:10:13][C][api:095]: API Server:
[09:10:13][C][api:096]:   Address: loungeesp.local:6053
[09:10:14][VV][esp32_ble_tracker:326]: Parse Result:
[09:10:14][VV][esp32_ble_tracker:343]:   Address: 5F:D6:92:28:DD:2F (RANDOM)
[09:10:14][VV][esp32_ble_tracker:345]:   RSSI: -82
[09:10:14][VV][esp32_ble_tracker:346]:   Name: ''
[09:10:14][VV][esp32_ble_tracker:357]:   Service UUID: FE:9F
[09:10:14][VV][esp32_ble_tracker:360]:   Manufacturer data: 00.73.CA.80.71.A0 (6)
[09:10:14][VV][esp32_ble_tracker:363]:   Service data:
[09:10:14][VV][esp32_ble_tracker:364]:     UUID: FE:9F
[09:10:14][VV][esp32_ble_tracker:365]:     Data: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (20)
[09:10:14][VV][esp32_ble_tracker:368]: Adv data: 03.03.9F.FE.17.16.9F.FE.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.09.FF.E0.00.00.73.CA.80.71.A0 (38)
[09:10:14][D][esp32_ble_tracker:544]: Found device 5F:D6:92:28:DD:2F RSSI=-82
[09:10:14][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[09:10:14][VV][esp32_ble_tracker:326]: Parse Result:
[09:10:14][VV][esp32_ble_tracker:343]:   Address: 1F:14:1A:1D:7F:E4 (RANDOM)
[... 1744 lines cut ...]
[09:11:49][VV][esp32_ble_tracker:326]: Parse Result:
[09:11:49][VV][esp32_ble_tracker:343]:   Address: 88:C6:26:9F:76:F5 (PUBLIC)
[09:11:49][VV][esp32_ble_tracker:345]:   RSSI: -86
[09:11:49][VV][esp32_ble_tracker:346]:   Name: ''
[09:11:49][VV][esp32_ble_tracker:354]:   Ad Flag: 26
[09:11:49][VV][esp32_ble_tracker:357]:   Service UUID: FE:61
[09:11:49][VV][esp32_ble_tracker:360]:   Manufacturer data: 00.60.63.00.00.02.10.4E.89.6E.48.BA.32 (13)
[09:11:49][VV][esp32_ble_tracker:360]:   Manufacturer data: 00.60.63.00.00.02.10.4E.89.6E.48.BA.88.C6.26.9F.76.F5.32 (19)
[09:11:49][VV][esp32_ble_tracker:368]: Adv data: 02.01.1A.02.08.00.03.03.61.FE.10.FF.03.00.00.60.63.00.00.02.10.4E.89.6E.48.BA.32.16.FF.03.00.00.60.63.00.00.02.10.4E.89.6E.48.BA.88.C6.26.9F.76.F5.32 (50)
```

Updated BLE config:

```
sensor:
  - platform: ble_rssi
    service_uuid: '08885DD0-111B-11E4-9191-0800200C9A66'
    name: "Dog Tracker"
  - platform: ble_rssi
    service_uuid: 'EBB1B9BD-9ACE-4C66-B335-D9CC3BA1188F'
    name: "Tims Phone"
  - platform: ble_rssi
    mac_address: 'F0:99:19:2B:F0:43'
    name: "Tim's Garmin"
```

buxtronix, 2020-07-18:
Your service_uuid '08885DD0-111B-11E4-9191-0800200C9A66' doesnt seem to correspond with the service_uuid in the logs? Seems to be from 'manufacturer data' instead.

For this device:

[09:11:48][VV][esp32_ble_tracker:326]: Parse Result:
[09:11:48][VV][esp32_ble_tracker:343]:   Address: 00:EA:24:06:DC:64 (RANDOM)
[09:11:48][VV][esp32_ble_tracker:345]:   RSSI: -66
[09:11:48][VV][esp32_ble_tracker:346]:   Name: 'XY-DC64-5F'
[09:11:48][VV][esp32_ble_tracker:354]:   Ad Flag: 6
[09:11:48][VV][esp32_ble_tracker:357]:   Service UUID: 64:DC:06:24:10:ED:C9:4F:4F:03:D7:9B:06:42:C2:AB
[09:11:48][VV][esp32_ble_tracker:360]:   Manufacturer data: 02.15.08.88.5D.D0.11.1B.11.E4.91.91.08.00.20.0C.9A.66.24.06.DC.64.C4 (23)
[09:11:48][VV][esp32_ble_tracker:368]: Adv data: 02.01.06.1A.FF.4C.00.02.15.08.88.5D.D0.11.1B.11.E4.91.91.08.00.20.0C.9A.66.24.06.DC.64.C4.11.06.64.DC.06.24.10.ED.C9.4F.4F.03.D7.9B.06.42.C2.AB.0B.09.58.59.2D.44.43.36.34.2D.35.46 (60)

You should be using service_uuid: '64DC0624-10ED-C94F-4F03-D79B0642C2AB'

tjharman, 2020-07-18:
Thanks very much @buxtronix - I didn't realise the Service UUID would be different from that found using my Phone's BLE app.

I have put that in instead, but I still don't get anything detected.
I even changed it from an RSSI to just a plain out binary detector, in case there was some code path difference:

```\esp32_ble_tracker:
binary_sensor:
  - platform: ble_presence
    service_uuid: '64DC0624-10ED-C94F-4F03-D79B0642C2AB'
    name: "Dog Tracker"
```

But still nothing triggers - The dog tracker token is always shown as "Off" even though it's camped atop of the ESP32.

Do you have any other suggestions I could try, I appreciate your assistance thus far!

tjharman, 2020-09-26:
Just FYI I've tried this with 1.15.2 and I still can't get a 128bit UUID pulled from the logs in this format:

`04:E4:06:24:10:ED:C9:4F:4F:03:D7:9B:06:42:C2:AB`

To trigger like this:

```
binary_sensor:
  - platform: ble_presence
    service_uuid: '04E40624-10ED-C94F-4F03-D79B0642C2AB'
    name: "Tim's Keys"
```

closed 2021-02-07 by stale[bot]