esphome/issues #1364: esphome::fan::FanState::turn_on function has no effect

uryupinsk, 2020-07-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip on Windows 10 x64 and macOS 10.15 with Python 3.8

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32/lolin32

**ESPHome version (latest production, beta, dev branch)**

v1.14.5

**Affected component:**

[Fan](https://esphome.io/components/fan/index.html?highlight=fan) and [Speed Fan](https://esphome.io/components/fan/speed.html?highlight=fan)

**Description of problem:**

I created a lambda script to change the speed of multiple PWM fans based on temperature. The script is triggered on value change of a Dallas temperature sensor.

Changing the speed of fans by making a call to the `esphome::fan::FanStateCall::set_speed` function is working correctly, even when I force the speed manually on MQTT.

I also make a call on each turn to `esphome::fan::FanState::turn_on` if fan state is OFF. However, this isn't working when I force the state manually in MQTT by issuing `fan/command/OFF`. The state stays OFF.

**Problem-relevant YAML-configuration entries:**
```C
// This is the end of the loop of my lambda script
//
// fan_speed is the ID of the Speed Fan entity
//
// Turn fans state ON if current state is OFF
if ( !id(fan_speed).state ) {
  id(fan_speed).turn_on();
}
```

**Logs (if applicable):**

```
Nothing interesting
```

**Additional information and things you've tried:**

glmnet, 2020-07-20:
I don't understand you

uryupinsk, 2020-07-20:
I check regularly in the code that my fans are in state `ON`. If it's not the case, I use the function described above to set their state to `ON`.

To test this, I am forcing the state to `OFF `by issuing a command with MQTT (`fan/command/OFF`). The function should automatically turn the fans back `ON `on next check, but that's not happening. Fan state stays `OFF`.

glmnet, 2020-07-20:
duplicate of https://github.com/esphome/issues/issues/1359

closed 2020-07-20 by glmnet