esphome/issues #1367: Action triggered only once with Nextion display

baacktothefuture, 2020-07-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Home Hassistant docker raspberrypi3-homeassistant 0.112.5
ESPHome via docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 Wemos D1 mini

**ESPHome version (latest production, beta, dev branch)**
v1.14.5

**Affected component:**
https://esphome.io/components/binary_sensor/nextion.html
https://esphome.io/components/api.html

**Description of problem:** 
When pressing the button created on nextion the first time after boot, log shows the button is  pressed and homeassistant action is called
When pressing the button created on nextion after the first time after boot, log shows the button is  pressed and homeassistant action is NOT called

**Problem-relevant YAML-configuration entries:**
```
  name: nextion
  platform: ESP8266
  board: d1_mini
  on_boot:
    - wait_until:
        - wifi.connected  

logger:
  hardware_uart: UART1
  level: DEBUG

wifi:
  ssid: "XXXXX"
  password: "XXXXX"
  reboot_timeout: 0min

web_server:
  port: 80

api:
  reboot_timeout: 0min

ota:  

uart:
  rx_pin: D7
  tx_pin: D2
  baud_rate: 9600

 
text_sensor:

  - platform: homeassistant  
    id: all_climate_away_mode
    name: "All Climate Away Mode Sensor From Home Assistant"
    entity_id: sensor.all_climate_away_mode     

display:
  - platform: nextion
    id: nextion_display
    update_interval: 300ms

binary_sensor:
  - platform: nextion
    page_id: 0
    component_id: 8
    id: btn_all_climate_away_mode 
    name: "Pulsante Modalità Pulizie Globale"
    on_release:
      - homeassistant.service:
          service: homeassistant.turn_on
          data:
            entity_id: script.toggle_away_mode_all_climate_thermostats
```

**Logs (if applicable):**
```
INFO Reading configuration nextion.yaml...
INFO Starting log output from xxx.local using esphome API
INFO Connecting to xxx.local:6053 (xxx.xxx.xxx.xxx)
INFO Successfully connected to nextion.local
[10:08:52][I][app:100]: ESPHome version 1.14.5 compiled on Jul 21 2020, 10:08:20
[10:08:52][C][wifi:415]: WiFi:
[10:08:52][C][wifi:283]:   SSID: 'xxx'
[10:08:52][C][wifi:284]:   IP Address: xxx
[10:08:52][C][wifi:286]:   BSSID: xxx
[10:08:52][C][wifi:287]:   Hostname: 'nextion'
[10:08:52][C][wifi:291]:   Signal strength: -42 dB ▂▄▆█
[10:08:52][C][wifi:295]:   Channel: 1
[10:08:52][C][wifi:296]:   Subnet: xxx
[10:08:52][C][wifi:297]:   Gateway: xxx
[10:08:52][C][wifi:298]:   DNS1: xxx
[10:08:52][C][wifi:299]:   DNS2: (IP unset)
[10:08:52][C][uart:140]: UART Bus:
[10:08:52][C][uart:142]:   TX Pin: GPIO4
[10:08:52][C][uart:145]:   RX Pin: GPIO13
[10:08:52][C][uart:147]:   Baud Rate: 9600 baud
[10:08:52][C][uart:148]:   Stop bits: 1
[10:08:52][C][uart:152]:   Using software serial
[10:08:52][C][logger:175]: Logger:
[10:08:52][C][logger:176]:   Level: DEBUG
[10:08:52][C][logger:177]:   Log Baud Rate: 115200
[10:08:52][C][logger:178]:   Hardware UART: UART1
[10:08:52][C][homeassistant.text_sensor:011]: Homeassistant Text Sensor 'All Climate Away Mode Sensor From Home Assistant'
[10:08:52][C][homeassistant.text_sensor:012]:   Entity ID: 'sensor.all_climate_away_mode'
[10:08:52][C][captive_portal:169]: Captive Portal:
[10:08:52][C][web_server:123]: Web Server:
[10:08:52][C][web_server:124]:   Address: xxx.local:80
[10:08:52][C][ota:029]: Over-The-Air Updates:
[10:08:52][C][ota:030]:   Address: xxx.local:8266
[10:08:52][C][api:095]: API Server:
[10:08:52][C][api:096]:   Address: xxx.local:6053
[10:09:33][D][nextion:208]: Got touch page=0 component=8 type=PRESS
[10:09:33][D][binary_sensor:036]: 'Pulsante Modalità Pulizie Globale': Sending state ON
[10:13:09][D][nextion:208]: Got touch page=0 component=8 type=PRESS
[10:14:34][I][ota:046]: Boot seems successful, resetting boot loop counter.
```

**Additional information and things you've tried:**
I also tried to remove nextion component and try to perform the action via a physical switch: the action is performed everytime the switch is pressed, so the problem occurs only using the nextion.
Also tried to use different gpios for nextion communication but with no luck.

jhoult, 2020-07-22:
In the nextion editor, did you set it up to send an event on touch press *and* touch release?  Click on the button, then the event window,  and check the box for "send component ID" on both press and release

baacktothefuture, 2020-07-26:
I didn't. Now it works, thank you!

closed 2020-07-26 by baacktothefuture