esphome/issues #1371: ESP32 cannot connect to WiFi reason="Auth Expired"

Mirabis, 2020-07-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Tried both Hassio (Raspberry Pi 4B) and from Windows 10 (python esphome/esptool)

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 DEVKITv1
**ESPHome version (latest production, beta, dev branch)**

v1.14 both from Windows and Hassio

**Affected component:**
https://esphome.io/components/wifi.html

**Description of problem:**
No matter what I do it does not connect to any Wireless network. Even mobile hotspots do not work. Tried changing to legacy mode, disabling 5ghz on same SSID, disabling bandsteering etc. Using Unifi AC AP Pro's.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32_radiator_office
  platform: ESP32
  board: esp32doit-devkit-v1
  arduino_version: dev
  platformio_options:
    platform: espressif32@1.12.4

wifi:
  networks:
  - ssid: "Redacted-IoT"
    password: !secret iot_wifi_password
    bssid: !secret bssid_ap_overloop
    priority: 3
  - ssid: "Redacted-IoT"
    password: !secret iot_wifi_password
    bssid: !secret bssid_ap_woonkamer
    priority: 2
  - ssid: "Redacted-IoT"
    password: !secret iot_wifi_password
    bssid: !secret bssid_ap_zolder
    priority: 1    
  # Other WiFi options here
  domain: .home
  
  reboot_timeout: 15min
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ESP32-32S Office"
    password: "redacted"

status_led:
  pin: GPIO2 # blink slowly = warning, blink quickly = error, no light = OK

# Fix time to be sure    
time:
  - platform: homeassistant
    id: homeassistant_time

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  password: !secret ota_api_password
  reboot_timeout: 5min
  
ota:
  password: !secret ota_api_password

dallas:
[... 2913 lines cut ...]
[20:53:06][V][esp-idf:000]: I (348227) wifi: state: init -> auth (b0)
[20:53:07][V][esp-idf:000]: I (349229) wifi: state: auth -> init (200)
[20:53:07][V][esp-idf:000]: I (349231) wifi: n:11 0, o:11 2, ap:11 2, sta:11 0, prof:11
[20:53:07][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[20:53:07][W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[20:53:07][W][wifi_esp32:336]: Event: Disconnected ssid='REDACTED-IoT' bssid=F0:9F:C2:74:EC:10 reason='Auth Expired'
[20:53:07][W][wifi:451]: Error while connecting to network.
[20:53:07][W][wifi:488]: Restarting WiFi adapter...
[20:53:07][V][wifi_esp32:035]: Disabling STA.
[20:53:07][V][esp-idf:000]: I (349272) wifi: mode : softAP (24:62:ab:ff:8b:dd)
[20:53:07][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[20:53:07][V][wifi_esp32:315]: Event: WiFi STA stop

```

**Additional information and things you've tried:**

- Running without sensors
- Changing from multiple networks defined to just 1
- Fastconnect true/false
- Changing wifi to hidden (both AP and config)
- Building from both Hassio (pi) and Windows
- Erasing entire flash first multiple times before flashing firmware
- powering from 5v usb
- powering from 3.3v and gnd

mamartins, 2020-08-02:
I’m having a similar issue, in my case I have 2 unifi ac pro APs and the device can only connect to one of them.

h0jeZvgoxFepBQ2C, 2020-08-13:
I'm having similar issues, not only with the esp32 but also esp8266. Also having Unifi AC PRO AP, but also one magenta modem with 2,4 Ghz Wifi.

Mirabis, 2020-12-23:
I ended up replacing the ESP's and the newer ones connected without issues. Not sure what the culprit is/was.

closed 2021-05-06 by stale[bot]

cromulus, 2021-05-17:
I'm having the same issue. m5stack atom lite.

Francisk0, 2021-06-15:
same problem

nickw444, 2021-06-21:
Likewise, been having this issue with both of my MagicHome LED Strip Controllers flashed with ESPHome.

FWIW I am running UniFi Nano HD single AP in my apartment, which appears to be the culprit, as re-assigning to a test network running off of a Netgear Modem/Router/AP allows it to connect without issues.

I've tried all kinds of tweaks to the AP settings to no avail unfortunately. Unsure whether this is truly an ESP problem, or something more sinister in the UAP firmware.

```
[22:26:05][I][wifi:235]: WiFi Connecting to '<redacted>'...
[22:26:05][W][wifi_esp8266:444]: Event: Disconnected ssid='<redacted>' bssid=<redacted> reason='Authentication Failed'
[22:26:05][W][wifi:532]: Error while connecting to network.
[22:26:05][I][wifi:235]: WiFi Connecting to '<redacted>'...
[22:26:10][W][wifi_esp8266:444]: Event: Disconnected ssid='<redacted>' bssid=<redacted> reason='Auth Expired'
[22:26:10][W][wifi:532]: Error while connecting to network.
```

cromulus, 2021-06-23:
For what it’s worth, I also have UniFi kit. The new nano wifi6 AP. Everything worked with my older UniFi equipment at my prior home. Switched to new stuff when we moved and it doesn’t seem to work anymore. Let me see if I can fiddle with the settings on my iot network.

nickw444, 2021-06-24:
> The new nano wifi6 AP.

I'm on a Wifi AC Wave 2 (UAP Nano HD). Was your old gear AC wave 2? Possibly something which affects Unifi AC Wave 2 and also WiFi 6.

> Let me see if I can fiddle with the settings on my iot network.

I fiddled with pretty much every setting on my Wifi Network. Made a separate "Smart" SSID, turned off all other SSIDs to rule them out. Fiddled with the settings without any luck unfortunately. Enabling `Enable multicast enhancement (IGMPv3)` seemed to improve connection speed when running Tasmota, but I think that was just sheer luck.

sebbalex, 2021-07-16:
same here, with unifi kit also, I have UniFi UAP-AC Lite

cromulus, 2021-07-16:
Huh. Super weird. Let’s bring this up with Ubiquiti. Perhaps they can figure it out?

nickw444, 2021-07-17:
> Let’s bring this up with Ubiquiti.

I'm not convinced this is a problem on the Unifi side, given I'm running Tasmota without problems on the two chips which were not connecting when running ESPHome...

cromulus, 2021-07-17:
For what it’s worth, these settings on the iOS unifi app seemed to make things work with my M5stack atom lites:
[image]

I think the “optimize iot WiFi connectivity” is the one that did the trick.  Don’t know what it actually does.

zewelor, 2021-09-09:
Has the same problem, after hours of searching finally got this solution: https://github.com/espressif/arduino-esp32/issues/2144#issuecomment-701739289 . So far seems to work. Finger crossed !