esphome/issues #1372: Thermostat: Logging nan instead of the default target temperature

edenhaus, 2020-07-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Docker

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266

**ESPHome version (latest production, beta, dev branch)** 1.15-dev

**Affected component:** thermostat

**Description of problem:**
The component is not logging correctly the default target temperature and other temperatures. It is showing nan (not a number) instead (See logs below).

**Problem-relevant YAML-configuration entries:**
```yaml
climate:
  - platform: thermostat
    name: ${name} Heizung
    sensor: temperature
    default_target_temperature_low: 20 °C
    away_config:
      default_target_temperature_low: 15 °C
    heat_action:
      - switch.turn_on: heater
    idle_action:
      - switch.turn_off: heater
```

**Logs (if applicable):**

```
[18:57:05][C][thermostat.climate:509]: Thermostat 'Hobbyraum Heizung'
[18:57:05][C][thermostat.climate:511]:   Default Target Temperature Low: nan°C
[18:57:05][C][thermostat.climate:514]:   Hysteresis: 0.5°C
... (Removed as irrelevant)
[18:57:05][C][thermostat.climate:534]:   Supports AWAY mode: YES
[18:57:05][C][thermostat.climate:537]:     Away Default Target Temperature Low: nan°C
```

**Additional information and things you've tried:**
I tried also to specify only the number without unit, but this also didn't worked.

@kbx81 Can you give it a look? If you need other details or someone to test, please let me know.
Thanks in advance

kbx81, 2020-07-24:
Yup, I already see what the issue is. I'll do a PR to fix the logging this weekend. 😄

kbx81, 2020-07-26:
@edenhaus if you have a moment, please confirm if the issue is fixed for you here: https://github.com/esphome/esphome/pull/1201

closed 2020-07-26 by OttoWinter

edenhaus, 2020-07-27:
sorry for the later answer. I can also confirm that it's working. @kbx81 thanks :)