esphome/issues #1375: FastLed clockless crashes the esp32

mvturnho, 2020-07-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):** 
Esp32 T-Display unit from TTGO

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.15.0b1

**Affected component:**

FastLed clockless

https://next.esphome.io/components/light/fastled.html

**Description of problem:**
I use the simplest esphome fastled config as possible, when the device reoots after flash it crashes. I tested the same config under the current 1.14.5 then there is no issue.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: tdisplay
  upper_devicename: TDISPLAY

esphome:
  name: ${devicename}
  platform: ESP32
  board: nodemcu-32s
  platformio_options:
    upload_speed: 921600
    monitor_speed: 19200

logger:
    level: VERBOSE
  
wifi:
  ssid: !secret wifi_kelpnet_up_ssid
  password: !secret wifi_kelpnet_password
  reboot_timeout: 0s

api:
  password: 'secret'
  reboot_timeout: 0s

light:
  - platform: fastled_clockless
    chipset: WS2811
    pin: 17
    num_leds: 1
    rgb_order: GRB
    name: ${upper_devicename}" status_led"

```

**Logs (if applicable):**

```
Hard resetting via RTS pin...
========================= [SUCCESS] Took 20.47 seconds =========================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[13:30:01]\x82\xea\xe9 I (361) wifi: wifi firmware version: 7997e4b
[13:30:01]
[13:30:01][V][esp-idf:000]: I (362) wifi: config NVS flash: enabled
[13:30:01]
[13:30:01][V][esp-idf:000]: I (363) wifi: config nano formating: disabled
[13:30:01]
[13:30:01][V][esp-idf:000]: I (370) wifi: Init dynamic tx buffer num: 32
[13:30:01]
[13:30:01][V][esp-idf:000]: I (377) wifi: Init data frame dynamic rx buffer num: 32
[13:30:01]
[13:30:01][V][esp-idf:000]: I (384) wifi: Init management frame dynamic rx buffer num: 32
[13:30:01]
[13:30:01][V][esp-idf:000]: I (392) wifi: Init management short buffer num: 32
[13:30:01]
[13:30:01][V][esp-idf:000]: I (400) wifi: Init static rx buffer size: 1600
[13:30:01]
[13:30:01][V][esp-idf:000]: I (406) wifi: Init static rx buffer num: 16
[13:30:01]
[13:30:01][V][esp-idf:000]: I (413) wifi: Init dynamic rx buffer num: 32
[13:30:01]
[13:30:01][V][esp-idf:000]: I (588) wifi: mode : sta (24:6f:28:96:50:18)
[13:30:01]
[13:30:01][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[13:30:01][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[13:30:01][V][wifi_esp32:361]: Event: WiFi STA start
[13:30:01][V][wifi_esp32:352]: Event: WiFi ready
[13:30:01][V][esp-idf:000]: I (615) wifi: Set ps type: 1
[13:30:01]
[13:30:01]
[13:30:01][D][wifi:305]: Starting scan...
[13:30:01]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[13:30:01]Core 1 register dump:
[13:30:01]PC      : 0x400ddc59  PS      : 0x00060031  A0      : 0x800851c5  A1      : 0x3ffbe770  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /home/esphome/dev-config/tdisplay -t idedata
WARNING Decoded 0x400ddc59: ClocklessController<17, 77, 77, 154, (EOrder)66, 0, false, 5>::doneOnChannel(rmt_channel_t, void*) at /home/esphome/dev-config/tdisplay/.piolibdeps/tdisplay/FastLED/controller.h:171
[13:30:03]A2      : 0x00000000  A3      : 0x3ffc1378  A4      : 0x00000000  A5      : 0x00000000  
[13:30:03]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80089924  A9      : 0x3ffbe770  
[13:30:03]A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000000  
[13:30:03]A14     : 0x00060023  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
[13:30:03]EXCVADDR: 0x0000001c  LBEG    : 0x40154d49  LEND    : 0x40154d62  LCOUNT  : 0x00000000  
[13:30:03]Core 1 was running in ISR context:
[13:30:03]EPC1    : 0x400ddc59  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40086fd1
[13:30:03]
[13:30:03]Backtrace: 0x400ddc59:0x3ffbe770 0x400851c2:0x3ffbe7a0 0x4008494d:0x3ffbe7d0 0x4008c4ac:0x3ffb1e10 0x400e0b57:0x3ffb1e30 0x400dc51d:0x3ffb1e50 0x400ddaea:0x3ffb1ea0 0x400e154f:0x3ffb1fb0 0x40089c5d:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400ddc59: ClocklessController<17, 77, 77, 154, (EOrder)66, 0, false, 5>::doneOnChannel(rmt_channel_t, void*) at /home/esphome/dev-config/tdisplay/.piolibdeps/tdisplay/FastLED/controller.h:171
WARNING Decoded 0x400851c2: rmt_driver_isr_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c:862
WARNING Decoded 0x4008494d: _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
WARNING Decoded 0x4008c4ac: _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:435
WARNING Decoded 0x400e0b57: yield at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:170
WARNING Decoded 0x400dc51d: esphome::Application::setup() at /home/esphome/dev-config/tdisplay/src/esphome/core/application.cpp:55
WARNING Decoded 0x400ddaea: setup() at /home/esphome/dev-config/tdisplay/.piolibdeps/tdisplay/FastLED/controller.h:171
WARNING Decoded 0x400e154f: loopTask(void*) at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:14
WARNING Decoded 0x40089c5d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[13:30:04]
[13:30:04]Rebooting...

```

**Additional information and things you've tried:**

Yaml config works under esphome 1.14.5

OttoWinter, 2020-07-28:
Ok, looks like a bug in FastLED. I bisected on the version and the issue first appeared in 3.3.3

I'll downgrade to 3.3.2

Probably related to some of the changes here: https://github.com/FastLED/FastLED/compare/be47737f0481ab02362987fb46ffe10660d64495..fefc5f563a33963b9998f0475ad050813b30adcb#diff-ddee7bebf2789191d7022a173e2771a8R117

OttoWinter, 2020-07-28:
If someone could look into this issue more and report this to FastLED, that would be great! I don't have the time currently to do that.

closed 2020-07-28 by OttoWinter

keukenrol, 2020-08-30:
I also noticed my ESP32 using FastLED crashing since the last verison. I'll downgrade to 3.3.2