esphome/issues #1376: upgrading to 1.14.5 killed sonoff basic wifi

bensuffolk, 2020-07-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASS Supervisor running in Docker, not HASS.IO O/S

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic ESP8266, Revision 1.1

**ESPHome version (latest production, beta, dev branch)**

V1.14.5

B1.**Affected component:**

**Description of problem:**

I was using 1.14.3, lots of Sonoff  Basics running fine.  I know that in that version I had a problem sometimes with WiFi when I upgraded to 1.14.3 due to the power output issue, so I already has the 'output_power: 17db' line in which let them work fine.

After this upgrade (Single unit only as a test) I find it will not connect to the WiFi

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bathroom_ls
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  output_power: 17db

logger:

api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    filters:
      - delayed_on: 100ms
      - delayed_off: 100ms
    id: button
    on_state:
      then:
        - light.toggle: light_1
  - platform: status
    name: "Bathroom LS Status"

output:
  - platform: gpio
    pin: GPIO12
    id: relay_1

light:
  - platform: binary
    name: "Bathroom"
    id: light_1
    output: relay_1

status_led:
  pin:
    number: GPIO13
    inverted: yes

```

**Logs (if applicable):**

```
[17:01:22][I][wifi:193]: WiFi Connecting to 'HASS'...
[17:01:25][W][wifi_esp8266:391]: Event: Disconnected ssid='HASS' reason='Probe Request Unsuccessful'
[17:01:25][W][wifi:451]: Error while connecting to network.
[17:01:25][W][wifi:488]: Restarting WiFi adapter...
[17:01:25][I][wifi:193]: WiFi Connecting to 'HASS'...
[17:01:28][W][wifi_esp8266:391]: Event: Disconnected ssid='HASS' reason='Probe Request Unsuccessful'

....

[17:03:58][I][app:127]: Forcing a reboot...
[17:03:58]
[17:03:58] ets Jan  8 2013,rst cause:1, boot mode:(3,7)
[17:03:58]
[17:03:58]load 0x4010f000, len 1384, room 16 
[17:03:58]tail 8
[17:03:58]chksum 0x2d
[17:03:58]csum 0x2d
[17:03:58]v8b899c12
[17:03:58]~ld
[17:03:58][I][logger:156]: Log initialized
[17:03:58][C][status_led:014]: Setting up Status LED...
[17:03:58][C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[17:03:58][I][app:028]: Running through setup()...
[17:03:58][C][light:097]: Setting up light 'Bathroom'...
[17:03:58][D][light:264]: 'Bathroom' Setting:
[17:03:58][D][binary_sensor:034]: 'Bathroom LS Status': Sending initial state OFF
[17:03:58][C][wifi:033]: Setting up WiFi...
[17:03:58][I][wifi:193]: WiFi Connecting to 'HASS'...
[17:03:58][D][binary_sensor:034]: 'button': Sending initial state OFF
[17:04:01][W][wifi_esp8266:391]: Event: Disconnected ssid='HASS' reason='Probe Request Unsuccessful'
[17:04:01][W][wifi:451]: Error while connecting to network.
[17:04:01][W][wifi:488]: Restarting WiFi adapter...
[17:04:01][I][wifi:193]: WiFi Connecting to 'HASS'...
[17:04:04][W][wifi_esp8266:391]: Event: Disconnected ssid='HASS' reason='Probe Request Unsuccessful'

```

**Additional information and things you've tried:**

It simply loops through the above error logs repeatedly.  

I have seen a similar issue raised for 1.14.4, but the answer to that was the wifi power, which I already had set (which I had to do to get it to work with 1.14.3 in the first place so I do not think its the same issue again). 

I have reduced the power level down to 15db and all the way down to 10db, with the same results.

Changed logger to only show INFO level not the default DEBUG and that has worked and the sonoff connects to the WiFI as it should.

The same code runs find on a Revision 2 Sonoff Basic

Using fast_connect because its a hidden network I connect to

lustefaniak, 2020-07-28:
I can confirm that, upgrade to 1.14.5 from 1.14.4 killed wifi on 2 sonoff basic and 2 sonoff pow r2.

glmnet, 2020-07-28:
This error showed up here too: https://github.com/esphome/issues/issues/952
This was before version 1.14.4 so most likely related to those power / program corruption on the flash.

between 1.14.3 and 1.14.4 there were a few changes, DHT / Dallas, also some WiFi hostname stuff there.
between 1.14.4 and 1.14.5 we only fixed [an issue which is related to ESP32 only.](https://github.com/esphome/esphome/commit/bf0cce4ad8e45b6f0b0b331263fd2a99abf45d6b)

bensuffolk, 2020-07-28:
So there is another so off basic I have that just used as a remote relay. Setting logger to info did not work on this one, neither did changing the output power to lower than 17db.

The only way I could get this one to connect was to remove the logger completely.

What exactly is the power / corruption issue? Just for reference I have tried the same bi are flashed directly and ota, no difference if the corruption is about flashing via WiFi

manuelbarbosa, 2020-07-29:
I had the same issue, fixed it by removing some components for the Sonoff Basics from EspHome:

- removed mqtt
- removed wifi signal
- removed uptime

And they started working properly

lustefaniak, 2020-07-30:
For me problems usually happen when I enable `time` component, no matter if it is `smtp` or recently (like this current time) `ha`.

It looks like there were some changes to power_output default value (https://github.com/esphome/esphome/commit/072cd5b83ec51f81a68fc6f7969b9e48524caa8b), things worked for me well when limit was downed to 17db. I didn't have explicit value set this time and maybe this is source of the problem as in the past - OTA corrupts firmware when too high power is used?

bapowell, 2020-08-16:
Same issue, with Sonoff Basic, using "--pre" ESPHome v1.15.0b3.

Fixed by removing both of the following:
* captive_portal
* sensor: wifi_signal

vobo70, 2020-09-14:
Are there any chance that will be fixed in next release?

user897943, 2020-09-25:
I am still running 1.13.6 because any build after that seems to have wifi issues. It's not an ethernet device so having to remove it from the electric cable, solder a serial flasher, connect to computer, etc is a pain in the ass, why can't the developers figure this out? I think it sucks that it's been like this for more than a year and nobody cares on the development team. Sonoff basic is a very popular and cheap device, this should get the attention it deserves. I do feel it is wrong to keep releasing stuff that breaks user's devices. Yes, it breaks them, nobody should have to flash using a serial to usb adapter after the initial flash. Bricking devices is bad development, I am calling this out, it doesn't matter if the developers are giving their time for free, it's still a product and shouldn't break things.

vobo70, 2020-09-25:
for me 1.15.2 works without any problems

user897943, 2020-09-25:
> for me 1.15.2 works without any problems

Could you please share your yaml configuration for your sonoff basic?

vobo70, 2020-09-25:
esphome:
  name: grzalka_zasobnik
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "voboIOT"
  password: "mE@5yVy%7P?QTSLy"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "grzalka zasobnik"
    password: "p30aQHrGWdMx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "h8R>AXxUY2ZMbkau"

ota:
  password: "h8R>AXxUY2ZMbkau"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "przycisk grzalka zasobnik"
    on_press:
      - switch.toggle: relay_grzalka_zasobnik

switch:
  - platform: gpio
    name: "przekaznik grzalka zasobnik"
    pin: GPIO12
    id: relay_grzalka_zasobnik

status_led:
  pin:
    number: GPIO13
    inverted: yes

vobo70, 2020-09-25:
esphome:
  name: grzalka_zasobnik
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "xxxxxxxx"
  password: "xxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "grzalka zasobnik"
    password: "xxxxxxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxxxxx"

ota:
  password: "xxxxxxxx"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "przycisk grzalka zasobnik"
    on_press:
      - switch.toggle: relay_grzalka_zasobnik

switch:
  - platform: gpio
    name: "przekaznik grzalka zasobnik"
    pin: GPIO12
    id: relay_grzalka_zasobnik

status_led:
  pin:
    number: GPIO13
    inverted: yes

user897943, 2020-09-25:
Just in case, I'd advise you to change your passwords because despite deleting the post it still went through via email.

vobo70, 2020-09-25:
thank you - you're right

user897943, 2020-09-25:
You're welcome and thanks for posting the config... I'll try to update one and see how it goes

bapowell, 2020-11-22:
Results from testing my Sonoff Basic, using "dev" branch ESPHome v1.16.0.dev0 (11/21/2020):

Re-enabling the following works:

-     captive_portal
-     sensor: wifi_signal

Adding back in the "time" component partially works. The following (without an on_time automation) works fine:

time:
  - platform: homeassistant
    id: hass_time
    timezone: "America/Chicago"

But when I add an on_time automation, the Sonoff become non-functional:
...
    on_time:
      # Every night at 11:59pm
      - seconds: 0
        minutes: 59
        hours: 23
        then:
          - light.turn_off: light_relay

closed 2021-04-05 by stale[bot]