esphome/issues #1380: permission denied

s1nbad, 2020-07-29:
ubuntu 20.04 pip3 install
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

N/A
**ESP (ESP32/ESP8266, Board/Sonoff):**

N/A
**ESPHome version (latest production, beta, dev branch)**

latest
**Affected component:**

N/A

**Description of problem:**
followed exactly :- https://esphome.io/guides/getting_started_command_line.html

To start the ESPHome dashboard, simply start ESPHome with the following command 
esphome config/ dashboard

error: error creating directories /config/.esphome: [Errno 13] permission denied: /config

i note: (with config/ pointing to a directory where you want to store your configurations)
how do you "point" to a directory ? 
(perhaps an example would be useful in the docs)

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2020-07-29:
With the `config/` argument you tell esphome where to store the dashboard files. It's relative to the current directory you're in.

In your case, you're executing the command in the root directory `/`, and your user doesn't have permissions to write to the `/config` directory.

You can instead go to your home directory (by typing `cd ~`) or wherever you want to store the files, then execute `esphome config/ dashboard`

s1nbad, 2020-07-29:
ok, i can sort out permissions but to be clear -  /config refers to a directory as opposed to a esphome switch

eg:  esphome /any/directory dashboard   would attempt to write to /any/directory

glmnet, 2020-07-29:
Yes. You're right. It's a bit counterintuitive  the 'program location verb switches' order

Btw. If you're in the config folder you can use `.`

s1nbad, 2020-07-29:
understood, thanks for your help

closed 2020-07-29 by s1nbad