esphome/issues #1381: RGBWW and Strobe Effect: Can't control warm/cold white independently

Adrian-at-CrimsonAuzre, 2020-07-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

Strobe Effect

**Description of problem:**
I cannot control warm and cold white independently on an RGBWW light with the Strobe Effect (looking at the documentation I think the actions are also affected).

**Problem-relevant YAML-configuration entries:**

I would expect this to work, but `cold_white` and `warm_white` don't exist, only `white`. I would also like to see a color temperature option if possible, that would make pulsing different temperatures easier as a sort of "notification" to get your attention.
```yaml
light:
  - platform: rgbww
    name: "RGB CCT Test "
    red: R
    green: G
    blue: B
    cold_white: CW
    warm_white: WW
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    effects:
    - strobe:
        name: Warm/Cold flicker
        colors:
          - brightness: 100%
            red: 0%
            green: 0%
            blue: 0%
            cold_white: 0%
            warm_white: 100%
            duration: 50ms
          - brightness: 100%
            red: 0%
            green: 0%
            blue: 0%
            cold_white: 100%
            warm_white: 0%
            duration: 50ms

```

OttoWinter, 2020-07-30:
Yeah color temperature could be added there.

But cold_white, warm_white can not be supported. ESPHome chose to use Home Assistant's color model, which uses `color_temperature`, and it's up to the integration how to translate that to cold/warm white percentage, thus effects can't define cold/wnite percentage directly.

Adrian-at-CrimsonAuzre, 2020-07-30:
Yeah, that'll work. I don't really have a need for it _yet_ but I have ideas that would probably benefit from strobing color temperature.

closed 2020-12-11 by stale[bot]