esphome/issues #1382: Clear RTS and DTR in serial console

jbliesener, 2020-07-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10, pip installation

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-07 (512K Flash)

**ESPHome version (latest production, beta, dev branch)**

1.14.5

**Affected component:**

https://esphome.io/guides/getting_started_command_line.html

**Description of problem:**
I'm using the ESP-07 with only 512K Flash, so I can't use OTA updates, which requires me to use the serial cable. Upload works fine, but after uploading I need to disconnect the RTS and DTR pins to the serial converter, because both lines are held active (low) in the serial terminal and therefore the ESP doesn't start. As soon as I quit the serial terminal from esphome or disconnect RTS and DTR, the program starts.

**Problem-relevant YAML-configuration entries:**
```yaml
NOT RELATED TO YAML

```

**Logs (if applicable):**

```
NO LOG

```

**Additional information and things you've tried:**

I would suggest to change the code of `run_miniterm` to something like this:

```python
    ...
    backtrace_state = False
    # Not specifying a port name does not open the port
    with serial.Serial(baudrate=baud_rate) as ser: 
      ser.port = port
      ser.rts = False
      ser.dtr = False
      ser.open()  # open port, set DTR and RTS inactive
      while True:
      ...
```

jbliesener, 2020-11-27:
Still open

jbliesener, 2021-07-26:
Any chance to get this implemented?

agners, 2021-07-28:
Hm, I have a similar case where RTS# (CHIP_EN) and DTR# (GPIO9/boot loader strapping) are directly connected.

[image]

Normally there is a logic which makes modules ignore active RTS/DTR signals ("active" is low since those are low active signals).

Anyways, unfortunately it seems that pyserial at least in Linux causes a change of RTS/DTR signals when trying to set them to False by default (high). On modules with the regular logic this leads to resets according to my testing.

But maybe adding it as an option makes sense.

agners, 2021-07-28:
I wasn't able to set RTS/DTR false without causing a toggling (and hence a reset on other ESP boards).

See also: https://github.com/pyserial/pyserial/issues/124

closed 2021-08-10 by jesserockz