esphome/issues #1384: GPIO Binary Sensor initiated in unexpected state, or even no state at all

Basanon, 2020-07-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS 4.11, hassio 0.113.2, pi4

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-WROOM-32D, pcf8575

**ESPHome version (latest production, beta, dev branch)**

ESPhome v1.14.5

**Affected component:**

GPIO Binary Sensor https://esphome.io/components/binary_sensor/gpio.html
GPIO Extender pcf8575 https://esphome.io/components/pcf8574.html

**Description of problem:**
I'm using multiple reed sensors and switches in my project, so have to use two GPIO extenders to feet all of them, among some LED lights as well (20 reed, 20 switch, 10 LED) 
All of them can be fitted with one ESP32 WROOM and two pcf8575 GPIO extenders

I'm noticed what GPIO16 and GPIO17 is always initiated in wrong "ON" state, instead "OFF" as all other GPIO's, with provided below YAML  
Nothing attached to any GPIOs so far
Here is a screenshot from ESP32 Web interface
[image]

In this example only GPIO on extender affected, but in case Im adding some extra ESP32 GPIO in that YAML, they will show up in "empty" state as well 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32reed
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "WiFi"
  password: "Pass"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.79
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32reed Fallback Hotspot"
    password: "Pass2"

captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
web_server:
  port: 80

# Enable Home Assistant API
api:

ota:

i2c:
  - id: ext_1
    sda: 27
[... 543 lines cut ...]
[01:08:12][VV][i2c:113]:     Received 0b11111111 (0xFF)
[01:08:12][VV][i2c:083]: Requesting 2 bytes from 0x20:
[01:08:12][VV][i2c:113]:     Received 0b11111111 (0xFF)
[01:08:12][VV][i2c:113]:     Received 0b11111111 (0xFF)
[01:08:12][VV][i2c:083]: Requesting 2 bytes from 0x21:
[01:08:12][VV][i2c:113]:     Received 0b11111111 (0xFF)
[01:08:12][VV][i2c:113]:     Received 0b11111111 (0xFF)
[01:08:12][VV][i2c:083]: Requesting 2 bytes from 0x20:
[01:08:12][VV][i2c:113]:     Received 0b11111111 (0xFF)
[01:08:12][VV][i2c:113]:     Received 0b11111111 (0xFF)
[01:08:12][VV][i2c:083]: Requesting 2 bytes from 0x21:
[01:08:12][VV][i2c:113]:     Received 0b11111111 (0xFF)
[01:08:12][VV][i2c:113]:     Received 0b11111111 (0xFF)
[01:08:12][VV][api.service:271]: on_list_entities_request: ListEntitiesRequest {}
[01:08:12][V][app:076]: A component took a long time in a loop() cycle (0.48 s).
[01:08:12][V][app:077]: Components should block for at most 20-30ms in loop().
[01:08:12][C][i2c:028]: I2C Bus:
[01:08:12][C][i2c:029]:   SDA Pin: GPIO33
[01:08:12][C][i2c:030]:   SCL Pin: GPIO32
[01:08:12][C][i2c:031]:   Frequency: 50000 Hz
[01:08:12][I][i2c:033]: Scanning i2c bus for active devices...

```

**Additional information and things you've tried:**

OttoWinter, 2020-07-30:
Duplicate of https://github.com/esphome/issues/issues/806

Basanon, 2020-07-30:
Its not about "web server page" its about wrong initial state of sensor
Its in "ON" state even in Home Assistant

glmnet, 2020-07-30:
Yes this one seems different, a lots of IOs but more importantly this is an ESP32, the other issue is ESP8266 centric.

I believe there are two issues here, there might be one regarding the queue filling up at 32 on the ESP32, the other deserves more analysis, I haven't tested the module but `INPUT` as is is not pulled high nor low for a code point of view and if you have nothing connected there (a path to ground or to vcc via a resistor) then there is no guarantee that the read value will be ON or OFF.

Can you try connecting some pull down resistor and a button to check if it works?

glmnet, 2020-07-30:
Also use a Home Assistant API interface to debug too, don't relay on WebServer only otherwise we don't know where the use really is.

Basanon, 2020-07-30:
> Can you try connecting some pull down resistor and a button to check if it works?

Yes, I can do that
But eventually on 20 GPIO will be connected reed sensors, which mean simple wire between GPIO pin and ground to get ON state, and no connection - OFF state.
I already tested this config with two reed and two relay (switch in YAML), that prototype worked like a charm

Now, with full amount of sensors and switches I got that strange behavior 
But, I notice what If I reorganize GPIO, connect less binary sensors to extender, but more to ESP32 itself, it stop showing ON, but still empty on web (this is another issue)
Did not test it well yet, but seems to be it will work. Will put update here, as soon tests will be finished

I notice couple more issues too (not critical for me, coz easily can be avoided)
1) Name of sensor or switch should be latin only, in case of Cyrillic it show unexpected result on web interface - 10 switches, toggle button on first only, and it changes ON/OFF state on multiple switches
2) Same YAML on 1.5 dev version coursing ESP32 reboot in circle. After 10 reboots device switches to safe mode.

May be it will explain a bit
My project is a window shutter automation with HA and ESPHome. I have 9 windows and one terrace door.
Two reed sensors for shutter to detect its opened or closed (reed in a wall, magnet on the edge of curtain)
Two relay per shutter, to enable close/open shutter motor rotation
And 10 led to reflect current state of each shutter 
Manual control available, with HA priority (manual will be ignored in case different command from HA available)
Protection from open/close command at the same time included in HA and manual control (rotary control, not 2 buttons)

Basanon, 2020-08-03:
Just tested extenders GPIO 10-17 only on both of them

pins 16 and 17 on extenders does not work
all 16 switches on HA changing state, but 16ths and 17ths does not affect relays
10-15 GIPIOs works fine, relay switches as it should

glmnet, 2020-08-03:
PCF8575 are 16 channels, you are going from A0 to A17 which are 18 channels. What I'm missing here?

glmnet, 2020-08-03:
May be pin numbering is confusing you, see here https://www.ti.com/lit/ds/symlink/pcf8575.pdf?ts=1596476589258&ref_url=https%253A%252F%252Fwww.google.com%252F where they skip P08 and P09 thus having only 16 channels even though there are P0 and P17

In any case P10-P17 might be mapped to channel number 8-15 in ESPHome

Basanon, 2020-08-03:
> In any case P10-P17 might be mapped to channel number 8-15 in ESPHome

Exactly
There is mismatch between module marked GPIO (0:7 & 10:17) and ESPHome GPIO (0-15) for PCF8575
Discovered that after adding third extender :) Now will remove it...

Basanon, 2020-08-04:
Finally whole project tested
Here is YAML
```

esphome:
  name: esp32reed
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "WiFi"
  password: "Pass"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.79
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Reed Fallback Hotspot"
    password: "Pass2"

captive_portal:

# Enable logging
logger:
#  level: VERY_VERBOSE

# Enable Home Assistant API
web_server:
  port: 80

# Enable Home Assistant API
api:

ota:

i2c:
  - id: ext_1
    sda: 27
    scl: 26
    scan: True 
  - id: ext_2
    sda: 33
    scl: 32
    scan: True 
 
    
pcf8574:
  - id: 'ext_hub_A'
    i2c_id: ext_1
    address: 0x20
    pcf8575: True   
  - id: 'ext_hub_B'
    i2c_id: ext_2
    address: 0x21
    pcf8575: True   
   
binary_sensor:

  - platform: gpio
    id: sensor_closed_shutter_6 
    name: "Closed Shutter 6"
    pin: 
      pcf8574: ext_hub_A
      number: 0
[... 494 lines cut ...]
    chipset: WS2812
    pin: GPIO13
    num_leds: 1
    rgb_order: RGB
    name: "Shutter 10 LED"  
  - platform: fastled_clockless
    chipset: WS2812
    pin: GPIO12
    num_leds: 1
    rgb_order: RGB
    name: "Shutter 9 LED"  
  - platform: fastled_clockless
    chipset: WS2812
    pin: GPIO14
    num_leds: 1
    rgb_order: RGB
    name: "Shutter 8 LED"  
  - platform: fastled_clockless
    chipset: WS2812
    pin: GPIO25
    num_leds: 1
    rgb_order: RGB
    name: "Shutter 7 LED"  

```

Basanon, 2020-08-04:
Seems to be that was not issue, but difference between PCF8575 module GPIO marks and ESPHome GPIO numbers
[image]

Basanon, 2020-08-04:
I think if this info will appear on this page https://esphome.io/components/pcf8574.html that will be great!
Thanks

closed 2020-08-04 by Basanon