esphome/issues #1385: sun sensor returns NaN when time: homeassistant with last dev build

SeByDocKy, 2020-07-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Supervised HA (Odroid C4) w/ HA 0.103.2

Please provide details about your environment below this line. -->

ESP8266 NodeMCU V3

**ESPHome version (latest production, beta, dev branch)**

Dev build  1.16 from 07/30/2929

sun sensor (both azimuth & elevation)

**Description of problem:**

sun sensors return NaN values when time object is set to homeassistant (was working fine few days ago). Works again when set to sntp

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: suiveur_1axe
  platform: ESP8266
  board: nodemcuv2
  includes:
    - BMM150_custom_sensor.h

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Suiveur 1Axe Fallback Hotspot"
    password: "O6hc3FK8UPsh"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  services:
    - service: control_stepper
      variables:
        target: int
      then:
        - stepper.set_target:
            id: stepper_suiveur_1axe
            target: !lambda 'return target;'

ota:

sun:
  latitude: !secret esphome_lat
  longitude: !secret esphome_long 
  
time:
  - platform: homeassistant #sntp
    id: homeassistant_time  

output:
  - platform: gpio
    pin:
      number: GPIO02
      inverted: True
    id: gpio_02
    
[... 458 lines cut ...]
[09:20:38][C][ota:029]: Over-The-Air Updates:
[09:20:38][C][ota:030]:   Address: suiveur_1axe.local:8266
[09:20:38][D][api.connection:604]: Client 'Home Assistant 0.113.2 (192.168.1.20)' connected successfully!
[09:20:39][C][api:095]: API Server:
[09:20:39][C][api:096]:   Address: suiveur_1axe.local:6053
[09:20:39][C][sntp:043]: SNTP Time:
[09:20:39][C][sntp:044]:   Server 1: '0.pool.ntp.org'
[09:20:39][C][sntp:045]:   Server 2: '1.pool.ntp.org'
[09:20:39][C][sntp:046]:   Server 3: '2.pool.ntp.org'
[09:20:39][C][sntp:047]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
[09:20:39][D][sensor:092]: 'suiveur_1axe_azimuth_soleil': Sending state 93.67138 ° with 1 decimals of accuracy
[09:20:39][D][sensor:092]: 'suiveur_1axe_azimuth_difference': Sending state 93.67138 ° with 1 decimals of accuracy
[09:20:39][D][sensor:092]: 'suiveur_1axe_azimuth_plateau_smoothed': Sending state 0.00000 ° with 1 decimals of accuracy
[09:20:40][C][wifi_signal.sensor:009]: WiFi Signal 'WiFi puissance_suiveur_1axe'
[09:20:40][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[09:20:40][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[09:20:40][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[09:20:40][D][sensor:092]: 'suiveur_1axe_azimuth_plateau': Sending state 0.00000 ° with 2 decimals of accuracy
[09:20:40][D][sensor:092]: 'WiFi puissance_suiveur_1axe': Sending state -57.00000 dB with 0 decimals of accuracy
[09:20:40][D][sensor:092]: 'suiveur_1axe_azimuth_soleil': Sending state 93.67435 ° with 1 decimals of accuracy
[09:20:40][D][sensor:092]: 'suiveur_1axe_azimuth_difference': Sending state 93.67435 ° with 1 decimals of accuracy
[09:20:40][D][sensor:092]: 'suiveur_1axe_azimuth_plateau_smoothed': Sending state 0.00000 ° with 1 decimals of accuracy
[09:20:41][D][sensor:092]: 'suiveur_1axe_azimuth_plateau': Sending state 0.00000 ° with 2 decimals of accuracy
[09:20:41][D][sensor:092]: 'suiveur_1axe_azimuth_soleil': Sending state 93.67732 ° with 1 decimals of accuracy
[

OttoWinter, 2020-07-30:
Bisected this to this PR: https://github.com/esphome/esphome/pull/1208

Probably due to `PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH` build flag, which disables some LwIP features (maybe including `settimeofday`?)

OttoWinter, 2020-07-30:
Ok, not related to `PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH`, but to the new arduino framework core.

Looks like their settimeofday function no longer works.

OttoWinter, 2020-07-30:
I see, the newer ESP8266 arduino framework has a broken `settimeofday` implementation. It returns `EINVAL` when the timezone parameter is not NULL (but ESP32 core expects it NOT to be null, so annoying)

SeByDocKy, 2020-07-30:
> I see, the newer ESP8266 arduino framework has a broken `settimeofday` implementation. It returns `EINVAL` when the timezone parameter is not NULL (but ESP32 core expects it NOT to be null, so annoying)

Annoying definitively ....

closed 2020-07-30 by OttoWinter