esphome/issues #1386: xtensa-lx106-elf-g++: not found

gitolicious, 2020-07-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

_Hass.io on Odroid N2_

 Hass.io Add-on: ESPHome (beta)
 Beta version of ESPHome Hass.io add-on.

 Add-on version: 1.15.0b3
 You are running the latest version of this add-on.
 System: Debian GNU/Linux 9 (stretch)  (armv7 / odroid-xu)
 Home Assistant Core: 0.113.2
 Home Assistant Supervisor: 229

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, but irrelevant

**ESPHome version (latest production, beta, dev branch)**

1.15.0b3 - fails
latest dev - fails
1.14.5 - works fine

**Affected component:**

_n/a_

**Description of problem:**
Compilation is giving me `xtensa-lx106-elf-g++: not found`

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP8266
  board: esp01_1m
```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/test/.pioenvs
INFO Deleting /data/test/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test
Processing test (board: esp01_1m; framework: arduino; platform: espressif8266@2.6.0)
--------------------------------------------------------------------------------
PackageManager: Installing toolchain-xtensa @ ~2.40802.191122
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
toolchain-xtensa @ 2.40802.200502 has been successfully installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20702.0 (2.7.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/test/.pioenvs/test/src/esphome/core/application.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/test/.pioenvs/test/src/esphome/core/component.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/test/.pioenvs/test/src/esphome/core/controller.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/test/.pioenvs/test/src/esphome/core/esphal.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/test/.pioenvs/test/src/esphome/core/helpers.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/test/.pioenvs/test/src/esphome/core/application.cpp.o] Error 127
Compiling /data/test/.pioenvs/test/src/esphome/core/log.cpp.o
*** [/data/test/.pioenvs/test/src/esphome/core/component.cpp.o] Error 127
*** [/data/test/.pioenvs/test/src/esphome/core/controller.cpp.o] Error 127
*** [/data/test/.pioenvs/test/src/esphome/core/esphal.cpp.o] Error 127
*** [/data/test/.pioenvs/test/src/esphome/core/helpers.cpp.o] Error 127
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/test/.pioenvs/test/src/esphome/core/log.cpp.o] Error 127
========================= [FAILED] Took 24.86 seconds =========================
```

**Additional information and things you've tried:**
* Tried master branch with gitpod.io, works fine there
* Searched for the binary within the docker container, exists at
```
root@15ef4d2f-esphome-beta:~/.platformio/packages/toolchain-xtensa/bin# ls -la ./xtensa-lx106-elf-g++ 
-rwxr-xr-x 1 root root 819856 Jul 17  2019 ./xtensa-lx106-elf-g++
```

gitolicious, 2020-07-30:
#594 and #236 might be related

OttoWinter, 2020-08-06:
Odroid N2 is an armv7 device, right?

Notable is that it installs an extra toolchain here:

```
PackageManager: Installing toolchain-xtensa @ ~2.40802.191122
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
toolchain-xtensa @ 2.40802.200502 has been successfully installed!
```

Not sure why that's happening as the image is supposed to contain the right toolchain already.

Command not found could also mean a broken symlink, so maybe check with `readlink` too.

In the docker base image build the xtensa-lx106-elf-g++ command does exist: https://github.com/esphome/esphome-docker-base/runs/925103486?check_suite_focus=true#step:9:133 (ignore the `-fail`, that's just to create a fake build so the toolchain gets downloaded).

gitolicious, 2020-08-06:
Odroid N2 has ARM Cortex-A73 and A53 CPUs which are armv8.

```
root@15ef4d2f-esphome-dev:/# lscpu 
Architecture:        aarch64
Byte Order:          Little Endian
CPU(s):              6
On-line CPU(s) list: 0-5
Thread(s) per core:  1
Core(s) per socket:  3
Socket(s):           2
Vendor ID:           ARM
Model:               4
Model name:          Cortex-A53
Stepping:            r0p4
CPU max MHz:         1896.0000
CPU min MHz:         100.0000
BogoMIPS:            48.00
Flags:               fp asimd evtstrm aes pmull sha1 sha2 crc32
```

Where do you want me set the `readlink` command?
```
root@15ef4d2f-esphome-dev:/# readlink ~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++ 
root@15ef4d2f-esphome-dev:/# 
```

gitolicious, 2020-08-06:
Is https://github.com/esphome/esphome-docker-base/issues/5 related in any way?

gitolicious, 2020-08-09:
Investigating further, I saw that builds actually fail for ESP32 boards in the current stable version (1.14.5) as well:
```yaml
esphome:
  name: test32
  platform: ESP32
  board: esp32dev
```
There the _Download_ part is not shown in the logs...
```
INFO Reading configuration /config/esphome/test32.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test32
Processing test32 (board: esp32dev; framework: arduino; platform: espressif32@1.12.1)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
Compiling /data/test32/.pioenvs/test32/src/esphome/core/application.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/test32/.pioenvs/test32/src/esphome/core/component.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/test32/.pioenvs/test32/src/esphome/core/controller.cpp.o
Compiling /data/test32/.pioenvs/test32/src/esphome/core/esphal.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
sh: 1: xtensa-esp32-elf-g++: not found
Compiling /data/test32/.pioenvs/test32/src/esphome/core/helpers.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/test32/.pioenvs/test32/src/esphome/core/application.cpp.o] Error 127
Compiling /data/test32/.pioenvs/test32/src/esphome/core/log.cpp.o
*** [/data/test32/.pioenvs/test32/src/esphome/core/component.cpp.o] Error 127
*** [/data/test32/.pioenvs/test32/src/esphome/core/controller.cpp.o] Error 127
*** [/data/test32/.pioenvs/test32/src/esphome/core/esphal.cpp.o] Error 127
*** [/data/test32/.pioenvs/test32/src/esphome/core/helpers.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/test32/.pioenvs/test32/src/esphome/core/log.cpp.o] Error 127
========================== [FAILED] Took 2.35 seconds ==========================
```

OttoWinter, 2020-08-11:
@gitolicious No https://github.com/esphome/esphome-docker-base/issues/5 is not related.

I tried to manually set up a qemu aarch64 VM using the `raspi3` machine profile (appears to match your platform the most), and it worked fine there. Don't know what could be the issue here.

gitolicious, 2020-08-12:
Thx for looking into it. Any idea how I could investiage further on my end? Tried uninstalling the add-on and deleted everything I found in the data folder other than the config YAMLs. Reinstalled and ended up with the same error.
This weekend probably I can boot from a different SD with a fresh HomeAssistant image just to be sure there is no interference from any other component or setting.

motwok, 2020-08-31:
I have exactly the same problems with aarch64 on raspi4:

just the name of the executable is different

esphome on some older version (from january):
ESP8266 compiles
ESP32 compiles

esphome 1.14.5:
ESP8266 compiles
ESP32 not compiles (did not find the compiler)

esphome 1.15.5:
ESP8266 not compiles (did not find the compiler)
ESP32 not compiles (did not find the compiler)

after changing to arm71 it works.

Regards
Emmo

gitolicious, 2020-09-01:
Good to know I am not alone on this... I didn't find the time yet to set up a fresh system on my N2 to reconfirm the issue.
@motwok: arm71 is the 32bit version of HomeAssistant for Raspi4?

gitolicious, 2020-09-16:
This nasty bugs now sneaked its way into the stable release `1.15.1`. Will be interesting to see if this brings in more reports from affected users.

keteflips, 2020-09-18:
Same problem here.

keteflips, 2020-09-18:
> I have exactly the same problems with aarch64 on raspi4:
>
> just the name of the executable is different
>
> esphome on some older version (from january):
> ESP8266 compiles
> ESP32 compiles
>
> esphome 1.14.5: [...]

How I can change it in the ESPHome addon for Home Assistant?

Haringstad, 2020-09-19:
And I can confirm: Same issue here.

```
INFO Reading configuration /config/esphome/19_inch_fans.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/19_inch_fans
Processing 19_inch_fans (board: esp01_1m; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20502.0 (2.5.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa 2.100100.200706 (10.1.0)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
Compiling /data/19_inch_fans/.pioenvs/19_inch_fans/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/19_inch_fans/.pioenvs/19_inch_fans/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/19_inch_fans/.pioenvs/19_inch_fans/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/19_inch_fans/.pioenvs/19_inch_fans/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/19_inch_fans/.pioenvs/19_inch_fans/src/esphome/components/api/api_server.cpp.o
*** [/data/19_inch_fans/.pioenvs/19_inch_fans/src/esphome/components/api/api_pb2.cpp.o] Error 127
*** [/data/19_inch_fans/.pioenvs/19_inch_fans/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/19_inch_fans/.pioenvs/19_inch_fans/src/esphome/components/api/api_server.cpp.o] Error 127
========================== [FAILED] Took 3.26 seconds ==========================
```

nordscan, 2020-09-20:
for me too

```
rp4
2020-08-20-raspios-buster-arm64-lite.img
EspHome 1.15.2
```

`INFO Reading configuration /config/esphome/kotolna.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/kotolna
Processing kotolna (board: nodemcuv2; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/kotolna/.pioenvs/kotolna/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/kotolna/.pioenvs/kotolna/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/kotolna/.pioenvs/kotolna/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/kotolna/.pioenvs/kotolna/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling /data/kotolna/.pioenvs/kotolna/src/esphome/components/api/api_server.cpp.o
*** [/data/kotolna/.pioenvs/kotolna/src/esphome/components/api/api_pb2.cpp.o] Error 127
*** [/data/kotolna/.pioenvs/kotolna/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/kotolna/.pioenvs/kotolna/src/esphome/components/api/api_server.cpp.o] Error 127
========================== [FAILED] Took 2.89 seconds ==========================`

gitolicious, 2020-09-21:
Thanks a lot for the MeToos.
@keteflips @Haringstad @nordscan 
Can you guys please share your environments (hardware, HomeAssistant version and installation method) so we can identify root causes?

nordscan, 2020-09-21:
`rasbian - docker - hassio`

```
arch | aarch64
-- | --
chassis |  
dev | false
docker | true
docker_version | 19.03.13
hassio | true
host_os | Debian GNU/Linux 10 (buster)
installation_type | Home Assistant Supervised
os_name | Linux
os_version | 5.4.51-v8+
python_version | 3.8.5
supervisor | 245
timezone | UTC
version | 0.115.2
virtualenv | false

```

gitolicious, 2020-09-21:
Can't figure out what hardware you are running on from what you have posted.

triphoppingman, 2020-09-21:
Hardware:  Odroid-N2
Ambian - Docker - hassio

Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/bnc60_01/.pioenvs/bnc60_01/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/bnc60_01/.pioenvs/bnc60_01/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/bnc60_01/.pioenvs/bnc60_01/src/esphome/components/api/api_pb2_service.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/bnc60_01/.pioenvs/bnc60_01/src/esphome/components/api/api_server.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling /data/bnc60_01/.pioenvs/bnc60_01/src/esphome/components/api/list_entities.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found

Haringstad, 2020-09-21:
Raspberry Pi 4, 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux

arch | aarch64
-- | --
chassis |  
dev | false
docker | true
docker_version | 19.03.13
hassio | true
host_os | Debian GNU/Linux 10 (buster)
installation_type | Home Assistant Supervised
os_name | Linux
os_version | 5.4.51-v8+
python_version | 3.8.5
supervisor | 245
timezone | Europe/Amsterdam
version | 0.115.2
virtualenv | false

gitolicious, 2020-09-22:
So up to now we only have Odroid N2 and RPi 4 affected. @OttoWinter, any idea what could be different for these devices compared to others? Or is it just these devices being the most popular ones in the community?

triphoppingman, 2020-09-22:
It appears we are dealing with a problem with aarch64 platforms (64bit ARM).   https://en.wikipedia.org/wiki/AArch64

triphoppingman, 2020-09-25:
More information:

So I did find where platformio (which esphome uses) stores the compilers:

`root@15ef4d2f-esphome:/root/.platformio$ find . -name '*g++*' -print                                                  
./packages/toolchain-xtensa@2.40802.200502/xtensa-lx106-elf/bin/g++
./packages/toolchain-xtensa@2.40802.200502/bin/xtensa-lx106-elf-g++
./packages/toolchain-xtensa@2.40802.200502/share/man/man1/xtensa-lx106-elf-g++.1
./packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++
./packages/toolchain-xtensa/share/man/man1/xtensa-lx106-elf-g++.1
./packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++
./packages/tool-scons/engine/SCons/Tool/g++.py
root@15ef4d2f-esphome:/root/.platformio$ ./packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
`

It looks like an issue with platformio / esphome.

triphoppingman, 2020-09-27:
I was able to get this working by adding the following to the core block:

`  arduino_version: 2.5.1`

These seems to be the last version of the arduino framework that provides the correct compiler.   Sigh.   This is not a great solution as we are missing out on all the fixes that come with the Arduino code.

gitolicious, 2020-09-28:
Hooray, at least we have a workaround! Thanks for the info. For now, a little outdated version is better than no version. Also we have a starting point for investigations.

gitolicious, 2020-09-28:
Unfortunately for ESP32 I didn't find any version that works. Tried all listed in [the docs](https://esphome.io/components/esphome.html#arduino-version).

keteflips, 2020-09-28:
I have a custom Home Assistant instalation in a TVBOX with Armbian Linux (arm64).

> I was able to get this working by adding the following to the core block:
>
> ` arduino_version: 2.5.1`
>
> These seems to be the last version of the arduino framework that provides the correct compiler. [...]

This workarround didnt work for me.

motwok, 2020-10-05:
> How I can change it in the ESPHome addon for Home Assistant?

the arch can be changed in boot/config.txt

In **my installation** i changed the line
`arm_64bit=1`
to
`arm_64bit=0`

Please check documentation for your installation first, i take no responsibility for you installation!

Emmo

triphoppingman, 2020-10-05:
> > How I can change it in the ESPHome addon for Home Assistant?
> 
> the arch can be changed in boot/config.txt
> 
> In **my installation** i changed the line
> `arm_64bit=1`
> to
> `arm_64bit=0`
> 
> Please check documentation for your installation first, i take no responsibility for you installation!
> 
> Emmo

This only applies for Raspberry Pi 4.0.

motwok, 2020-10-13:
> This only applies for Raspberry Pi 4.0.
You are right... had overread that a pi3 was mentioned... then it might be just running the "original armv7 kernel"!?

buettnc, 2020-11-11:
Wondering if there was any progress on this issue.  I am seeing this missing compiler error in container image 1.27.4.  I have tried changing the RASPBERRY Pi 4 Model B 4GB (Linux raspberrypi 5.4.72-v8+ #1356 SMP PREEMPT Thu Oct 22 13:58:52 BST 2020 aarch64 GNU/Linux)  from arm_64bit=1 to 0 with no success.  I have also confirmed that the compiler is in the container.  I updated that PATH variable in the container to include /root/.platformio/packages/toolchain-xtensa32/bin but that resulted in a different error.

INFO Reading configuration config/home_esp32_bluetooth_hub_1.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d config/home_esp32_bluetooth_hub_1
Processing home_esp32_bluetooth_hub_1 (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.12.4)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/home_esp32_bluetooth_hub_1/src/esphome/components/api/api_connection.cpp.o
xtensa-esp32-elf-g++: error trying to exec '/config/config/home_esp32_bluetooth_hub_1/ "/"/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus': execv: No such file or directory
Compiling .pioenvs/home_esp32_bluetooth_hub_1/src/esphome/components/api/api_pb2.cpp.o
xtensa-esp32-elf-g++: error trying to exec '/config/config/home_esp32_bluetooth_hub_1/ "/"/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus': execv: No such file or directory
Compiling .pioenvs/home_esp32_bluetooth_hub_1/src/esphome/components/api/api_pb2_service.cpp.o
xtensa-esp32-elf-g++: error trying to exec '/config/config/home_esp32_bluetooth_hub_1/ "/"/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus': execv: No such file or directory
*** [.pioenvs/home_esp32_bluetooth_hub_1/src/esphome/components/api/api_connection.cpp.o] Error 1
Compiling .pioenvs/home_esp32_bluetooth_hub_1/src/esphome/components/api/api_server.cpp.o
*** [.pioenvs/home_esp32_bluetooth_hub_1/src/esphome/components/api/api_pb2.cpp.o] Error 1
*** [.pioenvs/home_esp32_bluetooth_hub_1/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
xtensa-esp32-elf-g++: error trying to exec '/config/config/home_esp32_bluetooth_hub_1/ "/"/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus': execv: No such file or directory
*** [.pioenvs/home_esp32_bluetooth_hub_1/src/esphome/components/api/api_server.cpp.o] Error 1
========================== [FAILED] Took 3.42 seconds ==========================

gitolicious, 2020-11-11:
Unfortunate no progress and no ideas or hints how to fix this. For now I am stranded with the `arduino_version: 2.5.1` workaround for ESP8266.  ESP32 I have to compile elsewhere.

buettnc, 2020-11-19:
Here is some additional information:
I updated the container path to include the path to the compiler and tried to compile, received the following:
Compiling .pioenvs/bhive_esp32_bluetooth_hub_1/src/esphome/components/api/api_connection.cpp.o
xtensa-esp32-elf-g++: error trying to exec '/config/config/bhive_esp32_bluetooth_hub_1/ "/"/.
platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus': execv: No such file or directory
Compiling .pioenvs/bhive_esp32_bluetooth_hub_1/src/esphome/components/api/api_pb2.cpp.o
xtensa-esp32-elf-g++: error trying to exec '/config/config/bhive_esp32_bluetooth_hub_1/ "/"/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus': execv: No such file or directory
Compiling .pioenvs/bhive_esp32_bluetooth_hub_1/src/esphome/components/api/api_pb2_service.cpp.o
xtensa-esp32-elf-g++: error trying to exec '/config/config/bhive_esp32_bluetooth_hub_1/ "/"/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus': execv: No such file or directory
*** [.pioenvs/bhive_esp32_bluetooth_hub_1/src/esphome/components/api/api_connection.cpp.o] Error 1
Compiling .pioenvs/bhive_esp32_bluetooth_hub_1/src/esphome/components/api/api_server.cpp.o
*** [.pioenvs/bhive_esp32_bluetooth_hub_1/src/esphome/components/api/api_pb2.cpp.o] Error 1
*** [.pioenvs/bhive_esp32_bluetooth_hub_1/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
xtensa-esp32-elf-g++: error trying to exec '/config/config/bhive_esp32_bluetooth_hub_1/ "/"/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/cc1plus': execv: No such file or directory
*** [.pioenvs/bhive_esp32_bluetooth_hub_1/src/esphome/components/api/api_server.cpp.o] Error 1

Checked the cc1plus compiler at that path to see if all .so files were available
root@18986ad250c6:/root/.platformio/packages/toolchain-xtensa32/libexec/gcc/xtensa-esp32-elf/5.2.0# ldd cc1plus
        libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0xf7733000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xf7720000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xf769f000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xf75a7000)
        /lib/ld-linux-armhf.so.3 (0xf7846000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xf757e000)

All good, attempt to ls the file referenced by the path 
root@18986ad250c6:/# ls -l /config/config/bhive_esp32_bluetooth_hub_1/ "/"/.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0
ls: cannot access '//.platformio/packages/toolchain-xtensa32/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0': No such file or directory

Seems to collapse the "/" into a // instead of the /\"/\"/,  any thoughts on why this may be happening?

Vartkat, 2021-02-01:
> I was able to get this working by adding the following to the core block:
> 
> ` arduino_version: 2.5.1`
> 
I'd love to test this solution but in which file do you modify the "core block" ?

gitolicious, 2021-02-01:
Just put the following in your device config:
```
esphome:
  arduino_version: 2.5.1
```

Vartkat, 2021-02-01:
> Just put the following in your device config:
> 
> ```
> esphome:
>   arduino_version: 2.5.1
> ```
Wonderfull ! Thanks very much, that one did the trick.

netweaver1970, 2021-02-24:
any update on this? the workaround above doesn't work for me, similar errors (after install of the pinned arduino version.
I'm running 64 bit kernel on raspi 3B, in a managed hassio install on Buster. 1st time tying to compile after going to the 64 bit kernel, most likely due to that change.

`Processing bedleft (board: esp01_1m; framework: arduino; platform: espressif8266@2.1.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20501.190515 (2.5.1) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa 1.40802.2 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/bedleft/.pioenvs/bedleft/src/esphome/components/api/api_connection.cpp.o
xtensa-lx106-elf-g++: error trying to exec 'cc1plus': execvp: No such file or directory
Compiling /data/bedleft/.pioenvs/bedleft/src/esphome/components/api/api_pb2.cpp.o
xtensa-lx106-elf-g++: error trying to exec 'cc1plus': execvp: No such file or directory
Compiling /data/bedleft/.pioenvs/bedleft/src/esphome/components/api/api_pb2_service.cpp.o
xtensa-lx106-elf-g++: error trying to exec 'cc1plus': execvp: No such file or directory
*** [/data/bedleft/.pioenvs/bedleft/src/esphome/components/api/api_connection.cpp.o] Error 1
Compiling /data/bedleft/.pioenvs/bedleft/src/esphome/components/api/api_server.cpp.o
*** [/data/bedleft/.pioenvs/bedleft/src/esphome/components/api/api_pb2.cpp.o] Error 1
*** [/data/bedleft/.pioenvs/bedleft/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
xtensa-lx106-elf-g++: error trying to exec 'cc1plus': execvp: No such file or directory
*** [/data/bedleft/.pioenvs/bedleft/src/esphome/components/api/api_server.cpp.o] Error 1
========================= [FAILED] Took 12.87 seconds =========================`

Tryfos, 2021-03-30:
Has this ever been solved? I'm also running Raspbian 64 (Debian) on a Pi4 Buster with a supervised install of HA. All compiles, ESP32 and ESP8266, fail with the same error.
I don't know what are the consequences of using the older Arduino framework. It solves the problem, but is just a workaround, not the solution.

motwok, 2021-03-31:
> Has this ever been solved? I'm also running Raspbian 64 (Debian) on a Pi4 Buster with a supervised install of HA. All compiles, ESP32 and ESP8266, fail with the same error.
Not that i know...

> I don't know what are the consequences of using the older Arduino framework. It solves the problem, but is just a workaround, not the solution.
The "Optimal Workaround" is switching to 32 bit...

Emmo

Bobbik1, 2021-04-03:
Exactly same issue here, debian-docker-HA Rpi3

Bobbik1, 2021-04-03:
> > Just put the following in your device config:
> > ```
> > esphome:
> >   arduino_version: 2.5.1
> > ```
> 
> Wonderfull ! Thanks very much, that one did the trick.

where actually i could find it?

grizzlylife, 2021-07-07:
raspberry pi 3b+ same issue, RaspberryOS-docker-HA.
get when i changed boot/config.txt the line
arm_64bit=1
to
arm_64bit=0
but i need that else traccar dosn't work =(
Anything else that helps besides arduino_version: 2.5.1?

triphoppingman, 2021-07-07:
Speaking for myself, I moved off the ARM64 platform.

gitolicious, 2021-07-07:
After quite some time I retried today and now everything works fine when removing the `arduino_version: 2.5.1` workaround with both `dev` and latest stable version.

```
 Add-on version: 1.19.4
 You are running the latest version of this add-on.
 System: Debian GNU/Linux 10 (buster)  (aarch64 / odroid-n2)
 Home Assistant Core: 2021.6.6
 Home Assistant Supervisor: 2021.06.8
```

How about the others?

motwok, 2021-07-08:
> ... removing the `arduino_version: 2.5.1` workaround ...
How to do i do this?

> How about the others?
sounds good... will try on Weekend.

Emmo

gitolicious, 2021-07-08:
> > ... removing the `arduino_version: 2.5.1` workaround ...
> How to do i do this?

Undo this:

> Just put the following in your device config:
> 
> ```
> esphome:
>   arduino_version: 2.5.1
> ```

If you haven't done it before, you didn't implement the workaround.

grizzlylife, 2021-07-08:
Add-on version: 1.19.4
aarch64
core-2021.6.6
supervisor-2021.06.8

..it hasn't been worked yet :(

Lipown, 2021-08-03:
Workaround flashes the board however the board keeps restarting with this error:

Connected and then restart ..

[13:46:38]
[13:46:38] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[13:46:38]
[13:46:38]wdt reset
[13:46:38]load 0x4010f000, len 1384, room 16 
[13:46:38]tail 8
[13:46:38]chksum 0x2d
[13:46:38]csum 0x2d
[13:46:38]vac02aff5
[13:46:38]~ld
[13:46:38][I][logger:170]: Log initialized
[13:46:39][C][ota:378]: There have been 9 suspected unsuccessful boot attempts.
[13:46:39][I][app:029]: Running through setup()...
[13:46:39][C][wifi:037]: Setting up WiFi...

eisengrau, 2021-08-09:
Can confirm the issue under Alpine Linux current edge/testing brach. Esphome installed via pip3 install tornado esptool esphome, nothning else really. This is not an Alpine docker container, instead it is running under LXC...
The workaround posted above would not work in my case as I get the below when validating:

  Unfortunately the arduino framework version '2.5.1' is unsupported at this time. You can override this by manually using espressif32@<platformio version>.

Esphome version is v1.20.4. I should mention that I have tried the same within an LXC Debian and Ubuntu containers as well, the error is the same:

Compiling .pioenvs/test/src/esphome/components/api/api_connection.cpp.o
sh: xtensa-esp32-elf-g++: not found
*** [.pioenvs/test/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/test/src/esphome/components/api/api_pb2.cpp.o
sh: xtensa-esp32-elf-g++: not found
*** [.pioenvs/test/src/esphome/components/api/api_pb2.cpp.o] Error 127
========================== [FAILED] Took 2.60 seconds ==========================

So this is not really something distribution specific, however in case of Alpine linux, that might be the case, as there is no "sh".

yannpub, 2021-09-14:
I tried to compile for a Sonoff TH16. 

My setup is a supervised HA, VM running Debian Bullseye on an ARM64 box.
Add-On Version is 2021.8.2

INFO Reading configuration /config/esphome/sonoff.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/sonoff
Processing sonoff (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
Tool Manager: Installing toolchain-xtensa @ ~2.40802.191122
Tool Manager: Warning! More than one package has been found by toolchain-xtensa @ ~2.40802.191122 requirements:
 - platformio/toolchain-xtensa @ 2.100300.210717
 - mcspr/toolchain-xtensa @ 5.100300.210830
 - tasmota/toolchain-xtensa @ 5.100200.210303
Tool Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa @ 2.40802.200502 has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/sonoff/.pioenvs/sonoff/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [/data/sonoff/.pioenvs/sonoff/src/esphome/components/api/api_connection.cpp.o] Error 127
========================= [FAILED] Took 158.41 seconds =========================

mburwell80, 2021-09-27:
I am on Win 10 64 machine and I also have this same error trying to flash an esp D1

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mike\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\mike\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\mike\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=4M2M,led=1,sdk=nonosdk_190703,ip=hb2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10815 -build-path C:\Users\mike\AppData\Local\Temp\arduino_build_998064 -warnings=none -build-cache C:\Users\mike\AppData\Local\Temp\arduino_cache_228381 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -verbose C:\Users\mike\Documents\Arduino\sketch_sep27a\sketch_sep27a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mike\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\mike\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\mike\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=4M2M,led=1,sdk=nonosdk_190703,ip=hb2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=1A86_7523 -ide-version=10815 -build-path C:\Users\mike\AppData\Local\Temp\arduino_build_998064 -warnings=none -build-cache C:\Users\mike\AppData\Local\Temp\arduino_cache_228381 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -verbose C:\Users\mike\Documents\Arduino\sketch_sep27a\sketch_sep27a.ino
Using board 'generic' from platform in folder: C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2
Using core 'esp8266' from platform in folder: C:\Users\mike\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2
Detecting libraries used...
"C:\\Users\\mike\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE "-IC:\\Users\\mike\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/include" "-IC:\\Users\\mike\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/lwip2/include" "-IC:\\Users\\mike\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\mike\\AppData\\Local\\Temp\\arduino_build_998064/core" -c -w -Werror=return-type -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_I [... cut ...] Users\\mike\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2\\variants\\generic" "C:\\Users\\mike\\AppData\\Local\\Temp\\arduino_build_998064\\sketch\\sketch_sep27a.ino.cpp" -o nul
Alternatives for cannot execute 'cc1plus': []
ResolveLibrary(cannot execute 'cc1plus')
xtensa-lx106-elf-g++: fatal error: cannot execute 'cc1plus': CreateProcess: No such file or directory  -> candidates: []
compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

barryodonovan, 2021-10-29:
I upgraded ESPHome in my HomeAssistant (Raspberry Pi 4 core-21.10.6 ) earlier today and hit this issue.
After a couple of hours of reading and trying to figure out how to access the underlying OS (I really like to low effort of the OOB image but hate how 'user friendly' it is!) I finally just uninstalled and then re-installed the add-on. This time I updated my devices (diy ESP8266 stuff) and walked away while it installed all the bits (instead of being impatient and presuming it was done when it stopped responding for 15 minutes!).

xtensa toolchain 2.40802.0 installed, all good, everything working again...

ymmv

yannpub, 2021-11-02:
@barryodonovan I just tried the same uninstall/install the addon within HA. But unfortunately not with the same result :-(
Still the infamous `sh: 1: xtensa-lx106-elf-g++: not found`

motwok, 2021-11-03:
> ... I finally just uninstalled and then re-installed the add-on. 

What Addon did you exactly reinstall... something INSIDE the Docker-Image oder the ESP-Home HA Plugin?

Emmo

barryodonovan, 2021-11-03:
I uninstalled the add-on in the Add-On Store, then re-installed it.

Barry

Riva34, 2021-11-04:
I have tried various methods but none worked for me.  I have given up and used Linux esphome command to compile and flash.

wijnandtop, 2021-11-15:
Hi Guys,

Running version 2021.11.3 of HA and version 2021.10.3 of ESPHome.

Had the same issue while compiling code for: ESP8266 / esp01_m1

Solved the issue by using arduino framework version 3.0.2

So changing code to:
`esphome:
  name: device-name
  platform: ESP8266
  board: esp01_1m
  arduino_version: 3.0.2`

K-4U, 2021-11-21:
None of these have worked for me. Even tried installing everything myself on the docker container but that made things even worse. How is this issue running for so long!?

mmakaay, 2021-11-22:
> How is this issue running for so long!?

Because nobody came up with a solid fix so far?
What might help is to provide some more information on your setup and what exact steps you tried, since you report that proposed solutions don't work for you. That indicates that there might be another variable at play in your case.

gitolicious, 2021-11-22:
The original issue was opened by me and for my setup it was resolved somehow. I left the ticket open as others are still facing problems. +1 for the proposal from @mmakaay, everyone who has problems should share their setup and exact problems so we can identify similarities.

K-4U, 2021-11-22:
Fair enough, i was venting after a few hours of sunday frustration. Apologies.

I reinstalled the latest version of the esphome manager in Home Assistant and created a brand new esp8266 nodemcuv2 project and wanted to check if it could install on the esp8266 i had.

That failed compilation with the same errors as above, even with different arduino_version. I even tried the same version that i had running for another project.
But just the barebones project with
```yaml
esphome:
  name: 3d-printer-ledstrip
  platform: esp8266
  board: nodemcuv2```
doesn't work.

[140 more replies]

closed 2022-08-02 by ssieb