esphome/issues #1387: HASSIO: ESPHome esptool not found - Error 127

Kuenstler79, 2020-07-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io Dev 5.0
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8285
**ESPHome version (latest production, beta, dev branch)**

1.14.5
**Affected component:**

**Description of problem:**
When I trying to flash an ESP8266 component with the ESPHome Dashboard or only compile the file, I get the error message: "sh: 1: esptool: not found
***** [/data/steckdose2/.pioenvs/steckdose2/firmware.bin] Error 127"

All my ESP32 components can be flashed over-the-air via the dashboard without any problems.
Also on my MacBook I can flash the ESP8266s with the same YAML file without problems.

I have newly installed Home Assistant on a Raspberry Pi 4 (4GB). However, I had to use the dev V 5.0 (64 bit), because all the others did not boot from USB (SSD).

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: steckdose1
  platform: ESP8266
  board: esp8285
  arduino_version: 2.4.2

wifi:
  ssid: "SSID"
  password: "PASSWORD"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Steckdose1 Fallback Hotspot"
    password: "PASSWORD"

captive_portal:

web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "PASSWORD"

ota:
  password: "PASSWORD"

binary_sensor:
  - platform: gpio
    name: "Schalter Steckdose 1"
    pin:
      number: GPIO13
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      - switch.toggle: relay

switch:
- platform: gpio
  id: blue_led
  pin:
    number: GPIO02
    inverted: True
- platform: gpio
  name: "Relais Steckdose 1"
  pin: GPIO15
  id: relay
  on_turn_on:
  - switch.turn_on: blue_led
  on_turn_off:
  - switch.turn_off: blue_led

# Use red LED for connectivity status indicator
status_led:
  pin:
    number: GPIO0
    inverted: True

sensor:
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: True
    cf_pin: GPIO05
    cf1_pin: GPIO14
    # Higher value gives lower watt readout
    current_resistor: 0.00221
    # Lower value gives lower voltage readout
    voltage_divider: 871
    current:
      name: "Strom Steckdose 1"
      accuracy_decimals: 2
    voltage:
      name: "Spannung Steckdose 1"
    power:
      name: "Leistung Steckdose 1"
    update_interval: 10s
    
  - platform: wifi_signal
    name: "WiFi-Signalstärke Steckdose 1"
    update_interval: 10s
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Reading configuration /config/esphome/steckdose1.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/steckdose1
Processing steckdose1 (board: esp8285; framework: arduino; platform: espressif8266@1.8.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 423.98KB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20402.4 (2.4.2) 
 - tool-esptool 1.413.0 (4.13) 
 - toolchain-xtensa 1.40802.2 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
Compiling /data/steckdose1/.pioenvs/steckdose1/src/main.cpp.o
Linking /data/steckdose1/.pioenvs/steckdose1/firmware.elf
Retrieving maximum program size /data/steckdose1/.pioenvs/steckdose1/firmware.elf
Building /data/steckdose1/.pioenvs/steckdose1/firmware.bin
Checking size /data/steckdose1/.pioenvs/steckdose1/firmware.elf
sh: 1: esptool: not found
*** [/data/steckdose1/.pioenvs/steckdose1/firmware.bin] Error 127
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  50.2% (used 41152 bytes from 81920 bytes)
Flash: [====      ]  40.0% (used 409580 bytes from 1023984 bytes)
========================= [FAILED] Took 11.33 seconds =========================
```

**Additional information and things you've tried:**

OttoWinter, 2020-08-06:
Probably due to `arduino_version: 2.4.2` and a broken package over at platformio (hence I'm marking this as an external issue).

You could try remove that line to see if the error goes away, that would at least confirm it's a broken upstream package. Then we can report that to platformio.

Kuenstler79, 2020-08-06:
Yes, very good! You were right.
Without the Arduino reference, the compiling and uploading works.
Thanks for the tip!

I switched from Tasmota and somehow the first flash run (flashed with MacBook) with ESPHome didn't run through until I explicitly specified the line `arduino_version: 2.4.2`.

closed 2020-08-06 by Kuenstler79

qguv, 2021-02-04:
The docs for the Sonoff S20 include a line for arduino_version in the example configuration—maybe a warning could be added for Home Assistant users?